nRF24/RF24 #346: Sharing SPI port with rf24 (arduino)

noeldum, 2017-03-28:
Using atmega328 I plan to use another chip also connected on SPI (different CS pin obviously).

What are the recommendations to pause the rf24 on SPI so the other chip can also start comms?

I am using a sketch from the mesh example. I introduced my own SPI request just after the mesh.update() but they seem to interfere. Even if forcing the CS pin on rf24. Here is why I am wondering there must be something else to do.

sherikapotein, 2017-03-30:
I use a SD card and nrf24l01p together on my uno. The only preemptive precaution that i take is to deliberately pull the other  modules'(SD module in my case)  CS pin to High(Disabled) every time before the NRF transmission and setting it back to low before any SD transaction. Use port manipulation to achieve it rather than digitalwrite, if your task demands lower latency. This  library, if i remember correctly automatically enables and disables the CS pin low/high before/after every transaction, so no need to worry there.

TMRh20, 2017-04-01:
I think I remember looking at this and the following two lines in [RF24_config.h](https://github.com/nRF24/RF24/blob/master/RF24_config.h#L28) which might need to be moved after the SPI includes so the SPI Transaction code is used properly. Can't remember and don't have time to test, so feel free to try that out. See https://www.arduino.cc/en/Tutorial/SPITransaction

closed 2019-05-10 by TMRh20