nRF24/RF24 #348: default values for delay and count in setRetries

Slartitorto, 2017-04-08:
Hi all, thanks for your great job.
just a question:
I omitted both setAutoAck() and setRetries() statements in my code, but I  see setAutoAck() is enabled by default.
Does setRetries() linked to setAutoAck() ? I mean: if the ack is not received, I think the sender retries as many time as is set in setRetries().
In that case, if setRetries() is not set in the code, what is the default values for delay and count ?

wmarkow, 2017-04-12:
If I see correctly in the source code, then we have:
 * setAutoAck() is never called from the library code (it must/may be called from the final application)
 * setRetries(5, 15) is called from begin() method which sets the delay to 1500us and retry count to 15

Since the library doesn't touch the auto-ack feature by default, the question is what is the default auto-ack status after nRF24L01 power up. According to the documentation, the **EN_AA** register (Auto Acknowledgment) after power up is configured to enable the auto-ack. The default value of retry mechanism (after nRF24L01 power up) is:delay of 250us and retry count 3.

After that analysis we can say that - when using RF24 library:
 * auto-ack is enabled
 * the retry mechanism is configured as: delay 1500 us and 15 retries count

For deep information please take a look at the [7.5 Automatic packet transaction handling](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf)

wmarkow, 2017-04-12:
One more thing: i just was interested what happens when auto-ack is disabled but retries mechanism is enabled. At first you may think that transmitter will put that retry mechanism in action because the receiver will never send the auto-ack.
After a deep look into the nRF24L01 datasheet, it may be that retry mechanism is not used when aut-ack is disabled (even when you set some values in **SETUP_RETR** register). I think it is not written clearly in the datasheet but take a look a [PTX operation Figure 8 in the datasheet](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf#G1153676). In the schematic there is a question **Is Auto Retransmit enabed?** If **Yes** then there is another question **No_ACK?** I think the second question is related to auto-ack feature. If so, then **Auto Retransmission** will work only when **Auto Acknowledgement** is enabled.

closed 2019-05-10 by TMRh20