nRF24/RF24 #353: alternative hardware SPI pins (teensy using audio shield)

matajumotors, 2017-04-30:
I am trying to hook up RF24 module to Teensy 3.2 while using audio shield also.
Audio library moves MISO from pin11->7 and SCK pin 13->14.

Is there a way to make RF24 library to use alternative hardware pins?
I have tried to put  SPI.setMOSI(7); and SPI.setSCK(14); in my setup, but it does not work.

I see that there is an option to use software SPI, but I was not able to compile it sucessfully. Does anyone have working example?

I would prefer to get it working on the alternative hardware SPI pins of Teensy.

Any help would be deeply appreciated.
Thanks

joe318, 2018-04-26:
Hello,
i have this configuration, just add
 SPI.setMOSI(7);
       SPI.setMISO(8);
       SPI.setSCK(14);       // SCK on pin 14
       SPI.begin();

in your setup and it's ok

rica01, 2018-07-13:
WHAT???? THE AUDIO LIBRARY CHANGES THE PIN SETUP WITHOUT TELLING IT ANYWHERE IN THE DOCS???
yes im screaming.

lenapizarro, 2018-07-13:
Hi, how did you find out which pins are being moved? And exactly which audio objects do this? I was using a Teensy 3.6, an audio board and a nRF24, but it's not working. I'm using pins 28 as ce, 39 as csn, mosi as 11, miso as 12 and sck as 13.

mech06, 2018-07-13:
They must be hardware pins.

13 Tem 2018 Cum 03:06 tarihinde lenapizarro <notifications@github.com> şunu
yazdı:

rica01, 2018-07-13:
oh really. hardware pins. really. cuz there is a thing called NON HARDWARE
pins? >.>

Avamander, 2018-07-13:
@rica01 Pretty sure he meant pins with hardware SPI support vs. pins that would require bitbanging spi.

closed 2019-05-10 by TMRh20