nRF24/RF24 #360: radio.available() returns true when nothing is transmitting

mattinmir, 2017-05-24:
I am running the `gettingstarted.cpp` file under RF24/examples_linux on a Raspberry Pi 3B and am attempting to have it communicate to an Arduino running the Arduino version of gettingstarted. (Arduino ping_out, RPi pong_back)

However, the RPi claims to have received messages even when I have disconnected power from the 
Arduino:
[image]

I'll also add that even when the Arduino is powered and transmitting, the RPi shows the same output. 

I have checked my wiring and confirmed the RF24 module is connected using `isChipConnected()` so I don't think that is the problem. Also there is no one nearby also using these modules that could be interfering. 

Is there any other conceivable reason why `radio.available()` might return true? Here is the output of `printDetails()` in case it helps:

[image]

matt769, 2017-07-14:
Power supply issues can definitely produce the effect that you have described in your issue.

I had what appeared to be an identical issue where the GettingStarted sketch was stuck in the  while loop (line 107) since `radio.available()` was constantly returning true, even though there appeared to be nothing to read (other radio was off, or in receive mode).

Note that I was actually able to transmit (sometimes) so the module was definitely connected and I suspect `isChipConnected()` would also have returned true for me (although I was previously unaware of this function so hadn't actually tried it).

Getting to the point - it appeared to be a power supply issue as when I added a proper 3.3V voltage regulator to the supply the problem completely disappeared. So my suggestion would be to check this first and follow all the available guidance on this topic.

rnicolas, 2018-01-05:
I think there's a bug on how pipes are cleared. I'm having more or less the same problem, but when I'm using RF24Network, when I execute network.update() the code is stucked here, inside the loop that is asking for radio.available() and radio.isValid(); when I change the library and I return to the original from maniacbug, I can see this loop is repeated like 30 or 40 before the pipe and the registers are cleared and after that I can send packages. So, I suppose problem might be here. I will update if I can resolve it.

Avamander, 2018-01-05:
I had a similar problem about how my code seemingly froze on .update(), further tracing it I ended up at a spi transaction call. It turned out my code just messed wrote a bit of data to the wrong place. I suggest you do the same and determine the exact line where it freezes.

rnicolas, 2018-01-05:
I was reading you just [here](https://github.com/nRF24/RF24Mesh/issues/45) because the initial post explains the same problem that I'm facing.

The code that I'm running is just the example called helloworld_tx.ino under RF24Network/examples/helloworld_tx/

haseebehsan, 2018-02-17:
@mattinmir that is most certainly power supply issue or pipe clearing issue as explained by @rnicolas , if your issue is resolved let us know. So that it could be closed
I was redirected from [here](https://github.com/nRF24/RF24Mesh/issues/45)

linkhat, 2018-07-14:
Nice to meet you guys. My case is opposite of you all. In mine, radio.available( ) returns always false. I use RF24master library file. I use two arduino uno boards. Pin Configuration is correct. Earlier time, transmitter didn't send successfully and the receiver didn't receive nth. So I rewrite the address 0xE8E8F0E1LL in both codes. Then transmitter sends successfully (Most often it fails but if I touch the input pin, it sends successfully again) However receiver still receive nothing. 4.7uF is connected between 3.3V and Gnd too in the receiver. I don't know where I miss. I hope someone can help me.

linkhat, 2018-07-15:
@Avamander I did as you said but still return false. I doubted nrf so I replaced it with other that is used in transmitter. I tested both with the transmitter program on each arduino board whether both nrf's are good or not. I found that both nrf are fine but one of arduino uno board (used for receiver in earlier time) don't work with new nrf and showing me fail. 

My transmitter code is:

#include<SPI.h>
#include<nRF24L01.h>
#include<RF24.h>

RF24 radio(9,10);
int msg[1];
boolean er;
const uint64_t pipe=0xE8E8F0E1LL;
int btnPin=3;

void setup( ){
  Serial.begin(9600);
  radio.begin( );
  radio.openWritingPipe(pipe);
  radio.stopListening( );
  pinMode(btnPin, INPUT_PULLUP);
}

void loop(){
  Serial.println(digitalRead(btnPin));
  if(!digitalRead(btnPin)){
    msg[0]=11;
    er=radio.write(&msg, sizeof(msg));
    if(er==true)
    Serial.println("Success");
    else
    Serial.println("Fail");
  }
}

My receiver code is:

#include<SPI.h>
#include<nRF24L01.h>
#include<RF24.h>

RF24 radio(9,10);
int msg[1];
boolean er;
const uint64_t pipe=0xE8E8F0E1LL;
int ledPin=3;

void setup( ){
  Serial.begin(9600);
  radio.begin( );
  radio.openReadingPipe(pipe);
  radio.startListening( );
  pinMode(ledPin, INPUT_PULLUP);
}

void loop(){
  bool done=false;
  if(radio.available()){
    while(!done){
       done=radio.read(&msg, sizeof(msg));
       Serial.println(msg[0]);
       if(msg==11)
       digitalWrite(ledPin, HIGH);
       delay(1000);
    }
  else
  Serial.println("No Radio Available");
}

wmarkow, 2018-07-15:
I think there was some similar issue (but do not remember its number): disabling **auto ack** helped to "establish the connection". Just call **setAutoAck(false)** for transmitter and receiver code. I'm not sure if it helps in your case but you can give it a try.

In your **setup()** method try to call **isChipConnected()** and inspect its value; if it is **false** then you have some wiring issues.

linkhat, 2018-07-21:
Hey Guys, now my transmitter transmitts correctly and receiver senses that radio is available. But message is not correctly received. I sent "111" but received "ÿÿÿÿÿÿÿÿ". Also I don't understand why TX led is on at the receiver and RX led is on at the transmitter. I hope you could help me. Thank you.

Avamander, 2018-07-21:
@linkhat Shorten your wires and do not use ribbon cable. Also please use one of the examples, modified as little as possible, then we can be sure it isn't your code and it's something in the hardware.

linkhat, 2018-07-22:
It's ok now. I found error. One of the module is broken down. I just found it today by testing with new and old ones. Thank you all, for helping me. Thanks a lot.

closed 2019-05-10 by TMRh20