nRF24/RF24 #361: Please add RF24Network library support for arduino M0 board.

rafi021, 2017-05-31:
I have 2 arduino m0 board (samd21). I have connected them with 2 nr24l01 modules. but unfortunately RF24 and RF24Network library does not seems to work with arduino m0. I have tested for both the configuration for NRF and Mo board. 
miso=D12, mosi=11,sck=13, cs=8,ce=7 and also with ICSP header pins. But both of the connections are failed to initialize the NRF24l01 module. The NRF24l01 module works fine with aruino pro mini. 
please experts of this fork, need your valuable advice.

pmdmartins123, 2017-06-01:
Hi,

I am also using the nrf24l01+ radios with several arduino's uno. 
I had the same problem when i started to work with the radios. 
I realized that the radio was constantly restarting so I put a capacitator 10u Farad parallel with ground and 3.3V.
And i also realized that the pinout connections were different using different libraries. So the libraries I use are: RF24 and RF24Network that i download it from: https://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo

and the pinout I use is:
9->CE
10->CSN
13->SCK
11->MOSI
12->MISO
for the IRQ i didn't connect it.

I hope this helps.

pmdmartins123, 2017-06-01:
Very important:
The RF24 and the RF24Network libraries are not the same of the ones that can be downloaded from the libraries manager of the arduino ide.

pmdmartins123, 2017-06-01:
I used this site to learn and start to work with the radio.
https://maniacbug.wordpress.com/2011/11/02/getting-started-rf24/

Once more i hope this helps.

rafi021, 2017-06-01:
i have already tired your suggestion. (100uF cap with vcc and gnd pin). But doesn't work for me case. Arduino M0 has different spi pin connection rather than arduino uno. Arduino M0 has spi pinout in ICSP header. I want to use regular spi pinout like arduino uno (miso=12,mosi=11, sck=13).

Avamander, 2017-06-01:
@pmdmartins123 They are rather up-to-date. No major changes have been made since the last release. But just in case I will make a new one.

pmdmartins123, 2017-06-01:
I tried also the 100 u Farad and change it to the 10 uFarad, the first try didn't work so well.

StefanMeGit, 2019-04-06:
I got the exact same issue.. can't get the NRF24 working on the SAMD21... has anyone got it working?

SimonMerrett, 2019-05-20:
@StefanMeGit yes, I have used the GettingStarted with a clone Arduino M0, Arduino IDE 1.8.7 and library version 1.3.2 installed from the library manager. I didn't use an interrupt connection. A couple of changes I had to make:

- Arduino M0 uses SerialUSB instead of Serial (e.g. SerialUSB.begin(115200)). So Find & Replace that.

- Use the 3.3V power pin, CE/CSN as you wish (e.g. D7 and D8) but **DO** use the ICSP connector for SCK, MISO, MOSI, **NOT** D13, D12, D11.

john--, 2019-08-10:
I can confirm what @SimonMerrett  suggests. Using the ICSP pins is the only way I've gotten NRF24 to work on my SAMD21G18A. Has anyone gotten any other SERCOM's to work? I've tried a few different things with no luck.

SimonMerrett, 2019-10-01:
@john-- I have now got a custom board up and running based on SAMD51 with a dedicated SPI port for the NRF24L01. My proof of concept was on an Adafruit ItsyBitsy and there may be some information of general interest for you [here](https://forums.adafruit.com/viewtopic.php?f=62&t=153987).

el-samiyel, 2020-03-08:
Hi @SimonMerrett @john--, are you able to confirm how you got this working on the SAMD21.

I have installed the 1.3.2 library, Used ICSP and SerialUSB but not getting any response from any of the examples. 

Any info will be much appreciated.

SimonMerrett, 2020-03-08:
@el-samiyel do you have any luck using isChipConnected() or isPVariant() ? Do you definitely have CE/CSN correctly numbered? Can you get an oscilloscope or logic analyzer on the SPI lines? If the HW is all set up correctly, did you try any of the workaround in the link I posted above?

el-samiyel, 2020-03-16:
@SimonMerrett Ahh yes, I finally got it connected. My issue was that I was trying to use the scanner example which returned nothing due to the printf function. Has anybody managed to get he scanner working with the samd21? Many Thanks

SimonMerrett, 2020-03-20:
> @SimonMerrett Ahh yes, I finally got it connected. My issue was that I was trying to use the scanner example which returned nothing due to the printf function. Has anybody managed to get he scanner working with the samd21? Many Thanks

Do you have to use printf?

TMRh20, 2020-08-11:
Sounds like its all working, and the scanner example has been recently updated to use Serial.prints instead of printf. Closing.

closed 2020-08-11 by TMRh20