nRF24/RF24 #362: Can the receiver determine the address of the transmitter?

nkolban, 2017-06-13:
Assuming I have two device ... one called Transmitter and the other called Receiver.  If I perform openWritingPipe() specifying address R on Transmitter and call openReadingPipe() specifying address T also on the receiver and finally call openReadingPipe() specifying address R on the receiver ... I believe I can now perform a write() to send data from the Transmitter to the Receiver.  My question is:

1) Does the incoming message on the Receiver "know" the address of the Transmitter so that it can send an automatic acknowledgment?
2) Can code on the Receiver determine the address of the Transmitter so that it can subsequently choose to send a message to it?

TMRh20, 2017-06-13:
1. The transmitter and receiver both use address R so auto ack uses that address. Yes.

2. Each address is assigned to a radio "pipe". The address of received packets can be determined by pipe number. See the class reference for radio.available();

nkolban, 2017-06-13:
Aha ... wow ... thank you kind sir.    That helped me tremendously.  So to paraphrase it back in case there are other readers in the future:

"When a transmitter using auto-ack sends a message to a receiver, after having transmitted the message to the receiver it immediately starts listening ... using the address of the receiver.  The actual receiver receives the message and transmits an acknowledgement using its OWN address (i.e. the address on which the message was received).  This address is now being used by the original transmitter to receive the ack.

closed 2017-06-13 by nkolban