nRF24/RF24 #363: ESP-12F + RF24 stability issue (crashing)

strange-v, 2017-06-14:
Hi.
I'm using Arduino IDE 1.8.2, ESP8266 Community 2.3.0, RF24 from master. Module - ESP-12F (with all required capacitors and pull-up/pull-down resistors), I've selected NodeMCU 1.0 (ESP-12E Module) in Arduino IDE.

I'm experiencing a big stability troubles when using RF24 library and modified GettingStarted example.
1. In receive mode when no one transmit data it works for couple of minutes, then crash (reboot without restoring/continuous rebooting).
2. In receive mode when package arrive every 3 seconds it could work for 1-2 minutes or crash on first package.

On the other hand I've another project which doesn't use RF24 library (a few DS18B20 sensors, web server with static/dynamic pages, web client for posting data) and it quite stable, I've tested for couple of hours on the same hardware - no crashes at all.

Code: https://gist.github.com/strange-v/eb0be1a42eb44f4cb1e29775eeae0da0
Output:

> Opening port
Port open
rl��rߌ#�n�����p�<����8��ǒ��lp�nn��;�nĒ���b�$�rrp�n��܀���l�����b�n��n�$���b�ľ~�n�����l`���#�n��rnr���;����;r�ےn���b��`Ğ�l��Ìr��n�b��
Starting...
Connecting......
Connected, IP address: 172.16.0.110
0
1
2
3
4
Sent response 279426320
Sent response 282435180
Sent response 285446868
��2܃

I'd appreciate any advice to resolve this issue.

csnol, 2017-08-16:
Hi, strange-v:
Almost all of these questions be checked on 3.3V power supply.  
maybe you should recheck it. or try below codes use low speed and low transpower:
``` C++
...........
radio.begin();          // Initialize the nRF24L01 Radio
//radio.setChannel(108);            // set 2.4G channel
radio.setDataRate(RF24_250KBPS);   //250kbbs ,RF24_250KBPS，Low speed.
radio.setPALevel(RF24_PA_LOW);   // LOW Transmit Power.
............
```

TMRh20, 2020-08-11:
Closing as issue is very old and a number of updates have been made to ESP & RF24. Please re-open if still issues with this device.

closed 2020-08-11 by TMRh20