nRF24/RF24 #364: Data transfer with radio.startWrite

Rotario, 2017-06-16:
Hi,
So I've got two NRF24l01+ modules talking, I'm sending data every 2 seconds from a Teensy to a Raspi.
The Teensy calls radio.startWrite and the Raspi has a while(radio.available){ radio.read; .... }
The Teensy is always spitting out data regardless, but the raspi listens via a script. When I run the script, it receives data on the 2 second mark, but If I'm correct, there are 3 datasets on the teensy ready to be sent. When the script runs, only the first dataset into the teensy is read by the raspberry pi. This means it lags by 3 sets of data i.e the teensy writes 3rd data set to the radio while the first is actually sent. I get this is because it's a FIFO buffer, but is there any way to send all 3 packets in sequence until the TX fifo is empty?
Thanks,
Rowan

Raspi dataset read
`
Got payload size=12
 value=15 //hours
 value=23 //mins
 value=26 //secs
 value=19
 value=224
 value=191
 value=39
 value=64
 value=0
 value=0
 value=0
'

Teensy dataset send
'Sending Data
15:23:22
19
2.65
0
'

TMRh20, 2017-06-18:
You can load 3 packets then call radio.txStandBy(); to wait for the fifo to empty.

Rotario, 2017-06-19:
I've tried standby but the radio cuts out once I try to write after calling this, do I need to power it up again somehow once I've run standby?

wmarkow, 2017-07-21:
In your attached source code I see that you are using 
```
void RF24::startWrite( const void* buf, uint8_t len, const bool multicast )
```
 to send data. This method only adds the data to the transmission queue of the RF24 chip. The transmission should be started by the chip automatically. If you have auto ACK enabled and the remote (destination) chip is not operating then the transmitter chip will try to resend the packet until the retry limit is reached and then the **MAX_RT** bit of **STATUS** register is set to one (to indicate the the re-transmission limit has been reached). The nRF24 docs say that this but **must** be cleared by the user in order to enable further communication. The method **void RF24::startWrite( const void* buf, uint8_t len, const bool multicast )** seems to not clear that bit. Could you confirm that in your case?

Second thing: **txStandBy** checks for **MAX_RT** bit of **STATUS** and clears it automatically for you. Additionally this method flushes the entire transmission queue, so packets that are not sent are lost.

Maybe for now (just for tests) you could use the 
```
bool RF24::write( const void* buf, uint8_t len, const bool multicast )
```
method to send data? It will handle the **MAX_RT** bit of **STATUS** automatically and it should work fine when the receiver is off. It will not send data at the max speed but for now you try to send data every two seconds, right?

closed 2019-05-10 by TMRh20