nRF24/RF24 #367: radio.available() is executing constantly even if nothing is sent from master nrf device.

ajinkyaw, 2017-06-27:
I am two nrfs communicating to each other. one is connected to uno and other one is connected to another uno.

the problem i am facing here is , even if my master nrf is not sending anything , at the receiving end radio.available() is repeatedly getting executed and also not showing what i am sending from my master nrf. 

my code is as follows.

Master Nrf (Sender) :
`#include <SoftwareSerial.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(7, 8); // CNS, CE

int count;
char reqst[4] ;
const byte address[6] = "00111";
 
void setup() {
  
   Serial.begin(9600);     

 
   radio.begin();
   radio.openWritingPipe(address);
   radio.stopListening();

}

void loop() {
  
   while ( Serial.available() > 0 )  { 
   
    char inChar = Serial.read();
    
      reqst[count] = inChar;
      count++;
  
    if (inChar == '\n') {

      Serial.print(" String: ");
      Serial.println(reqst);
      radio.write(&reqst, sizeof(reqst));
      memset(reqst, 0, sizeof(reqst));
      count = 0;
      delay(1000);
    
    }
   }

   
}`

receiver nrf :
`#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(7, 8);
const byte rxAddr[6] = "00111";
char charStr[4];
boolean done;
String stri;
byte byteData;

void setup() {

  while (!Serial);
  Serial.begin(9600);
  radio.begin();
  radio.openReadingPipe(0, rxAddr);
  radio.startListening();

}

void loop() {

  if (radio.available()) {

    radio.read(&charStr, sizeof(charStr));
    done = true;

  }

  if (done) {

    Serial.println("received data is : ");
    stri = charStr;
    Serial.println(stri);
    memset(charStr, 0, sizeof(charStr));

  }

  done = false;

  delay(1000);

}`

and my output at receiver end is 

received data is : 
received data is : 
received data is : 
received data is : 
received data is : 
received data is : 

this is showing me per second. 
even if i try to send data , it won't show here.

Please suggest me where i am going wrong.
Its been a lot of time i have been solving this.

Avamander, 2017-06-27:
Check your wiring.

ajinkyaw, 2017-06-28:
@Avamander  : hi . i soldered another prototype just to ensure that there must be loose connections but the same thing happened..

wmarkow, 2017-06-28:
Is it possible that this issue is similar to #360?

Avamander, 2017-06-28:
I get garbage data when touching the data pins of the receiver sometimes or when the wiring is dodgy. One more thing you can try is soldering a large 10000uF electrolytic capacitor directly to the receiver's 3.3 and GND pins.

wmarkow, 2017-07-21:
If you have some hardware issues with the chip, then the **available()** method may give you a weird results; it may say that some data are available even when the transmitter is off. Could you please call the **isChipConnected()** and see what it returns? It could be that in your case it will return **false**, which means that you have some hardware issues between the **RF24** chip and **Arduino Uno**.

ajinkyaw, 2017-09-24:
this was one of the biggest coincidence of my life. three nrfs i bought and all were faulty. 
it was an hardware issue , i bought the new chips and its working fine now. 
Everyone thanks for so much support.

closed 2017-09-24 by ajinkyaw