nRF24/RF24 #368: Bad packets after ~1 min

Rotario, 2017-06-28:
Hi,
Thanks for the library - it's fantastic and the examples are a great start.

My architecture is as follows:
Sender: Teensy 3.6 using startFastWrite and tx.standby after 3 fails.
Receiver: Raspi using your library and BCM2835 using inter
[RadioProgram.txt](https://github.com/nRF24/RF24/files/1108416/RadioProgram.txt)

rupts and a while(true){sleep(10000)} to wait for interrupts.

I have 16-byte packets sending as fast as possible and a 12-byte packet sending every 5 seconds.

On the receiver end I see these packets, couting the packet lengths after every receive. After an amount of time, sometimes 1 min, sometimes 20 mins the packet lengths suddenly change to a seemingly random number - sometimes 28 - sometimes 14. I have a feeling the radio gets a bad packet - malfunctions and this data is just corrupt nonsense. Once this happens I have to restart the program and it works perfectly again. Would it be work doing a printDetails when this payload length suddenly changes to see what's happening with the radio?

Please find the C++ raspi receiver program attached:
[RadioProgram.txt](https://github.com/nRF24/RF24/files/1108418/RadioProgram.txt)

Rotario, 2017-06-28:
Ok, I've added some printDetails() statements where neccessary and the following is produced, the two working states continually switch between each other, these are printed whenever an interrupt occurs, so I assume one is for tx and one is for rx.

The not working state is printed whenever the payload length is not one of the two I'm expecting to receive - in this case it started to receive '6' byte payload lengths.
Hope this helps and I thank you for any help in solving this issue in advance.
Thanks,
Rowan

1st Working State
```
STATUS           = 0x43 RX_DR=1 TX_DS=0 MAX_RT=0 RX_P_NO=1 TX_FULL=1
RX_ADDR_P0-1     = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x22
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x06
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
```
2nd Working State
```
STATUS           = 0x62 RX_DR=1 TX_DS=1 MAX_RT=0 RX_P_NO=1 TX_FULL=0
RX_ADDR_P0-1     = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x22
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x06
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

```

Not Working State

```
Payload Length = 6

STATUS           = 0x63 RX_DR=1 TX_DS=1 MAX_RT=0 RX_P_NO=1 TX_FULL=1
RX_ADDR_P0-1     = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x22
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x06
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
```

Rotario, 2017-06-29:
I think this is a software issue with either the Raspi BCM library handling the interrupts or the Raspi itself, I've disconnected the sending radio competely, leaving the receiving program running, and lo and behold it starts to register ghost payloads. This continues even if I disconnect the interrupt pin between the NRF and the Raspi. This is what leads me to believe there's something in the libraries that just stops working correctly, I'm not enough of a software guy to debug it. I'm going to try the same program on an Odroid and see if the issue persists.

Rotario, 2017-06-29:
Ok, it's not a library issue or a BCM issue! Whenever the radio receives a bad packet length, instead of reading it to a throwaway buffer or flushing it from the FIFO I constantly read that it's available. I've added a radio.read(&throwawaybuffer, len) to the else statement and the problem shouldn't be an issue anymore.

closed 2017-06-29 by Rotario

soligen2010, 2017-06-29:
Just a thought, but you might want to add your own checksum to your packet as well just in case the"bad length" randomly happens to match your good length.  This also helps detect errors transferring data in SPI.  The module has a checksum, but it is checked before you read the data. I have had rare instances of data errors that I have assumed were introduced during the SPI transfer.  Your own checksum would help catch things like this.