nRF24/RF24 #370: Odroid C1+ Compatibility

Rotario, 2017-07-04:
Hi,
I don't think the pinMode and digitalWrite functions are working correctly with the Odroid C1+, the SPI works perfectly fine but the CE pin does not go LOW, nor does it change mode to OUTPUT when called in the code. I think there is a library it's using to handle pinMode and digitalWrite that does not work with this system, when I try to include the Odroid version of wiringPi I get errors presumably because there's already a library included that uses the same function names. and they're colliding in compilation. Any Ideas?

Rotario, 2017-07-04:
Works when using GPIO pin nos - no wPi pin numbers, attachInterrupt doesn't seem to work but more investigation required

Rotario, 2017-07-05:
As per 
https://forum.odroid.com/viewtopic.php?f=115&t=27544&sid=d89d1f74fa2e3bd15f2800839a81915e
it's been suggested that I change the module used from GPIO to standard wiringPi option

Rotario, 2017-07-05:
so I'm ok to leave the GPIO utility running CE pins and SPI etc? just changing the interrupt.c to have wiringPi functions, when I #include wiringpi won't the functions defined in there clash with the existing functions being used in RF24 SPI and GPIO such as pinMode and digitalWrite which are defined as GPIO::open etc?

Avamander, 2017-07-05:
I'm rather confused of what you what you want to do now. You're saying the attachInterrupt() function doesn't work on your board. Just make the installer script autodetect your board and use preprocessor (#ifndef #ifdef etc.) to use the right code that works on your device.

Rotario, 2017-07-05:
The installer script defaults to SPIDEV on the board, which works fine for the SPI, but not for attachInterrupt(), so I'm editing interrupt.c in /utility/SPIDEV/ to include wiringPi and for attachInterrupt to just point to the wiringPi function that should work to interrupt. There's no driver that works on the device for interrupts as there's no option to use wiringPi specifically.

closed 2018-01-08 by Rotario