nRF24/RF24 #373: Compilation Warning  with DUE. (RF24 1.3.0 & IDE 1.8.2)

pierre83, 2017-07-15:
In file included from E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24.h:18:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_modified_sketch_155011\Mouchard_nRF24_1_1.ino:4:

E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24_config.h:154:0: warning: "PSTR" redefined [enabled by default]

   #define PSTR(x) (x)

 ^

In file included from C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:31:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_build_234123\sketch\Mouchard_nRF24_1_1.ino.cpp:1:

C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:34:0: note: this is the location of the previous definition

 #define PSTR(str) (str)

 ^

In file included from E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24.h:18:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_modified_sketch_155011\Mouchard_nRF24_1_1.ino:4:

E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24_config.h:156:0: warning: "strlen_P" redefined [enabled by default]

   #define strlen_P strlen

 ^

In file included from C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:31:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_build_234123\sketch\Mouchard_nRF24_1_1.ino.cpp:1:

C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:69:0: note: this is the location of the previous definition

 #define strlen_P(a) strlen((a))

 ^

In file included from E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24.h:18:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_modified_sketch_155011\Mouchard_nRF24_1_1.ino:4:

E:\Pierre\Documents Pierre\Arduino\libraries\RF24/RF24_config.h:158:0: warning: "pgm_read_word" redefined [enabled by default]

   #define pgm_read_word(p) (*(p))

 ^

In file included from C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:31:0,

                 from C:\Users\Pierre\AppData\Local\Temp\arduino_build_234123\sketch\Mouchard_nRF24_1_1.ino.cpp:1:

C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:103:0: note: this is the location of the previous definition

 #define pgm_read_word(addr) (*(const unsigned short *)(addr))

 ^

Avamander, 2017-09-14:
Okay, this is a warning as you already know, redefining is done to avoid using unimplemented/broken functionality, if you can provide the preprocessor flag that is defined when someone compiles for your platform and the necessary header file name that contains these functions/macros I can fix it, until that you have to just ignore the warning.

pierre83, 2017-09-20:
Hello

Thank you. I'm not able to provide any information as I only compile on my Windows Vista with IDE 1.8.2 (last nRF24 library version and last Arduino boards version) and I have no idea about the rest as I don't see anything after clicking on the compile button, excuse me.

Moreover, my DUE runs very fine despite all warnings, and I don't want to provide any problem for a board wich is said in end of life (an error for me, but...). I'll buy another one because the price of 10 euros is very very low and the power of the CPU is enough for me.

Thank you again.

Pierre
  ----- Original Message ----- 
  From: Avamander 
  To: nRF24/RF24 
  Cc: pierre83 ; Author 
  Sent: Thursday, September 14, 2017 2:46 PM
  Subject: Re: [nRF24/RF24] Compilation Warning with DUE. (RF24 1.3.0 & IDE 1.8.2) (#373)

  Okay, this is a warning as you already know, redefining is done to avoid using unimplemented/broken functionality, if you can provide the preprocessor flag that is defined when someone compiles for your platform and the necessary header file name that contains these functions/macros I can fix it, until that you have to just ignore the warning.

pierre83, 2017-09-21:
Hi, 
I copied all the messages seen during compilation, sorry.

E:\arduino-1.8.2\arduino-builder -dump-prefs -logger=machine -hardware E:\arduino-1.8.2\hardware -hardware C:\Users\Pierre\AppData\Local\Arduino15\packages -hardware E:\Pierre\Documents Pierre\Arduino\hardware -tools E:\arduino-1.8.2\tools-builder -tools E:\arduino-1.8.2\hardware\tools\avr -tools C:\Users\Pierre\AppData\Local\Arduino15\packages -built-in-libraries E:\arduino-1.8.2\libraries -libraries E:\Pierre\Documents Pierre\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10802 -build-path C:\Users\Pierre\AppData\Local\Temp\arduino_build_700675 -warnings=all -build-cache C:\Users\Pierre\AppData\Local\Temp\arduino_cache_787800 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose E:\Pierre\Documents Pierre\Arduino\Meteo\Meteo_4.5r\Meteo_4.5r.ino
E:\arduino-1.8.2\arduino-builder -compile -logger=machine -hardware E:\arduino-1.8.2\hardware -hardware C:\Users\Pierre\AppData\Local\Arduino15\packages -hardware E:\Pierre\Documents Pierre\Arduino\hardware -tools E:\arduino-1.8.2\tools-builder -tools E:\arduino-1.8.2\hardware\tools\avr -tools C:\Users\Pierre\AppData\Local\Arduino15\packages -built-in-libraries E:\arduino-1.8.2\libraries -libraries E:\Pierre\Documents Pierre\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10802 -build-path C:\Users\Pierre\AppData\Local\Temp\arduino_build_700675 -warnings=all -build-cache C:\Users\Pierre\AppData\Local\Temp\arduino_cache_787800 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose E:\Pierre\Documents Pierre\Arduino\Meteo\Meteo_4.5r\Meteo_4.5r.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11
Using core 'arduino' from platform in folder: C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11
Les options de compilation ont été modifiées, tout sera recompilé
Detecting libraries used...
"C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "C:\Users\Pierre\AppData\Local\Temp\arduino_build_700675\sketch\Meteo_4.5r.ino.cpp" -o "nul"
"C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "-IE:\Pierre\Documents Pierre\Arduino\libraries\Pierre_ILI9488_v2d" "C:\Users\Pierre\AppData\Local\Temp\arduino_build_700675\sketch\Meteo_4.5r.ino.cpp" -o "nul"
"C:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\ [... cut ...] .11\system/CMSIS/CMSIS/Include/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\Pierre\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "-IE:\Pierre\Documents Pierre\Arduino\libraries\Pierre_ILI9488_v2d" "-IE:\Pierre\Documents Pierre\Arduino\libraries\Pierre_Si7021_v4b" "-IC:\Users\Pierre\AppDat

closed 2019-05-10 by TMRh20