nRF24/RF24 #378: Random 0 values sent to reveiver side

mohamedalichelbi, 2017-08-22:
hello , im using two arduino unos to control 2 servos and one brushless motor , i have displayed values on serial monitor from transmitting arduino and they are perfect but when i display them from receiver arduino there are random 0 values appearing between correct values which ruins all the work
                                 transmitter code`
/*
* Arduino Wireless Communication Tutorial
*     Example 2 - Transmitter Code
*                
* by Dejan Nedelkovski, www.HowToMechatronics.com
* 
* Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
*/
int value = 0; // set values you need to zero
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#define led 12
RF24 radio(7, 8); // CNS, CE
const byte addresses[][6] = {"00001", "00002"};
boolean buttonState = 0;
    #define MAX_SIGNAL 2100
    #define MIN_SIGNAL 700
void setup() {
  pinMode(12, OUTPUT);
  radio.begin();
  radio.openWritingPipe(addresses[1]); // 00001
  radio.openReadingPipe(1, addresses[0]); // 00002
  radio.setPALevel(RF24_PA_MIN);
  Serial.begin(115200); 
}
void loop() {
  delay(5);
  radio.stopListening();
  int potValue = analogRead(A0);
  int angleValue = map(potValue, 0, 1023, 0, 180);
  radio.write(&angleValue, sizeof(angleValue));
      int potValue1 = analogRead(A1);
      int angleValue1 = map(potValue1, 0, 1023, 0, 180);
      radio.write(&angleValue1, sizeof(angleValue1));
          int potValue2 = analogRead(A2);
          int angleValue2 = map(potValue2, 0, 1023, MIN_SIGNAL, MAX_SIGNAL);
          radio.write(&angleValue2, sizeof(angleValue2));
           if(Serial.available()) 
           value = Serial.parseInt();    // Parse an Integer from Serial
           Serial.println(angleValue2);
}
`
                         receiver code `
/*
* Arduino Wireless Communication Tutorial
*     Example 2 - Receiver Code
*                
* by Dejan Nedelkovski, www.HowToMechatronics.com
* 
* Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
*/
int value = 0; // set values you need to zero
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <Servo.h>
#define button 4
RF24 radio(7, 8); // CNS, CE
const byte addresses[][6] = {"00001", "00002"};
Servo myServo;
Servo myServo1;
Servo firstESC;
boolean buttonState = 0;
void setup() {
  pinMode(button, INPUT);
  myServo.attach(5);
  myServo1.attach(6);
  firstESC.attach(9);
  radio.begin();
  radio.openWritingPipe(addresses[0]); // 00002
  radio.openReadingPipe(1, addresses[1]); // 00001
  radio.setPALevel(RF24_PA_MIN);
  Serial.begin(115200);
}
void loop() {
  delay(5);
  radio.startListening();
  if ( radio.available()) {
    while (radio.available()) {
      int angleV = 0;
      int angleVTRUE = 0;
      radio.read(&angleV, sizeof(angleV));
      if (angleV < 180 )
             {
             angleVTRUE = angleV;
             }
      myServo.write(angleVTRUE);
         int angleV1 = 0;
         int angleV1TRUE = 0;
         radio.read(&angleV1, sizeof(angleV1));
         if (angleV1 < 180 )
              {
              angleV1TRUE = angleV1;
              }
         myServo1.write(angleV1TRUE);
             int angleV2 = 0;
             int angleV2TRUE = 0;
             radio.read(&angleV2, sizeof(angleV2));
             bitSet(angleV2, 0);
             if (angleV2 < 2100 &&   angleV2 > 700)
             {
             angleV2TRUE = angleV2;
             }
              angleV2TRUE = constrain(angleV2TRUE,700 , 2100);
                   
             firstESC.writeMicroseconds(angleV2TRUE);
               if(Serial.available()) 
               value = Serial.parseInt(); 
               Serial.println(angleV2TRUE);
    }
  }
}
`
PS: even after constrain() 0 values become 700 which is also not good
try displaying angleVTRUE and angleV1TRUE same problem
any help will be very appreciated and thank you in advance

Avamander, 2017-08-22:
Check your wiring and make sure you give a really good power supply to the nRF, it doesn't tolerate voltage spikes and brownouts.

mohamedalichelbi, 2017-08-22:
im really grateful for the response ,wiring is correct and power is pretty constant. i will also mention that ,if that helps,  the number of values received is more than number of values transmitted  which means if number 50 is sent 4 times i will get on the receiver side 5 or 6 values ( 4 times number 50 but with one or two random numbers between them). is there any way to ignore those values ?

Avamander, 2017-08-22:
Well, no way to ignore if you don't want to ignore "0". Again, make sure your wiring is good, no cheap ribbon cable and decoupling capacitors on your modules. It's pretty much the only possible cause there is.

mohamedalichelbi, 2017-08-22:
for the brushless motor my range is (700 - 2100) so i would be pleased to know a way to ignore values under 700 and i tried it with this  
` if (angleV2 < 2100 &&   angleV2 > 700)
             {
             angleV2TRUE = angleV2;
             }
              angleV2TRUE = constrain(angleV2TRUE,700 , 2100);`
 it did great with all values except values like (0-1-2) persisted and appeared on angleV 2TRUE before constrain

Avamander, 2017-08-22:
Then there's some other logic flaw in your code if you're saying `if(angleV2 > 700)` and angleV2TRUE is still assigned values 2 <= x <= 0.

mohamedalichelbi, 2017-08-22:
sorry i couldn't figure it out , is there any other way to do it ?

wmarkow, 2017-08-23:
My suggestions:

Test if motors and servos have impact to random data. Detach motor and servos from the Unos. Just print the data to Serial on both Arduinos (what is sent and what is received) to see if you still have random zero values.

In your transmitter code you sent three values  **angleValue**, **angleValue1**, **angleValue2** but you print only **angleValue2** to serial, so you do not know what are the values of other variables. The same is in your receiver code.

Maybe **map** or **constrain** work incorrect?

You are sending three integers in a three separate packets, constantly: AV, AV1, AV2, AV, AV1, AV2,.... You receive those values as well in three separate packets: AV, AV1, AV2. What if the first AV1 packet will not be delivered (because of transmission error): AV, AV2, AV, AV1, AV2? Your project will stop working because your code will silently assign AV2 as a AV1 and your hardware will be out of the sync. I suggest to create a structure that will hold all of three values together and send them all together in one packet.

The code below looks odd. Is it used for something?
```
if(Serial.available()) 
value = Serial.parseInt();    // Parse an Integer from Serial
```

mohamedalichelbi, 2017-08-23:
UPDATE : tried sending only one variable (AV) and it worked perfectly , i guess that i should send all my 3 variables in one packet . anyone knows a way to do it ?

Avamander, 2017-08-23:
See the GettingStarted_HandlingData example. The struct dataStruct (on line 31) contains two variables and then is later sent with `radio.write(&myData, sizeof(myData))` on line 77.

mohamedalichelbi, 2017-08-23:
so i tried to imitate that structure but now receiver in receiving nothing ;
Transmitter code 
`
/*
* Arduino Wireless Communication Tutorial
*     Example 2 - Transmitter Code
*                
* by Dejan Nedelkovski, www.HowToMechatronics.com
* 
* Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
*/
struct dataStruct{
  int angleValue;
  int angleValue1;
  int angleValue2;
}myData;

int value = 0; // set values you need to zero
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#define led 12
RF24 radio(7, 8); // CNS, CE
const byte addresses[][6] = {"00001", "00002"};
boolean buttonState = 0;
    #define MAX_SIGNAL 2100
    #define MIN_SIGNAL 700
void setup() {
  pinMode(12, OUTPUT);
  radio.begin();
  radio.openWritingPipe(addresses[1]); // 00001
  radio.openReadingPipe(1, addresses[0]); // 00002
  radio.setPALevel(RF24_PA_MIN);
  Serial.begin(115200); 
}
void loop() {
  delay(5);
  radio.stopListening();
  int potValue = analogRead(A0);
  int angleValue = map(potValue, 0, 1023, 0, 180);
  
      int potValue1 = analogRead(A1);
      int angleValue1 = map(potValue1, 0, 1023, 0, 180);
      
          int potValue2 = analogRead(A2);
          int angleValue2 = map(potValue2, 0, 1023, MIN_SIGNAL, MAX_SIGNAL);
                 radio.write(&myData, sizeof(myData));

                 
           if(Serial.available()) 
           value = Serial.parseInt();    // Parse an Integer from Serial
           Serial.println(angleValue);
           Serial.println(angleValue1);
           Serial.println(angleValue2);
}
`
receiver code is kept the same

mohamedalichelbi, 2017-08-23:
i also made this new receiver code but it is not receiving anything also
`
/*
* Arduino Wireless Communication Tutorial
*     Example 2 - Receiver Code
*                
* by Dejan Nedelkovski, www.HowToMechatronics.com
* 
* Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
*/
struct dataStruct{
  int angleV;
  int angleV1;
  int angleV2;
}myData1;

int value = 0; // set values you need to zero
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <Servo.h>
#define button 4
RF24 radio(7, 8); // CNS, CE
const byte addresses[][6] = {"00001", "00002"};
Servo myServo;
Servo myServo1;
Servo firstESC;
boolean buttonState = 0;
void setup() {
  pinMode(button, INPUT);
  myServo.attach(5);
  myServo1.attach(6);
  firstESC.attach(3);
  radio.begin();
  radio.openWritingPipe(addresses[0]); // 00002
  radio.openReadingPipe(1, addresses[1]); // 00001
  radio.setPALevel(RF24_PA_MIN);
  Serial.begin(115200);
}
void loop() {
  delay(5);
  radio.startListening();
  if ( radio.available()) {
    while (radio.available()) {
      radio.read(&myData1, sizeof(myData1));
      int angleV = 0;
      int angleVTRUE = 0;
     
      if (angleV < 180 )
             {
             angleVTRUE = angleV;
             }
      myServo.write(angleVTRUE);
         int angleV1 = 0;
         int angleV1TRUE = 0;
         
         if (angleV1 < 180 )
              {
              angleV1TRUE = angleV1;
              }
         myServo1.write(angleV1TRUE);
             int angleV2 = 0;
             int angleV2TRUE = 0;
             
             bitSet(angleV2, 0);
             if (angleV2 < 2100 &&   angleV2 > 700)
             {
             angleV2TRUE = angleV2;
             }
              angleV2TRUE = constrain(angleV2TRUE,700,2100);
                   
             firstESC.writeMicroseconds(angleV2TRUE);
               if(Serial.available()) 
               value = Serial.parseInt(); 
               Serial.println(angleV);
               Serial.println(angleV1);
               Serial.println(angleV2);
    }
  }
}
`

Avamander, 2017-08-23:
You're not storing anything in myData, you're creating new integers and assigning them values, see the example how you assign values to myData.

ortegafernando, 2017-08-23:
Hi, I will only say one thing. I am not an expert, and I have figth a lot with these chips. 

@mohamedalichelbi  the best thing that you can do, it to flash again the simple ping-pair example to check your hardware installation. After that, check your code. May be anything related with payload's size?
But first, based on my own experience, is to try again the simplest example.
Regards,

PeterSek, 2017-08-23:
Try additional code shown below

>>>TX
void loop() {
  delay(5);
  radio.stopListening();
  int potValue = analogRead(A0);
  int angleValue = map(potValue, 0, 1023, 0, 180);

  int potValue1 = analogRead(A1);
  int angleValue1 = map(potValue1, 0, 1023, 0, 180);

  int potValue2 = analogRead(A2);
  int angleValue2 = map(potValue2, 0, 1023, MIN_SIGNAL, MAX_SIGNAL);

 * //copy values to struct elements*
*  myData.angleValue = potValue;     //add these 3 lines*
*  myData.angleValue1 = potValue1;   //add these 3 lines*
*  myData.angleValue2 = potValue2;   //add these 3 lines*
  radio.write(&myData, sizeof(myData));

  if (Serial.available())

<<<

>>>RX
  if ( radio.available()) {
    while (radio.available()) {
      radio.read(&myData1, sizeof(myData1));
 *     //myData1.angleV    //these elements now contain received data*
*      //myData1.angleV1*
*      //myData1.angleV2*

      int angleV = 0;
      int angleVTRUE = 0;

   *   angleV = myData1.angleV;  //so move elemtes to local variables*
*      //angleV1= myData1.angleV1; //added*
*      //angleV1= myData1.angleV1; //added*
*      //comments, I would make send receive tyme names same myData both
txrx*
*      //also do not need angle V variables as i believe you can use
directly myData1.angleV instead of angleV etc*
*      //hope this helps, PeterSek*

      if (angleV < 180 )
      {
        angleVTRUE = angleV;
      }
      myServo.write(angleVTRUE);
      int angleV1 = 0;
      int angleV1TRUE = 0;
*      angleV1= myData1.angleV1; //added...**//also do not need angle V
variables as i believe you can use directly myData1.angleV instead of
angleV etc*

      if (angleV1 < 180 )
      {
        angleV1TRUE = angleV1;
      }
      myServo1.write(angleV1TRUE);
      int angleV2 = 0;
      int angleV2TRUE = 0;
*      angleV1= myData1.angleV1; //added...**//also do not need angle V
variables as i believe you can use directly myData1.angleV instead of
angleV etc*

      bitSet(angleV2, 0);
      if (angleV2 < 2100 &&   angleV2 > 700)
      {
        angleV2TRUE = angleV2;
      }
      angleV2TRUE = constrain(angleV2TRUE, 700, 2100);
<<<

mohamedalichelbi, 2017-08-23:
UPDATE :tried Peter's  with changing 
`
*  myData.angleValue = potValue;     //add these 3 lines*
*  myData.angleValue1 = potValue1;   //add these 3 lines*
*  myData.angleValue2 = potValue2;   //add these 3 lines*
`
to
 `
*  myData.angleValue =angleValue ;     //add these 3 lines*
*  myData.angleValue1 =angleValue1 ;   //add these 3 lines*
*  myData.angleValue2 = angleValue2;   //add these 3 lines*
`
  and  wooohooooo it's wooorking .
Thank you for caring and supporting me wholeheartedly. you guys are amazing

closed 2017-12-09 by TMRh20