nRF24/RF24 #379: Arduino to RPi communication

LLCFr, 2017-08-24:
Hello,

I’m LLCFr, a french total newbie on Github, so please excuse me for my English.
My goal is make Arduino and RPi communicate through SI24R1 (clone of NRF24L01): about 20 students have an Arduino, that will be used in classroom to answer some questions. These Arduino have a SI24R1, in order to send the answers to the teacher that has a RPi with a SI24R1. 

The difficulty is that the students can send their answer at any time, so the RPi should be able to receive many messages very quickly. Each message is about 10 Bytes long.

So the first step is to enable everything communicate.

1) Arduino MEGA <-> Arduino MEGA

This is perfect: the transmitter sends 4 bytes “ABCD”, then wait for the reply of the receiver, 4 bytes also “1234”, then sends 4 bytes…
A loop of 300 x (send + receive) takes about 1s, without any error, and with datarate = 250KBS.

The programs are based on some example found here:

Arduino Transmitter source code:
`//Test SI24R01 (Clone NRF24) - 2.4GHz SPI
//branchements:
// 50 : MISO
// 51 : MOSI
// 52 : SCK
// 7 : CE
// 8 : CSN
// 3 : IRQ
// 3.3V : VDD
// GND : GND

/*
* Getting Started example sketch for nRF24L01+ radios
* This is a very basic example of how to send data from one node to another
* Updated: Dec 2014 by TMRh20
*/

#include <SPI.h>
#include "RF24.h"
#include "rgb_lcd.h"

//#define DEBUG

/*rgb_lcd lcd;
const int colorR = 0;
const int colorG = 255;
const int colorB = 0;
*/

int pushButton = 5;
int lastButtonState = 0;     // previous state of the button

unsigned short usCount = 0;
unsigned char aucMsg[4] = {'A','B','C','D'};
unsigned char aucAck[4] = {'0','1','2','3'};
unsigned char aucEnd[4] = {'F','I','N','I'};
unsigned long T0;
unsigned long T1;
int EmLoop;
int EmSendOK;
int EmReceptOK;
int EmReceptKO;
int EmReceptNO;

/****************** User Config ***************************/
/***      Set this radio as radio number 0 or 1         ***/
bool radioNumber = 1;

/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
RF24 radio(7,8); // (pin CE, pin CSN)
/**********************************************************/

byte addresses[][6] = {"1Node","2Node"};
[... 365 lines cut ...]
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.03 KB/s 
> Payload Count: 1 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 0.00 KB/s 
> Payload Count: 0 

So, and after this very long post (sorry for this), do you have any clue, any idea to help me solve this problem? I am surely missing something obvious but I have to admit that I am stuck.
Thank you very much for any help, I innocently thought that RPi would be very easy to use, compared to Arduino. Seems that I’m wrong.

Avamander, 2017-08-24:
1) Clones are even pickier about the power supply, decouple it good for both LF and HF noise, don't use ribbon cables, make sure the connections are good.

2) If everything is within 5m, use higher data rates, you won't suffer any lost packets usually, you can obviously receive and process much more messages then.

3) Start from the examples, get the examples working nicely, then start modifying, please, you'll have less headache because if you know the examples don't work it's the transcievers, if they do work it's your code.

4) Arduino Mega has different pins for SPI than Uno etc. so make sure that is correct (that you aren't seeing SPI noise here).

5) Clones suck at dynamic payloads and auto-ack, maybe even CRC, disable all of them.

wmarkow, 2017-08-25:
> Arduino MEGA <-> Arduino MEGA
This is perfect: the transmitter sends 4 bytes “ABCD”, then wait for the reply of the receiver, 4 bytes also “1234”, then sends 4 bytes…
A loop of 300 x (send + receive) takes about 1s, without any error, and with datarate = 250KBS.

If so, then it means that on Arduino <-> Arduino system you have reached the maximum capacity of transmitted data over the channel (with 250kbps datarate). You send around 600 packets, which gives 1000ms / 600 = 1.7ms per packet. Is RPi able to operate at such a data rate? I know that RPi is more powerful than Arduino but it runs the Linux system and does a lot of other things (while Arduino executes only your program and nothing more). Maybe your data rate is to high for RPi and RPi is not able to query the RF24 chip every 1.7ms? Then the RX FIFO gets full and some packets are dropped. I don't know; I'm just throwing some ideas.
I would slow down the data transmission rate. I would start from one packet per second and observe in RPi if the packets are dropped. Then you can increase your data rate: like 10 packets per second, 50 packets per second and so on until you find the limit where RPi starts to drop packets. It should not be so hard to implement: you just need to rework your transmitter code.

LLCFr, 2017-08-25:
Thank you for your help !

Following your advice, I made the following tests with the "transfer" demo code":
- Decouple power supply : same (bad) result
- disable dynamicpayload, autoack and CRC : it works much better, thank you so much !

RPi's output is now:
> Rate: 10.88 KB/s 
> Payload Count: 340 
> Rate: 30.59 KB/s 
> Payload Count: 956 
> Rate: 28.77 KB/s 
> Payload Count: 899 
> Rate: 19.55 KB/s 
> Payload Count: 611 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 8.99 KB/s 
> Payload Count: 281 
> Rate: 30.08 KB/s 
> Payload Count: 940 
> Rate: 30.59 KB/s 
> Payload Count: 956 
> Rate: 20.74 KB/s 
> Payload Count: 648 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 8.03 KB/s 
> Payload Count: 251 
> Rate: 30.02 KB/s 
> Payload Count: 938 
> Rate: 30.75 KB/s 
> Payload Count: 961 
> Rate: 22.53 KB/s 
> Payload Count: 704 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 6.62 KB/s 
> Payload Count: 207 
> Rate: 30.85 KB/s 
> Payload Count: 964 
> Rate: 29.09 KB/s 
> Payload Count: 909 
> Rate: 24.00 KB/s 
> Payload Count: 750 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 5.41 KB/s 
> Payload Count: 169 
> Rate: 30.34 KB/s 
> Payload Count: 948 
> Rate: 31.14 KB/s 
> Payload Count: 973 
> Rate: 27.33 KB/s 
> Payload Count: 854 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 4.10 KB/s 
> Payload Count: 128 
> Rate: 31.26 KB/s 
> Payload Count: 977 
> Rate: 29.86 KB/s 
> Payload Count: 933 
> Rate: 26.62 KB/s 
> Payload Count: 832 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 2.24 KB/s 
> Payload Count: 70 
> Rate: 30.82 KB/s 
> Payload Count: 963 
> Rate: 30.05 KB/s 
> Payload Count: 939 
> Rate: 28.70 KB/s 
> Payload Count: 897 
> Rate: 0.00 KB/s 
> Payload Count: 0 
> Rate: 1.09 KB/s 
> Payload Count: 34 
> Rate: 30.02 KB/s 
> Payload Count: 938 
> Rate: 28.99 KB/s 
> Payload Count: 906 

This is much better, but the strange parts are:
- The payload count drops down to 0 every 5s. I made the test several times, and it is always every 5s.
- After a various time (20 seconds, several minutes, almost one hour, depending on the test), the payload count stay at 0, as if the RPi does not receive anything anymore.

So my next steps are:
- finding which parameter is the problem (dynamicpayload, CRC,...),
- check if payload received = payload sent
- find the reason why the payload count drops to 0 every 5s,
- find the reason why the receiver (Rpi) seems to fall into a coma and does not receive anything after a various time.

Thank you again for your help !

Avamander, 2017-09-14:
I heard someone once say "add debug prints until your software works" which applies here too, modify the library to basically print every single thing it does and then look at the last line printed, maybe it's a SPI call frozen, maybe it's a voltage spike.

LLCFr, 2017-09-14:
Hi,

I thought that this subject would not interest anyone anymore since I'm using "counterfeits" modules...

Latest news: when I connect the IRQ pin to GND, everything work perfectly (almost). Until then, I let IRQ pin non-connected. I don't know why I tried it but, hey, since it works :-)
From your experience, do you also connect IRQ to GND when using"true" nRF24 ?

So now, my next step: I have to imagine a protocol in order for the 20 Arduinos to communicate (two ways) with the Rpi without loss.
Allowing the Arduinos to take the initiative of the communication is perhaps not the best idea, because it may imply some collisions (I guess): some students may send data at the exact same time (unlikely case, but toasted bread always lands buttered-side-down). On the other hand, having the RPi initiate the communication implies that the Arduinos are always listening, and it costs battery power( about 20mA), and the RPi should often "ping" the Arduinos in case they have something to say.
Too many questions, too few answers ;-)

closed 2019-05-10 by TMRh20