nRF24/RF24 #383: ImportError: No module named RF24

mmittigg, 2017-10-09:
Ich habe die Datei led_test_24ghz.py im Verzeichnis ~/rf24libs/RF24/examples gespeichert. Wenn ich sie aber über sudo python .... ausführe, bekomme ich folgende Fehlermeldung:
"File "led_test_24ghz.py", line 4 in <module>
from RF24 import *
ImportError: No module named RF24"
Wo könnte der Fehler liegen?

mwwalewski, 2017-10-29:
Hello, I've got same problem when trying to use  RF24 with python3 

i@raspberrypi:~/rf24libs/RF24/pyRF24 $ sudo python3 setup.py build
running build
running build_ext
building 'RF24' extension
creating build/temp.linux-armv7l-3.5
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c pyRF24.cpp -o build/temp.linux-armv7l-3.5/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
creating build/lib.linux-armv7l-3.5
arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/pyRF24.o -lrf24 -lboost_python3 -o build/lib.linux-armv7l-3.5/RF24.cpython-35m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lboost_python3
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-g++' failed with exit status 1

Any idea what to do with: /usr/bin/ld: cannot find -lboost_python3   ?

thalesmaoa, 2017-11-30:
For Python3 you need to create a symbolic link
http://tmrh20.github.io/RF24/Python.html

```
sudo ln -s libboost_python-py35.so.1.62.0 libboost_python3.so
```

My case!

closed 2019-05-10 by TMRh20