nRF24/RF24 #386: Addresses convertion issue

deckbsd, 2017-11-01:
Hi,

I try to communicate with my arduino to my pi. On my arduino i'm using the nrf24 tmrh20 library. I declare my addresses like this
byte addresses[][6] = {"1Node","2Node"};

On my pi, i can't so i would like to know value in hex that i have to put in pipes = [[0xe7, 0xe7, 0xe7, 0xe7, 0xe7], [0xe7, 0xe7, 0xe7, 0xe7, 0xe7]] to get the same address ?
any idea ?
Thanks by advance

PaoloElMarinelo, 2017-11-02:
Hi,

In my Pi in python I use that : `pipes = [0xF0F0F0F0E1, 0xF0F0F0F0D2]` 
And in my Arduino Nano : `const uint64_t addresses[2] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0D2LL };`

I hope it will help you

deckbsd, 2017-11-04:
Thanks for your help 👍  it's now resolved

closed 2017-11-04 by deckbsd