nRF24/RF24 #389: Problems using library on RPI (AVR-GCC)

h4lifax, 2017-12-03:
Hello,

i recently downloaded the library and performed the steps to install it. I'm trying to use the library with avr-gcc to send and receive data with an attiny85. An raspberry pi is used for programming the attiny. 

The following error pops up, when including the library by #include "../../RF24/RF24.h":
[image]

Does someone have an idea to fix this error? I've already tried to add the features.h by appending -I /path/to/missingfile but this leads to further errors, where files weren't found.

Avamander, 2017-12-03:
Set up your build system properly. It clearly can't find any of the built-in libraries.

targodan, 2019-01-08:
I would like to know how to setup your build system "properly".

Does this library need the Arduino library-stack in order to work even on plain ATTinys?

TMRh20, 2019-01-08:
"Does this library need the Arduino library-stack in order to work even on
plain ATTinys?"

Nope, per https://tmrh20.github.io/RF24/Portability.html  the library needs
mainly, SPI(built into library for ATTiny) , GPIO ( digtalWrite() ) and
timer functionality (millis(), delay()),  which is provided by the Arduino
layer, or can be provided by the user separately.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

targodan, 2019-01-14:
Ok, that's interesting. I can't get the library to build properly on a bare ATTiny 861. I even passed `-D__AVR_ATtiny861__` to my compiler. It keeps trying to include `Arduino.h`.

I feel like the macros aren't setup properly for building without Arduino. Or am I missing any defines I should add? I can't find any documentation on the build defines necessary.

targodan, 2019-01-18:
Ok, after sifting though tons of preprocessor statements I think I got the gist of it. These are my findings using an ATTiny861 and thus giving the compiler the `-D__AVR_ATtiny861__` flag. Similar things may apply with other tinys.

1. Whatever you do it wants to include `<Arduino.h>` in [RF24_config.h:63](https://github.com/nRF24/RF24/blob/a3ab7ddb075360deac54fe2c8e2fab286793d21c/RF24_config.h#L63) which I don't understand [RF24_config.h:48](https://github.com/nRF24/RF24/blob/a3ab7ddb075360deac54fe2c8e2fab286793d21c/RF24_config.h#L48) is also executed including the ATTiny utility verifying that __AVR_ATtiny861__ is in fact defined correctly. Maybe a bug with my avr-gcc? Don't really know. Anyway [RF24_config.h:63](https://github.com/nRF24/RF24/blob/a3ab7ddb075360deac54fe2c8e2fab286793d21c/RF24_config.h#L63) needed to go in order for it to compile for me.
2. The ATTiny utility only provides SPI, but neither the digital IO interface or the timing interface of arduino. I find this a little strange as (at least in case of the 861) I need to use a third party arduino core anyway, which also brings an SPI implementation. This leads to compile errors due to SPIClass redefinition problems. I did not investigate this further because I decided I didn't want to use the arduino stack and instead supply my own drivers.
3. I then proceeded to (ab-)use the ATTiny utility folder to put my own drivers there. I did this because I don't think there is a compile/preprocessor flag that let's you easily include your own stuff and I had sussed out the include procedure for said utility folder.
4. I wrote a quick SPI driver, implemented the digital IO interface and setup a timer1 based dynamic delay interface as Arduino usually provides them and as the RF24 lib needs. If you do this keep in mind that in the current setup the supplied SPIClass functions are implemented within [RF24.cpp](https://github.com/nRF24/RF24/blob/a3ab7ddb075360deac54fe2c8e2fab286793d21c/RF24.cpp#L1625) so they need to be deleted or replaced as well if you want to use your own implementation.

At that stage I finally had it compiling but I ran into really strange problems with the `RF24::begin` function hanging in the while-loop for transfering data via SPI. But only sometimes. Different optimization flags and slight code variations sometimes seemed to fix this problem. However I never got successful transmission of packets via the RF24 chip. I have verified that my drivers work, I have verified that the chip works and I have verified that the receiving side works. I now suspect something like a stack overflow or out of RAM problem as this behavior is really strange and completely unpredictable. Maybe too many functions nested.

Anyway at this stage I give up on trying to avoid the Arduino stack and will just go the full Arduino  route even using the dreaded ArduinoIDE (yikes). I have already verified that the RF24 lib works flawlessly with the full Arduino stack and the [ATTiny Core by SpenceKode](https://github.com/SpenceKonde/ATTinyCore).

Maybe this helps someone.

A last piece of criticism to this library: IMO it lacks in documentation regarding the flag dependent build system, or said documentation is not as easy to find as it should be (I couldn't find it). Admittedly for your everyday ArduinoIDE users this is completely irrelevant but for anyone who tries to avoid the Arduino library stack or anyone who is just interested in how it works this is worth the characters in gold. I am for example still a little confused about which utility is used in what scenario.

akatran, 2019-01-18:
@targodan I have the same frustration for the ArduinoIDE so I may suggest to use [Sloeber](http://eclipse.baeyens.it/) which is doing much better as it is Eclipse IDE under the hood!

I use it for a couple of years now and I can say that it is way closer to what we are accustomed to than ArduinoIDE (If you can call it an IDE).

targodan, 2019-01-19:
Thank you very much @akatran I didn't even know there are any IDEs that have a prepared Arduino toolchain and build system. Sloeber looks really nice, you have saved my sanity friend! 👍

closed 2019-05-10 by TMRh20