nRF24/RF24 #390: How to get the correct data using esp-12f+nRF24L01?

wind-bsk, 2017-12-04:
Hello! I can not get the correct data receiver esp-12F + NRF24L01 from the transmitter Arduino pro mini + NRF24L01. I send and get data in this structure:
```
struct dataStruct {	
	unsigned int counter;
	byte id[8];
	int data;
	}data;
```
Example. 
Data sent by the transmitter Arduino pro mini + nRF24L01: 
```
counter=1;
id=28EE644A7161B3; 
data=2012
```

Data received by the receive esp-12F + nRF24L01:
```
counter=3995598849;
id=644A7161B3DC7; 
data=0
```

How to get the correct data by the receiver esp-12F + nRF24L01?

wmarkow, 2017-12-04:
It could be that your **dataStruct** size is different on Arduino than on esp-12F. I'm refereing to the **int** and **unsigned int** types. You can check that in the source code by using **sizeoff** keyword. If this is the issue, I suggest to use the types like **int_8t** or **int_16t** or **int_32t** (and so on, also the **uint_8t**, etc.) instead of common **int** or **unsigned int**.

wind-bsk, 2017-12-04:
**wmarkow**, if the receiver has the variable "counter" to change the type c **int** to **uint8_t**, then the data "counter" gets correct. For "id" and "data" the method does not work

wmarkow, 2017-12-04:
This is for sure: the structure on the transmitter's code should be exactly the same as defined in the receiver's code. Please check that and post the result. We will see later, what to do :)

wind-bsk, 2017-12-05:
I checked the size of the **int** data type. Indeed, for arduino it is 2 bytes for esp-12f 4 bytes. I replaced, in the data structure type int on **short unsigned int** and **short int**. 
```
struct dataStruct {	
	short unsigned int counter;
	byte id[8];
	short int data;
	}data;
```
As a result, I received the correct data by the receiver. 
**wmarkow**, thank you for helping me understand the problem. Good luck!

closed 2017-12-05 by wind-bsk

Avamander, 2017-12-05:
@wind-bsk Try using the standard `uint8_t`, `uint16_t` and `uint32_t` and their signed equivalents `int8_t`, `int16_t` and `int32_t`. That way you can be really sure what you're getting is unsigned 8 bits, unsigned 16 bits and so on.