nRF24/RF24 #391: Mixed tabs and spaces

myTonino, 2017-12-06:
Most source files contain a mix of tabs and spaces that renders in editors tab width different than 4 strangely. This is also the case on viewing the files online here on Github and applies also to the mesh and networking lib.

Avamander, 2017-12-19:
If you have the time to reformat all files according to NASA's C style guide then you could submit a pull request and I'll merge it.

wmarkow, 2017-12-23:
@Avamander, do you know some kind of source formatter (beautifier) software that would be able to format the code according to NASA C style? I have searched the Internet but I wasn't able to find such a software. I could find the only the docs related to the style:
 * [NASA C Style](http://homepages.inf.ed.ac.uk/dts/pm/Papers/nasa-c-style.pdf)
 * [NASA C++ Style](https://ntrs.nasa.gov/search.jsp?R=20080039927)

During my standard Arduino development I'm using [Eclipse CDT](https://www.eclipse.org/cdt/) with [Sloeber Plugin](http://eclipse.baeyens.it/). Eclipse CDT is able itself to format the code, by default only according to the following rules:
 * K&R
 * BSD/Allman
 * GNU
 * Whitesmiths

@Avamander, could you please write shortly why you prefer **NASA C Style**? Or what are pros for using **NASA**? I know that the code style treats about the "visual view of the code" (like the number of spaces in the indentation, the braces, where to put a new line and so on) but also it gives some clues about how to name correctly the functions, variables, and so on. 
NRF24 uses C++ so we would need some style guide related to C++ as well. Above I have put two link, the second one refers to NASA C++, I just found it in the Internet but I'm not quite sure if this is still valid. What are your thoughts?

Avamander, 2017-12-23:
@wmarkow It's both well written down, used in mission-critical places and it most closely resembles what people write (and what's used currently). 

If you have better suggestions, feel free to add them, and yeah, C++ style guide probably suits us more (and you probably can switch your IDE to Allman style as it's really close).

MAKOMO, 2017-12-23:
I am not even a C programmer, but I could assist in just replacing tabs by 4 spaces in all .h and .c files. That would increase readability on Github via a Web browser as well as editors that render tabs with a number of spaces different from 4. Let me know.

wmarkow, 2017-12-28:
Thanks @MAKOMO but we can make the process automatic. I'm using **Eclipse CDT** for Arduino development. Eclipse has embedded code formatter. It is possible to create your own formatting rules, so I have created a formatter according to [this NASA C++ guide](https://ntrs.nasa.gov/search.jsp?R=20080039927).

I have formatted two files with this formatter. There are commited in my RF24 fork in my [fix_for_#391](https://github.com/wmarkow/RF24/tree/fix_for_%23391) branch.
I think there is something still missing in this formatter (like I have noticed that preprocessor commands sometimes are not formatted correctly). Most basic thing: I have put 3 spaces (NASA recommends strongly that) in the indentation. But we can use 4 spaces as well - I think.

@Avamander, could you please take a look at my branch and make a short review? Thank you for your comment related to coding styles. I took a look and indeed you were right; Allman style is quite close to NASA.  When I started to learn C/C++ many years ago I was taught some good coding styles but I didn't know that it was according to NASA. I like NASA coding style :)

One more thing: if we decide to use this code formatter then it is better to take a look at the pull requests first and merge them before we format the code (we will avoid many merge conflicts that will be an effect of code formatting).

MAKOMO, 2017-12-28:
@wmarkow: please go forward with this. 3 or 4 spaces are ok for me, just tabs should be avoided. I also agree that it might be easier to process pull requests first. The same work should also be applied to the Networking and Mesh lib.

Avamander, 2017-12-30:
@wmarkow Your branch looks okay. Though the Arduino examples should be formatted with the Arduino IDE, should just be kept in mind when that bridge is to be crossed.

Pull requests can usually be updated by other people, if not opening a new one with the good style (without conflicts) would be totally okay imho.

wmarkow, 2018-01-02:
I took a look at Arduino IDE 1.8.5 and how its formatting works. To auto format the code in Arduino IDE just hit **Ctrl + T** (or choose **Tools->Auto Format**). This autoformat works strange. It looks like it doesn't care about "brackets". This piece of code is formatted "correctly" with **Arduino IDE**:

```
void setup()
{
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

  int  a = 5;
  if (a == 5)
  {
    a = 4;
  } else if (a == 3) {
    a = 2;
  }
}
```

I noticed that mostly it takes care about **2 spaces indentation** and it put some spaces where needed.

Now I'm just thinking that the **Built-In Examples** in **Arduino IDE** are the clue on what is their favorite code style (formatting).

Avamander, 2018-01-02:
@wmarkow So you think we should format everything with the Arduino IDE?

wmarkow, 2018-01-02:
No, I think we can stick to the **NASA** format as you suggested [before](https://github.com/nRF24/RF24/issues/391#issuecomment-353722513).

>Though the Arduino examples should be formatted with the Arduino IDE, should just be kept in mind when that bridge is to be crossed.

I just wanted to check what is the default formatting style in Arduino IDE but it looks like it allows some mixed formatting styles and actually you need manually take care of putting the brackets in a proper format.

per1234, 2018-02-03:
I think wmakow has the right idea with the example sketches: Follow the Arduino IDE's Tools > Auto Format conventions and then use the official Built-in Examples (and code from the Language Reference pages or other official code where that doesn't suffice) as a reference for any formatting choices not set by Auto Format. This is the style that Arduino users are comfortable with seeing in sketches.

The Arduino IDE uses AStyle for the Auto Format. The stock configuration is here:
https://github.com/arduino/Arduino/blob/master/build/shared/lib/formatter.conf

AStyle also allows you to configure more formatting options, as documented here:
http://astyle.sourceforge.net/astyle.html

I believe that adding the following options to the Arduino IDE's current AStyle configuration file will fully enforce the Arduino formatting style:
```
style=attach
break-closing-braces
attach-closing-while
attach-extern-c
add-braces
align-pointer=name
convert-tabs
```
Note that some of these AStyle configuration options require a newer version of AStyle than is used by the Arduino IDE. The version of AStyle used by the Arduino IDE does not permit formatting to Arduino style braces because `break-closing-braces` and `attach-closing-while` (which are not supported by that version) are required in addition to `style=attach` (which is supported).

Even though it's in the stock Auto Format configuration, I would suggest removing the `keep-one-line-statements` option because I think that just leads to code that's more difficult for beginners to read and more prone to bugs. I'm not aware of that coding style being used in any official Arduino code. And one statement per line certainly doesn't violate the Arduino formatting style.

wmarkow, 2018-03-05:
@per1234, thank you for your input about AStyle. I did some quick test on how Arduino IDE formats the code and I have found this a bit strange; take a look on [one of my comments before](https://github.com/nRF24/RF24/issues/391#issuecomment-354870883). I have noticed like Arduino IDE doesn't force a specific **brackets formatting**. I had the feeling that this:
```
if (a == 5)
{
  a = 4;
}
```
is correctly formatted, and also this is correctly formatted:
```
} else if (a == 3) {
  a = 2;
}
```
Maybe it is just a matter of AStyle configuration file (just like you pointed)? I'm not so familiar with AStyle. It could be worth to check.

Because the issue I have noted above and because I'm also a Java programmer (and I'm using Eclipse), I started to use Eclipse formatter, because that was (for me) the fastest way to start experimenting with formatting. But enyway, I think I will take a look again at AStyle formatter.

I think the best would be to have:
 * automated process of formatting (like executing some script that will format all of the sources)
 * two different formatting styles: **NASA formatter** for RF24 library and **Arduino formatter** for the code of examples.

Going back to my exepriments with Eclipse formatter.
In my https://github.com/wmarkow/Arduino/commits/eclipse-formatter-poc branch I have just batch formatted the Arduino's examples with Eclipse formatter. The fromtter XML has been prepared by me by modifying the formatting rules in Eclipse and checking how it works on the Arduino's examples (I wanted to have as small changes as possible while formatting). You can take a look. Mostly it has reformatted the comments.
In my fork of RF24 library I have did the same in my https://github.com/wmarkow/RF24/commits/fix_for_%23391 branch. Examples are formatted with **Arduino fromatter** and other source code is formatted with **NASA formatter**.  NASA fromatter rule for Eclipse have been prepared by me according to [this NASA C++ guide](https://ntrs.nasa.gov/search.jsp?R=20080039927).

Avamander, 2018-03-05:
@wmarkow The changes seem okay and really seem to remove quite a bit of uglyness, but I'm not sure about the `char *` array changes, old:
```c++
static const char * const rf24_crclength_e_str_P[] PROGMEM = {  rf24_crclength_e_str_0, rf24_crclength_e_str_1, rf24_crclength_e_str_2,};
```
vs. new:
```c++
static const char * const rf24_crclength_e_str_P[] PROGMEM = {
  rf24_crclength_e_str_0,	
  rf24_crclength_e_str_1,	
  rf24_crclength_e_str_2,	
};
```
What do you think?

per1234, 2018-03-05:
You're correct that the stock AStyle configuration doesn't enforce a brackets style. As established by the majority of official example sketches, the Arduino brackets style is definitely attached, which is why I suggested the addition of  `style=attach` to enforce this.

I had thought the closing bracket and else statement were generally put on separate lines (`break-closing-braces`) but after looking through some official examples it appears that is incorrect. Both on the same line is more common in the examples included with the IDE, although not by an overwhelming majority. Without `break-closing-braces` my suggestion of `attach-closing-while` is no longer needed.

I think the formatting script is a great idea. I would even suggest that a formatting style compliance check be done on all commits and pull requests via an online continuous integration service (e.g. Travis CI, CircleCI). AStyle is well suited for that usage because it's a very lightweight command line application but other options may be just as good.

wmarkow, 2018-03-07:
I did some further investigations with Eclipse formatter and I have found something, what I do not like. It seems that formatter behaves a bit different when I run in outside of Eclipse (by calling some formatter classes from Eclipse core jar) with comparing to how it behave when I format the code from the Eclipse.
I have put some comments to my commit. Take a look https://github.com/wmarkow/RF24/commit/af8c5d36c5329a14dabfe91cd096157ba3de24e4 for reference.

In the meantime I took a look at AStyle formatter. I did some preliminary work and I must say that I'm surprised with this tool. It is lightweight, has a lot of configuration switches and can be run from command line easily. I think I will continue with testing AStyle.
For *NASA* style we need to use ```style=allman```. I have rechecked and ```style=attach``` works perfect for *Arduino* style - just like @per1234 has mentioned before.

I will create a separate branch in my RF24 fork to test AStyle. I will try to make some "small" commits so we could see the differences in small steps.

@Avamander, related to your question. Was it in opposite way?
Old way of code:
```
static const char * const rf24_crclength_e_str_P[] PROGMEM = {
  rf24_crclength_e_str_0,	
  rf24_crclength_e_str_1,	
  rf24_crclength_e_str_2,	
};
```
vs new:
```
static const char * const rf24_crclength_e_str_P[] PROGMEM = {  rf24_crclength_e_str_0, rf24_crclength_e_str_1, rf24_crclength_e_str_2,};
```
It could be that I did it by mistake.

Avamander, 2018-03-07:
@wmarkow Yeah, mixed up the words. I think the expanded version is more readable, what do you think?

wmarkow, 2018-03-07:
Yes, I think expanded version is more readable, especially when you have a few items in the table. Formatter will break the line anyway if the line will be longer than maximal characters per line; then everything will be mixed up. With expanded version you have everything in order: opening bracket, items (one item per line), closing bracket.

I will try to prepare the AStyle config so it will put it expanded and we will see how it looks.

wmarkow, 2018-03-09:
I have just pushed a bunch of commits to my branch. I have used **AStyle-3.1** do reformat the sources. For now all the *.c, *.h and *.cpp are formatted with NASA Code Style. All *.ino and *.pde files are not formatted yet (they are waiting for a proper astyle configuration file).

2bndy5, 2020-09-19:
status on this?
I noticed the python examples don't follow PEP8 conventions as well (extremely hard to read if you're coming from python background). I could use the popular [black linter/formatter](https://github.com/psf/black) to fix them... @wmarkow objections? I personally prefer using VSCode's builtin formatter, but I haven't looked into if the configuration defaults need modification.

I realize this issue is a bit old and Github has introduced "actions" to serve as a free CI since this was opened. I have gladly made the switch to PlatformIO which has a [section in their docs about CI scripting](https://docs.platformio.org/en/latest/integration/ci/github-actions.html#). I wouldn't mind setting this up as it would be able to tell us if a PR or commit fails to build for a certain supported board(s). Free Github accounts have a limit on how many workflows can run simultaneously (I know 2 is allowed), but no limit on how many jobs are run in a single workflow. We could have a job for each board that is supported (in 1 workflow called "Build CI"). There is an [Github "action" in the marketplace](https://github.com/marketplace/actions/c-style-check) that can be used as a C++ style checker, but I haven't confirmed the configuration options would suite this repo's contribution guidelines. The C++ style checking would be a job on a separate workflow that also includes another job for python style checking.

This may sound childish, but... BADGES! Github actions come with status badges to quickly inform library prospectors that a CI workflow is passing or not. Not to mention they add a splash of color to the README.md.

per1234, 2020-09-19:
> I have gladly made the switch to PlatformIO which has a section in their docs about CI scripting.

The problem with using PlatformIO for CI of Arduino projects is that PlatformIO allows project structures that are incompatible with the official Arduino development software. So a change that breaks official Arduino compatibility can happily pass a PlatformIO CI check.

So it depends on who your target users are. If you only care about PlatformIO users, then PlatformIO is the best choice. But if you care about Arduino users, then PlatformIO is not an appropriate tool for CI.

Arduino now provides an official tool for this purpose:
https://github.com/arduino/compile-sketches

But either way you go, PlatformIO or Arduino, any CI for compiling sketches is way better than none at all.

>Free Github accounts have a limit on how many workflows can run simultaneously (I know 2 is allowed), but no limit on how many jobs are run in a single workflow.

The limit is on the number of concurrent jobs, not workflows:
https://docs.github.com/en/actions/reference/usage-limits-billing-and-administration#usage-limits
The free plan allows a maximum of 20 concurrent jobs

The reason the distinction matters is because multiple jobs in a workflow can (and do by default) run concurrently

>We could have a job for each board that is supported (in 1 workflow called "Build CI").

A convenient way to accomplish this is by using a [job matrix](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#jobsjob_idstrategymatrix)

Example:
https://github.com/107-systems/107-Arduino-UAVCAN/blob/104e8b1370e9eaba33e638b8f854a96e83175858/.github/workflows/compile-examples.yml#L25-L31

>There is an Github "action" in the marketplace that can be used as a C++ style checker

If you want to take the astyle approach I mentioned earlier in this thread, you could use this workflow:
https://github.com/arduino/arduino-examples/blob/main/.github/workflows/code-formatting-check.yml
It's configured to only cover the example sketches, but only a tiny change would be needed to make it cover the library code files as well.

>another job for python style checking.

There's a workflow for flake8 here:
https://github.com/arduino/compile-sketches/blob/main/.github/workflows/lint-python.yml
there's likely a GitHub Actions action for that, but it's so trivial to do it directly in the workflow that it didn't seem worth investigating.

>This may sound childish, but... BADGES! Github actions come with status badges to quickly inform library prospectors that a CI workflow is passing or not. Not to mention they add a splash of color to the README.md.

I've always been pretty "meh" about badges, but I'm starting to warm up to them. The reason is because I see a lot of repositories where the maintainers simply ignore the failing CI. I think having that big ugly red badge in the readme might give a little more motivation to keep it in the green.

Avamander, 2020-09-19:
@2bndy5 

> status on this?

Pretty much done. Arduino examples are formatted using its IDE, code is formatted as agreed and new PRs are fixed before merging. It should be fairly consistent.

Python PEP8 would need a bit more effort. Feel free to make a PR.

2bndy5, 2020-09-19:
@per1234 Clearly its been a while since I read the GitHub action's docs; thank you for the clarifications. TBH, I'm not a fan of black formatter; it feels too assertive, but I've started seeing a lot of libraries adopting it because its compatible with other linters/formatters. flake8 enable/disable check mechanisms require extra knowledge of what codes relate to what checks (please correct me again if I'm mistaken): `noqa: E731`. pylint on the other hand has some rather self explanatory disable/enable checks: `pylint: disable=invalid-names,missing-docstring`. This concern really comes in handy when writing entire libraries in python, but example scripts need some TLC too. I do realize that all these python formatters' configurations can be addressed in CLI args and/or config files (I'll cross that bridge when I get to it); black seems to only allow config files out of extensibility with other formatters where the above code snippets are in comments. 

>there's likely a GitHub Actions action for that, but it's so trivial to do it directly in the workflow that it didn't seem worth investigating.

agreed. In fact that was exactly what I was thinking as well. @Avamander PEP8 formatting is actually one of the easier items on my list because, well... its python. PR on that soon.

Since the library is setup to work in both PlatformIO and Arduino IDE, I was actually thinking of running actions for both, but I didn't research the Arduino side (thank you for that lead). I forgot this library already has actions running for RPi. About the PlatformIO project structure: Yes it can be quite alienating to Arduino IDE users, but it is way more flexible like building with specific versions of libraries for specific and/or multiple boards all in one project. On that note, this library's examples aren't importable as PlatformIO projects, but they [can be made so rather easily](https://docs.platformio.org/en/latest/librarymanager/config.html#id20) to compliment the Arduino IDE feature for importing examples (again please correct me if I'm wrong).

>I think having that big ugly red badge in the readme might give a little more motivation to keep it in the green.

I gotta have that green!

Avamander, 2020-09-19:
> PEP8 formatting is actually one of the easier items on my list because, well... its python. PR on that soon.

PEP8 doesn't concern only whitespace formatting, there are rules that if violated need more effort to fix.

I'd also remove any Python2 stuff when you're already improving the thing. Full Circuitpython support would also be nice, but that's an another issue :smile:...

> but I didn't research the Arduino side (thank you for that lead).

When I looked into it... say three years ago or so, Arduino building wasn't very pleasant to automate. Too fragile. Maybe with GH Actions and Arduino's own improvements it's a bit easier now.

> I gotta have that green!

I'd hold on to adding it to the top of the readme (bottom is fine) until it's certain that it works for a long period of time.

Avamander, 2020-09-19:
Oh, and @2bndy5, @wmarkow could you please turn on 2FA on your GitHub account (or let me know if you have any strong objections)?

2bndy5, 2020-09-19:
Last time I enabled 2FA, I ran into problems with git bash. If you know of a simple solution that doesn't involve custom passwords for separate applications, I'd gladly enable it again. Github still emails me when I log in on a device that hasn't previously been used for my account.

Avamander, 2020-09-19:
> If you know of a simple solution that doesn't involve custom passwords for separate applications

SSH key.

per1234, 2020-09-19:
>When I looked into it... say three years ago or so, Arduino building wasn't very pleasant to automate. Too fragile. Maybe with GH Actions and Arduino's own improvements it's a bit easier now.

The major enabler for automating installing dependencies and compiling Arduino sketches has been the creation of Arduino CLI. This is the engine under the hood of the `arduino/compile-sketches` action.

I have a lot of experience using the Arduino IDE's CLI for CI and it can really really a pain, especially in a headless application. I ended up with Bash script full of workarounds for each quirk the IDE presented. For the last eight months, I have been working extensively using Arduino CLI in the same applications and only a couple weeks ago did I run into the first thing I had to fight it a bit on to achieve my desired result.

The `arduino/compile-sketches` action makes the Arduino CLI interface more GitHub Actions workflow friendly and also adds some capabilities that are really useful for CI applications, but wouldn't really make sense for Arduino CLI itself to provide.

I don't know whether there is a lot of use of `arduino/compile-sketches` (née `arduino/actions/libraries/compile-examples`) in the community yet, but Arduino has been using it extensively for the last year. This is a tool created because the Arduino firmware developers needed it. I also hope it will be of use to the Arduino community.

I'd be happy to provide any assistance with setting up a workflow using `arduino/compile-sketches`, or just using Arduino CLI directly if for some reason you don't want to use the action. I can say with confidence that I'm the world's foremost authority when it comes to `arduino/compile-sketches` 😆

2bndy5, 2020-09-19:
Honestly full PEP8 compliance may not be possible on this wrapper (I'll look at it again) because of the naming scheme (casing - blanking on the correct term) used in the underlying C++  API.

>Full Circuitpython support would also be nice, but that's an another issue 😄...

It's [my other nRF24 repo that I'm actually working on improving right now](https://github.com/2bndy5/CircuitPython_nRF24L01/pull/14), but I don't have bells and whistles like RF24Network etc. Linked pull request is waiting on some testing of the PA/LNA modules. See also [my comment here about difference in API](https://github.com/nRF24/RF24/issues/589#issuecomment-694098166). That CircuitPython package is structured to include other python modules like a pure python port of RF24Network, FakeBLE, etc. under 1 pipe install.

Avamander, 2020-09-19:
@2bndy5 

>but I don't have bells and whistles like RF24Network etc.

Well, just having the base, RF24, would suffice, but yeah, this is a topic for an another issue.

2bndy5, 2020-09-19:
From an issue on the CircuitPython repo:
>We could update to newer MicroPython to get loadable native modules but that is a huge undertaking.

It would be a dead issue if you're talking about using C++ code for CircuitPython. Adafruit might be open to a native RF24 module (not package) built into the CircuitPython firmware, but I doubt it.

2bndy5, 2020-09-21:
I keep finding weird formatting in sub-directories' c++ files. Recent example:
```c++
static int sysFds[64] = {-1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                         -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                         -1, -1, -1, -1, -1, -1,};
```
where this would be better:
```c++
static int sysFds[64] = {
    -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
    -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
    -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
    -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1
};
```
which files are we mainly concerned with? All of them or just root directory?

Avamander, 2020-09-21:
> where this would be better:

It wouldn't, it'd be inconsistent with the style guide used as the base.

2bndy5, 2020-09-21:
141 character line length?

Avamander, 2020-09-21:
No it's that multiline initializators and function calls keep the indentation level of the starting (curly) brace.

2bndy5, 2020-09-21:
my concern was about the line length. I'm fine with maintaining indentation, but this makes me cringe:
[image]

that pdf in the CONTRIBUTING.md is old -- I was 10 years old in 1994, and I don't remember "word/line wrap" being a common-place feature back then. It doesn't specify a recommended max line length (that I could find). Not to mention its has no link-able table of contents. I would try copying and pasting text from it to use a direct reference, but selecting 1 line ends up copying the left half of the line and the right half of the paragraph its in.

I tried highlighting the words "four spaces", and this is how it reacts:
[image]

Avamander, 2020-09-21:
Your editor's hard-wrapping is causing that weirdness. The style guide base I _think_ indeed doesn't contain a line-length limit, but there's also no need to enforce the line length my VT-100 has. Abandoning such a tiny limit is also the approach the Linux project took just recently, they're now using a soft limit 100 characters.

Avamander, 2020-09-21:
So in conclusion, let's not force a line length, try not to run an autoformatter that tries to "intelligently" wrap lines - it probably needs a human to do so.

With the current example, if one was to keep the indentation, then I think it's totally fine to turn this:
```
static int sysFds[64] = {-1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                         -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                         -1, -1, -1, -1, -1, -1,};
```

into something like this, if it really annoys you:

```
static int sysFds[64] = {-1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, 
                         -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                         -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, 
                         -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, 
                         -1, -1, -1, -1};
```

2bndy5, 2020-09-22:
I know its a different syntax, but PEP8 has a soft 79 char line length limit (99 is the hard limit for black and pylint). Although, that's due to the fact that python is a scripting language while `gcc` ignores whitespace & comments when compiling so that its a non-issue. @Avamander please tell me you don't still code with your VT-100 (80 char wide display -- according to wikipedia).

Avamander, 2020-09-22:
> please tell me you don't still code with your VT-100 (80 char wide display -- according to wikipedia)

Nope :smile:

2bndy5, 2020-09-22:
yea, using the auto-formatter in VSCode turns the declaration in question into a 64 line declaration. <- Much much worse.

2bndy5, 2020-09-22:
@Avamander Since you made a reference the linux project, are there discrepancies using their C style guide (I can't find exactly what you were referring to)?

Avamander, 2020-09-22:
> are there discrepancies using their C style guide (I can't find exactly what you were referring to)?

I'd have to check again, but pretty sure there were.

2bndy5, 2020-09-22:
link too? I'm curious

Avamander, 2020-09-22:
https://www.kernel.org/doc/html/v4.10/process/coding-style.html

2bndy5, 2020-09-22:
well they prefer 8 space indentations... IMO that doc is rather opinionated/biased. So, no we can't consider linking to that.

I found the following comparison rather drastic
>There are heretic movements that try to make indentations 4 (or even 2!) characters deep, and that is akin to trying to define the value of PI to be 3.

docs should also avoid insulting statements like these:
>Encoding the type of a function into the name (so-called Hungarian notation) is brain damaged - the compiler knows the types anyway and can check those, and it only confuses the programmer. No wonder MicroSoft makes buggy programs.

2bndy5, 2020-09-22:
BTW I do prefer the readthedocs-sphinx theme over the doxy-gen output that http://tmrh20.github.io/RF24 uses. There is a sphinx extension that will wrap doxy-gen docs, but I haven't tried it out... yet. I'm getting off topic again; that should be a separate issue when I start understanding how to use doxy-gen.

Avamander, 2020-09-22:
> I do prefer the readthedocs-sphinx theme over the doxy-gen output that http://tmrh20.github.io/RF24 uses. 

Doxygen could be configured to use [m.css](https://mcss.mosra.cz/) [that looks like this](https://doc.magnum.graphics/magnum/). But yes, this is a separate topic.

wmarkow, 2020-10-01:
>status on this?

I just did my tests with formatters like 2 years ago and that's it - from my side.

>Pretty much done. Arduino examples are formatted using its IDE, code is formatted as agreed and new PRs are fixed before merging. It should be fairly consistent.

I think I lost the scope during those 2 years. What kind of formatter we use currently? Is it NASA code style? Personally I like NASA.

>well they prefer 8 space indentations...

This may be done in purpose. 8 space indentation plus for example 80 characters per line means that you can't write the code with many nested *if* or loops statements because it will not fit on your terminal screen. Besides many nested code looks bad and is hard to understand and analyse.

2bndy5, 2020-10-21:
>If you want to take the astyle approach I mentioned earlier in this thread, you could use this workflow:
https://github.com/arduino/arduino-examples/blob/main/.github/workflows/code-formatting-check.yml
It's configured to only cover the example sketches, but only a tiny change would be needed to make it cover the library code files as well.

@per1234 I tried the job in this workflow, using it as an attempt to try and prevent the long-running "build" job that "needs" it to pass. It found a lot wrong with the examples (may have been my fault), so I went through all the examples the workflow marked as erroneous. After committing the changes from auto-formatter tool in the Arduino IDE, the workflow still marked all the same files as erroneous. You can [see the results here](https://github.com/2bndy5/RF24/runs/1286346121?check_suite_focus=true#step:5:1) (if you're interested), but be warned its a 2000+ line log. Just thought I'd let you know (I think I'll go without the format checking job).

per1234, 2020-10-21:
>After committing the changes from auto-formatter tool in the Arduino IDE, the workflow still marked all the same files as erroneous.

Here's the clue for why this happened:
https://github.com/2bndy5/RF24/runs/1286346121?check_suite_focus=true#step:5:25
```
Cannot open default option file /home/runner/work/RF24/RF24/examples_formatter.conf
```
You didn't add the formatter configuration file:
https://github.com/arduino/arduino-examples/blob/main/examples_formatter.conf
to the repo when you added the formatting check. This caused Artistic Style to use its default configuration, which doesn't match the Arduino formatting style.

You can see here the passing check in my fork:
https://github.com/per1234/RF24/runs/1289296630?check_suite_focus=true

After I added https://github.com/arduino/arduino-examples/blob/main/examples_formatter.conf, there was still quite a bit of non-compliant formatting. The reason is that the formatter configuration provided in the arduino/arduino-examples repo adds some additional configuration options beyond that of the configuration used by the Arduino IDE's Auto Format. This means that, though code formatted with the "examples_formatter.conf" configuration will always be compliant with Arduino IDE's auto format configuration, the reverse is not always true.

My quick fix for the sake of a proof of concept was to just replace the configuration with the one used by the Arduino IDE:
https://github.com/arduino/Arduino/blob/1.8.13/build/shared/lib/formatter.conf
I do like the "examples_formatter.conf" configuration though, so if you chose to have another try at this, you might use that configuration. You can use Artistic Style directly from the command line to format everything. I use this command:
```
astyle --suffix=none --options=examples_formatter.conf --recursive ./*.pde,*.ino,*.h,*.hpp,*.hh,*.hxx,*.h++,*.cpp,*.cc,*.cxx,*.c++,*.cp,*.c,*.ipp,*.ii,*.ixx,*.inl,*.tpp,*.txx,*.tpl
```
You can get astyle compiled for Windows here: https://sourceforge.net/projects/astyle/files/
and instructions for installing for other OSs here: http://astyle.sourceforge.net/install.html

After changing the configuration to match the IDE's auto formatter, there were still a couple examples that weren't compliant. Maybe you just missed those by accident. After that, there was one last recalcitrant file:
https://github.com/per1234/RF24/runs/1289283492?check_suite_focus=true#step:5:79
```
ERROR: Non-compliant code formatting in /home/runner/work/RF24/RF24/examples/scanner/output/scanner.cpp
```
This `examples/scanner/output` folder is an artifact of an ancient Arduino IDE version from when it used to save the build output to the sketch folder. maniacbug accidentally committed that to the repo 10 years ago, and there it has stayed ever since. Because it can cause confusion and maintenance burden (as we now have proof), this folder should be removed from the repository. However, I took the minimalist approach of just formatting the file.

With that, the check now passes:
https://github.com/per1234/RF24/runs/1289296630?check_suite_focus=true

---
Since the time I made that comment about the `arduino/arduino-examples` repo's formatting check workflow, I have made what I feel to be some improvements on the system, when I added this check to the `SpenceKonde/megaTinyCore` repository. You can see the workflow here:
https://github.com/SpenceKonde/megaTinyCore/blob/master/.github/workflows/check-code-formatting.yml
and the formatter configuration file used by the workflow here:
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ci/arduino-code-style.conf
Because `SpenceKonde/megaTinyCore` is a pretty big project with multiple distinct components, I broke the formatting checks into multiple jobs in that workflow, with the idea that this might make it easier to understand the results of a failing check. However, that's completely optional. I could have defined all the paths to check in a single step

The functionality is the same, but I have moved that ugly bash code into a GitHub Actions action. This makes it  so the workflow can be very minimal and friendly to configure and the check code itself can be maintained from a single centralized location:
https://github.com/per1234/artistic-style-action

2bndy5, 2020-10-28:
@per1234 I'm loving the artistic-style-action. Ran into yet another yml snag: list of file formats requires a single quote `'`  where a double quote `"` caused an error saying `expected to find an alphabetical or numerical character`. Also I had to remove the option `remove-comment-prefixes` from the examples_formatter.conf file, but this is more of a continuity problem doxygen as the examples descriptions are actually located in the header file (which requires some extra copy-n-paste from example files' description to the header file that declares the example files).

I think I'm ready to call this issue resolved.

per1234, 2020-10-28:
>I'm loving the artistic-style-action.

I'm very happy if it can be useful! I actually haven't used it on any other CI projects since the megaTinyCore repo, but it was a fun opportunity to learn about GitHub's new "composite run steps" action type.

>Ran into yet another yml snag: list of file formats requires a single quote '

YAML certainly has some tricky things like that. The language seems so simple, right up until it isn't! I tend to always quote everything, with the most strict quotes possible, but it seems the common practice with YAML is just the opposite (which I suppose does work out well for the most basic use cases).

>I had to remove the option remove-comment-prefixes from the examples_formatter.conf file

That's not so bad. You're still pretty close to the standard Arduino formatting style. that `remove-comment-prefixes` is actually somewhat confusing because the Arduino IDE's editor automatically adds comment prefixes, but then its auto format removes them (https://github.com/arduino/Arduino/issues/3941). So you sort of end up with a tug of war over the comment prefixes.

2bndy5, 2020-12-07:
assigned myself this issue (as a reminder) because it will likely be closed after merging #691

2bndy5, 2020-12-14:
So, examples are properly formatted to AStyle preferences (configured to suite the Arduino IDE) as well as any source code. I rewrote the Linux examples with NASA C style guide in mind. All python examples follow PEP8 standards.

Some doxygen comment blocks seemed misaligned or not conformed with RF24.h comment block style, and there may still be some more like that. However, #691 changed 104 files with this issue in mind, so I'm going to call this resolved until otherwise noted. 

If there are any other formatting errors that may have been missed, we can re-open this or do a quick PR/commit.

closed 2020-12-14 by 2bndy5

2bndy5, 2021-08-17:
@Avamander I've been looking at clang-format as a tool (also available as GH actions) we can use to check C/C++source code format. It comes with some builtin/popular style-guides that can be specified as what rules to follow, but there is an additional option to make our own set of rules (which can inherit from the aforementioned style guides) that gets saved to a file in the root of the repository.

More info can be [found here](https://clang.llvm.org/docs/ClangFormatStyleOptions.html#configurable-format-style-options). BTW I really like the [Webkit style guide](https://www.webkit.org/coding/coding-style.html), but there are some rules I'd like to ignore/override. Still, I don't think we can beat the practicality and flexibility of the NASA C style guide (the link in _CONTRIBUTING.md_ works again - they must have finished their site update).

per1234, 2021-08-17:
Arduino decided to switch from Astyle to ClangFormat for Arduino IDE 2.x. There is also some work to do the same for the classic Arduino IDE: https://github.com/arduino/Arduino/pull/11543

The ClangFormat configuration file for the Arduino code style is here: https://github.com/arduino/Arduino/blob/7754b7fc8a127b46ee099fbea8e8a71b70d3486d/build/shared/clang-format-arduino

2bndy5, 2021-08-17:
@per1234 Good to hear (great minds seem to think alike). I also looked into using CPPCheck because Cmake has some preliminary support for environments that already have that installed. However, CPPCheck seems to focus on preventing undefined behavior and not much about code style. Still it wouldn't be bad to use both in a CI workflow.

Avamander, 2021-08-17:
Yep, clang-format is nice, it's just a bit of work to get nice results if you're trying to not cause too many changes in existing codebase.

2bndy5, 2021-08-17:
I just went through all the options and got a configuration that should match the style I've been seeing consistently (in RF24 libs) and from what I remember of NASA C style.

1 major problem that I've been fighting: the _.clang-format_ file CANNOT be utf-16 (I think it only works with utf-8). Although I think this may be more specific to windows because I generated the file from webkit style (running clang-format 32-bit exe).

per1234, 2021-08-17:
>it's just a bit of work to get nice results if you're trying to not cause too many changes in existing codebase.

It was very challenging to try to write a ClangFormat configuration that matched the Artistic Style configuration used by the Arduino IDE's auto format. That astyle configuration was very relaxed. You want attached braces? Cool. You want them broken? No problem. (Or more likely) you want a random mixture of attached and broken? Sure thing. ClangFormat doesn't play like that. You can pick any possible single style you could imagine, but there is usually no way to just shut it off.

In the end, we had to accept that it would not be possible to get a 100% match to the previous configuration.

[8 more replies]