nRF24/RF24 #392: Problem using PowerDown() transmission stops working

furiati, 2017-12-07:
Hi,

I have a sketch working where de PTX send a packet to the PRX, and the PRX send data from sensors included in the ACK payload. The sketch is working fine until I try to use the function powerDown to save battery on the PRX. When I use PowerDown() the PTX stops receiving the ACK payload.

PTX code:

`#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"

// Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8
RF24 radio(7, 8);

// Topology
const uint64_t pipes[2] = { 0xABCDABCD71LL, 0x544d52687CLL };              // Radio pipe addresses for the 2 nodes to communicate.

// A single byte to keep track of the data being sent back and forth
byte counter = 1;

struct datos {
  float T;
  float P;
};

datos tiempo;

void setup() {

  Serial.begin(115200);

  // Setup and configure rf radio

  radio.begin();
  radio.setAutoAck(1);                    // Ensure autoACK is enabled
  radio.enableAckPayload();               // Allow optional ack payloads
  radio.setRetries(15, 15);                // Smallest time between retries, max no. of retries
  radio.setPayloadSize(8);                // Here we are sending 1-byte payloads to test the call-response speed
  radio.setDataRate( RF24_2MBPS );
  radio.openWritingPipe(pipes[0]);
  radio.openReadingPipe(1, pipes[1]);
  radio.startListening();                 // Start listening
  radio.printDetails();                   // Dump the configuration of the rf unit for debugging
}

void loop(void) {
  radio.stopListening();                                  // First, stop listening so we can talk.
  Serial.print("Now sending ");
  Serial.print(counter);
  Serial.println(" as payload. ");
  byte gotByte;
  unsigned long time = micros();                          // Take the time, and send it.  This will block until complete
  //Called when STANDBY-I mode is engaged (User is finished sending)
  if (!radio.write( &counter, 1 )) {
    Serial.println(F("failed."));
  } else {
      if (!radio.available()) {
      Serial.println(F("Blank Payload Received."));
    } else {
      while (radio.available() ) {
        unsigned long tim = micros();
        radio.read( &tiempo, sizeof(tiempo) );
        Serial.print("Got response ");
        Serial.print(gotByte);
        Serial.print(" round-trip delay: ");
        Serial.print(tim - time);
        Serial.println(" microseconds");
        Serial.println(tiempo.T);
        Serial.println(tiempo.P);
        counter++;
      }
    }
  }
  // Try again later
  delay(2000);
}`

PRX Code:

`#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"
#include "LowPower.h"

// Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8
RF24 radio(7, 8);

// Topology
const uint64_t pipes[2] = { 0xABCDABCD71LL, 0x544d52687CLL };              // Radio pipe addresses for the 2 nodes to communicate.

// A single byte to keep track of the data being sent back and forth
byte counter = 1;

struct datos {
  float T;
  float P;
};

datos tiempo;

bool envio;

//Dallas DS18B20
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature.
DallasTemperature sensors(&oneWire);

// arrays to hold device address
DeviceAddress insideThermometer = { 0x28, 0xFF, 0x45, 0xC6, 0x60, 0x16, 0x05, 0x9F };

void setup() {

  Serial.begin(115200);
  //Dallas DS18b20
  sensors.begin();
  sensors.setResolution(insideThermometer, 12);
  printf_begin();

  // Setup and configure rf radio

  radio.begin();
  radio.setAutoAck(1);                    // Ensure autoACK is enabled
  radio.enableAckPayload();               // Allow optional ack payloads
  radio.setRetries(15, 15);                // Smallest time between retries, max no. of retries
  radio.setPayloadSize(8);               // Here we are sending 1-byte payloads to test the call-response speed
  radio.setDataRate( RF24_2MBPS );
  radio.openWritingPipe(pipes[1]);        // Both radios listen on the same pipes by default, and switch when writing
  radio.openReadingPipe(1, pipes[0]);
  radio.startListening();                 // Start listening
  radio.printDetails();                   // Dump the configuration of the rf unit for debugging
}

void loop() {
  // put your main code here, to run repeatedly:
  byte pipeNo;
  byte gotByte;
  sensors.requestTemperaturesByAddress(insideThermometer);
  tiempo.T = sensors.getTempC(insideThermometer);                                    // Dump the payloads until we've gotten everything
  tiempo.P = 1.03;
  while ( radio.available(&pipeNo)) {
    radio.read( &gotByte, 1 );
    radio.openWritingPipe(pipes[1]);     
    radio.writeAckPayload(pipeNo, &tiempo, sizeof(tiempo) );
    envio = true;
  }
  if (envio){
    Serial.println("PowerDown NRF24L01");
    radio.powerDown();
    delay(5000); //Power down radio for 5 seconds
    Serial.println("PowerUP NRF24L01");
    radio.powerUp();
    radio.startListening();
    envio = false;
  }
}`

What i am missing?

Thank you in advance.

Bye!!

TMRh20, 2017-12-09:
Try this:
```
if (envio){
    Serial.println("PowerDown NRF24L01");
    radio.powerDown();
    delay(5000); //Power down radio for 5 seconds
    Serial.println("PowerUP NRF24L01");
    radio.powerUp();
    radio.startListening();
    radio.writeAckPayload(pipeNo, &tiempo, sizeof(tiempo) );   //******** Adds this line
    envio = false;
  }
```

You can also write an ack payload in the setup, so the PTX is not receiving blank payloads.

furiati, 2017-12-10:
Hi,

I tried what you suggest and it works, the PTX receive the payload with the temperature and pressure. But, correct me if I am wrong, I assume this data is from 5 seconds ago, not from "now".

This "error" is acceptable even though I increase the sleep time. Although I would like to know what is happening. Why when I introduce the power down code, the PTX don't receive the payload? Is it a timing problem or what? Because when I load the sketch without any power down code the transmission works perfect. I am lost here.

Thank you very much for your help.

TMRh20, 2017-12-10:
The ack payload feature uses the same 3 level fifo as tx/rx. When you power down, fifo is flushed. Pay close attention to blank payloads, these are successful, just that the fifo was empty when the request was made. 

Ie: the same 5 minutes ago problem is apparent in your existing code, just hard to spot it

furiati, 2017-12-10:
So another approach could be delay the power down to give time to empty the fifo before it is flushed, am I right?

I think it is clear for me now.

Thank you again, this thing was driving me mad.

TMRh20, 2017-12-10:
Nope, SPI devices are not always reliably quick enough to buffer an ack payload after a request is made. They should be pre-loaded before reception.

furiati, 2017-12-10:
Thanks for the clarification.

All clear now!

:)

closed 2017-12-10 by furiati

wmarkow, 2017-12-11:
Let me put my few words about the scenario. RF24 module in PRX now sleeps for most of the time; every 5 seconds it wakes up and sends the sensor status to PTX. I think there is no need to send data from PTX to PRX because these data will not be received by PRX (because PRX is in a sleep mode for most of the time).

To make it simple you can reverse the roles of your boards: the board with temperature sensor will be a PTX. It will sleep for the most of the time and once per 5 second it will send a sensor status to the PRX (the other board). PRX must not sleep but constantly polls the RF24 chip if there are any new data with the temperature status. PRX can be then connected to a power supply while PTX can be even battery operated. @furiati , maybe that solution will be better for you?

furiati, 2017-12-11:
Ok, I think I was a little vague in my explanation. This sketches are not finished yet. My goal is to have one station that receive data from multiple stations with sensors (e.g. my house). To avoid collisions and to make the sketch simpler I thought this architecture would be better, taking advantage from de ack payload (poll data without changing between RX/TX).

The is a lot of work to do yet. One of this things is to modify the sketches to improve the timing so if the base station poll data every 6 seconds, the sensor station sleeps 5 seconds.

This are my intentions, what do you think? I tried to explain it as clear as possible.

wmarkow, 2017-12-12:
@furiati,

>One of this things is to modify the sketches to improve the timing so if the base station poll data every 6 seconds, the sensor station sleeps 5 seconds.

I thing this will never work. If the sensor station (actually the RF24 chips is powered down) sleeps for 5 second it will never receive your poll data while sleeping (because RF24 is powered down). The base station needs to wait until the sensor station is awake and then it needs to send a poll data. I think this requires a good synchronization algorithm.

Better option would be:
 * sensor station sleeps for 5 second, then wakes up, reads data from the sensors, sends the data to the main station (without polling) and goes to sleep again
 * have one main station that in most cases listens to the data from sensors stations
 * main station is able to do something with the received data from the sensor station. It can be connected to a controller (like **openhab** or **domoticz** running on a RapsberryPi), so you extend the functionality quite fast.

@furiati, I'm trying to built something similar at home. Maybe we could exchange some knowledge?

furiati, 2017-12-12:
I think this could work, which does not mean that it is the best option. The idea is: main station send a request to the sensor station, the sensor station responds with the sensor data and starts de power down sleep process for X seconds. When the main station receive the data, it calculates how long it has taken (like in my sketch and rf24 examples) and reset the time counter for next polling. So we know time until next polling will be something like: X + transmission time/2 + Y. Y will be a little more time to be safe. Maybe i am over complicating thing, anyway i am learning things.

I did not make any test using the 6 data pipes at the same time. I don't know if it will be collisions problems or not. have you been experimenting with this?

I did some test with a  demos d1 mini, it works with the RF24 library, unless the printf function. My plan is to upload the gathered data to Thingspeak or similar ioT service. I think it is enough for my purpose and it is way cheaper than a RaspberryPi which can be used for more complex things.

@wmarkow, Yes we can! I don't know if my little knowledge will help you but there it is.