nRF24/RF24 #393: port to the ESP32?

olivier-lefebvre, 2017-12-10:
I am trying to drive a nrf24l01+ with a ESP32 so that I can connect a bunch of coin cell nrf24 nodes to the web.
I have done some research but I have found very few things about using the nrf24 with the ESP32.
To have the RF24 library compile with ESP32 arduino core, I made to following change to the library:
in nRFL01.h in commented out this line: 
   `//#define NOP          0xFF`

in RF24.cpp I replaced
    `*buf++ = _SPI.transfer(NOP);`
by 
   ` *buf++ = _SPI.transfer(0xff);`

in RF24_config.h

I added 

   ```
 #elif defined (ESP_H)
    #include <pgmspace.h>
    #define PRIPSTR "%s"
```

just after line 135

Those changes allow the library to compile with no warnings.

I used the lastest ESP32 arduino core download from the github repo as well as the latest version of the RF24 library from github.

I am trying to run the getting started sketch from the library with a few modif:

		```
/*
		* Getting Started example sketch for nRF24L01+ radios
		* This is a very basic example of how to send data from one node to another
		* Updated: Dec 2014 by TMRh20
		*/

		#include <SPI.h>
		#include "RF24.h"

		/****************** User Config ***************************/
		/***      Set this radio as radio number 0 or 1         ***/
		bool radioNumber = 0;

		/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
		RF24 radio(5,18); //CE, CSN
		/**********************************************************/

		byte addresses[][6] = {"1Node","2Node"};

		// Used to control whether this node is sending or receiving
		bool role = 0;

		void setup() {
		  Serial.begin(115200);
		  Serial.println(F("RF24/examples/GettingStarted"));
		  Serial.println(F("*** PRESS 'T' to begin transmitting to the other node"));
		  SPI.begin();
		  SPI.setFrequency(2000000);
		  SPI.setBitOrder(MSBFIRST);
		  SPI.setDataMode(0);
		  if (radio.begin()) { Serial.println("OK!"); }
		  else{ Serial.println("too bad!"); }
		  delay(5000);
		  // Set the PA Level low to prevent power supply related issues since this is a
		 // getting_started sketch, and the likelihood of close proximity of the devices. RF24_PA_MAX is default.
		  radio.setPALevel(RF24_PA_LOW);
  
		  // Open a writing and reading pipe on each radio, with opposite addresses
		  if(radioNumber){
			radio.openWritingPipe(addresses[1]);
			radio.openReadingPipe(1,addresses[0]);
		  }else{
			radio.openWritingPipe(addresses[0]);
			radio.openReadingPipe(1,addresses[1]);
		  }
  
		  // Start the radio listening for data
		  radio.startListening();
		}
                void loop(){}
```

I have tried playing with different SPI parameters with no success. I lowered the frequency to 2Mbs and try to set the SPI in mode 0, MSBFIRST with no success.

On the hardware side I use:
- DOIT ESP32 DEVKIT v1
- a nrf24 adapter (the one with the power regulator and cap)
- a basic module

I tried swapping the cable, module, adapter with no success.

I power the dev board through USB and with an external beefy regulated power supply. I have a large cap (470uF) between (GND and 3.3V).

The connection are as follow:
 CE->GPIO18
CSN -> GPIO05
MISO -> GPIO12
MOSI -> GPIO13
CLK-> GPIO14
IRQ-> unconnected

My guess is there is a problem with the SPI communication protocol. The radio.begin() always fails....

I am a little bit out of my depth here, any help would be greatly appreciated!

Avamander, 2017-12-10:
Can you make a Serial.print([inser message here]) after each SPI transaction?

olivier-lefebvre, 2017-12-10:
Actually the code does not freeze. radio.begin() returns zero....

Avamander, 2017-12-10:
Okay, and then radio.printDetails()?

wmarkow, 2017-12-11:
@olivier-lefebvre, you can try to call **isChipConnected()** (in the **setup()** method) and see the result. If it is **false** then you have some hardware issues or SPI stack issues.
And please post the **radio.printDetails()** as @Avamander suggested.

olivier-lefebvre, 2017-12-11:
Thanks @wmarkow. 
I just did not wire the SPI properly, that is a little embarrassing! What through me off was that they are multiple SPI mapped in the pinout and it seems I wire the board to an alternative SPI port.

To sum up, the library almost work out of the box with ESP32. The only change to make are with NOP that is already defined in the ESP32 arduino core and that conflict with the definition in nRFL01.h.
I also added, in RF24_config.h after line 135
```
#elif defined (ESP32)
 #include <pgmspace.h>
 #define PRIPSTR "%s"
```
to match what was done with the ESP8266.

Compile error are thrown if I include <printf.h>
and printf.begin();
Without the two line above, the sketch compile and radio.printDetails() print the NRF24 details but cause the ESP32 to reset....

I close the issue for now and will report back if I figure how to print details without resetting the ESP32. Thanks again for your help!

closed 2017-12-11 by olivier-lefebvre

colesnicov, 2018-04-21:
Try this [https://github.com/espressif/arduino-esp32/issues/1219](https://github.com/espressif/arduino-esp32/issues/1219) :-)

sherif1717, 2018-11-23:
Kindly consider the following for Successful Configurations of (nRF24L01F + RF24 + ESP32):
Successful Configurations of (nRF24L01F + RF24 + ESP32)
1-	The R24 Library is working directly with ESP32 without any modifications except
        as you said, go to RF24_config.h and just after line 135 add the following code:

       #elif defined (ESP32)
       #include <pgmspace.h>
       #define PRIPSTR "%s"

2-	Use the following ESP32 pin configuration for wairing ESP32 to nRF24L01+
        CE   -> GPIO17  GPIO
        CSN  -> GPIO05  VSPI SS
        MISO -> GPIO19  VSPI MISO
        MOSI -> GPIO23  VSPI MOSI
        CLK  -> GPIO18  VSPI CLK
        IRQ-> unconnected

3-	For testing I used the standard R24 library example (getting started) as it is except in one setup (ESP32 with 1st nRF24L01+) I adjusted:
bool radioNumber = 0;  for 1st setup and 
bool radioNumber = 1;  for 2nd setup 

/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 17 & 5 */
//create an RF24 object
RF24 radio(17, 5);  // CE, CSN

5-	For the hardware setup I made the following:
-	Connected 10uF 16V capacitor electrolytic on pins 1&2 (GND & VCC) of nRF24L01+ in both transmitter and receiver sides.
-	In 1st setup I used separate breadboard power supply to power the 3.3Vdc to the nRF24L01+ Vcc pin. The GND pin to connected to both GND pins in power supply and the ESP32
-	In 2nd setup I powered the nRF24L01+ Vcc from the 3.3Vdc pin of ESP32 without using external power supply.

TESLA-UFPA, 2019-01-13:
@basic-user 
You could provide the code and pin connection?

sherif1717, 2019-05-20:
[image]

sherif1717, 2019-05-20:
//Include Libraries
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

/***
The connection are as follow:
CE   -> GPIO17  GPIO
CSN  -> GPIO05  VSPI SS
MISO -> GPIO19  VSPI MISO 
MOSI -> GPIO23  VSPI MOSI
CLK  -> GPIO18  VSPI CLK
IRQ-> unconnected

kindly add also below line to RF24_config.h and just after line 135 add the following code:

   #elif defined (ESP32)
   #include <pgmspace.h>
   #define PRIPSTR "%s"

***/
//===================================================
/*
* Getting Started example sketch for nRF24L01+ radios
* This is a very basic example of how to send data from one node to another
* Updated: Dec 2014 by TMRh20
*/
/****************** User Config ***************************/
/***      Set this radio as radio number 0 or 1        ***/
bool radioNumber = 0;  // 0 for Transmitter
                       //  radio.openWritingPipe(addresses[0]);
                       //  radio.openReadingPipe(1, addresses[1]);

/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 17 & 5 */
//create an RF24 object
RF24 radio(17, 5);  // CE, CSN
/**********************************************************/

byte addresses[][6] = { "1Node","2Node" };

IkpeohaGodson, 2019-06-17:
I have added the following to RF24_config.h after line 135 as suggested above:

#elif defined (ESP32)
#include <pgmspace.h>
#define PRIPSTR "%s"

I downloaded the patched library and added it to my Arduino IDE libraries. However, when I tried including it in my code, the R24 library that the IDE added was the old one and not the patched one. I've tried all I could to include the patched library to my code without success. I need help on this. How do I include the patched library to my code? I'm relatively new to coding, so any help will be useful to me. Thank you.

bhavikbhansali, 2020-03-15:
I had an issue for the initial connection where I am connecting a CE and CSN pin on other GPIO. but when I go through this page and make a connection accordingly I am able to send the data using ESP32 with NRF24L01.

Thanks for the comment on this.

Stevenaris1944, 2020-07-11:
Sorry, I kinda new in this, but what is this mean? *buf++ = _SPI.transfer(0xff);

should I create new line or replace it, and where? I search on RF24.cpp but I can't find it.

Please help

TMRh20, 2020-07-11:
@Stevenaris1944 This issue is closed because there is no longer an issue. ESP32 support is included.