nRF24/RF24 #398: NodemcuV3 Lolin + PROGMEM + RF24 (crash with last lib)

JLuisAlcaraz, 2018-01-01:
Hi all!,
I use a Nodemcu V3 Lolin board + nRF24L01+ in order to send/receive packs of 6 bytes from another Nano board with nRF24L01+ attached.

If I use the RF24 lib from inside Arduino IDE, installing with lib manager, all works properly except for PROGMEM (the lib installed is 1.3.0)
If I download the lib (and install manually) from this link:
https://github.com/nRF24/RF24
then PROGMEM works well, but I ave continuos soft reset.

Same wiring, one lib works without PROGMEM (1.3.0), other lib don't work but PROGMEM is ok (https://github.com/nRF24/RF24).

Wiring is:
nRF24L01+         Nodemcu
IRQ                           -
MOSI                      D7
CSN                        D0
3.3v                         3.3v
MISO                      D6
SCK                         D5
CE                           D3
GND                        GND

Avamander, 2018-01-01:
Reconfigure your IDE properly define the right preprocessor flags for the device your're using. `ARDUINO_ARCH_ESP8266` must be defined if you want the library to work properly on an ESP.

JLuisAlcaraz, 2018-01-01:
Thank you very much, Avamander!
How can I do that? Where I must define "ARDUINO_ARCH_ESP8266"?

closed 2018-01-02 by Avamander