nRF24/RF24 #399: Issue compiling with NodeMCU

Watson221, 2018-01-02:
Hi, I'm just using the LED remote example, because it's perfect for my project. The code is virtually identical to the LED  remote example just with the define part and the pins and led numbers edited for my project. 

I read on here that the MOSI, SCK, and MISO pins are hard wired into the code. 
http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo

The issue is that the NodeMCU doesn't have pins 11-13 that the code requires for the MOSI, SCK, and MISO pins.  I thought that by doing the #define thing that it would fix the issue, but it didn't.  The IDE won't compile the code (exit status 1).

I think  comment #398 had a similar issue, but in my IDE  my board is defined as the NodeMCU. So I really don't know how to work around that hard wire in the code. 

Any help would be appreciated,
Thanks!

`#define D6 13  // SCK
#define D7 12  // MISO
#define D8 11  // MOSI

#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

//
// Hardware configuration
//

// Set up nRF24L01 radio on SPI bus plus pins 9 & 10 (CE & CS)

RF24 radio(3,2);

// sets the role of this unit in hardware.  Connect to GND to be the 'led' board receiver
// Leave open to be the 'remote' transmitter
const int role_pin = A0;

// Pins on the remote for buttons
const uint8_t button_pins[] = { 8,9 };
const uint8_t num_button_pins = sizeof(button_pins);

// Pins on the LED board for LED's
const uint8_t led_pins[] = { 4,5 };
const uint8_t num_led_pins = sizeof(led_pins);

//
// Topology
//

// Single radio pipe address for the 2 nodes to communicate.
const uint64_t pipe = 0xE8E8F0F0E1LL;

//
// Role management
//
// Set up role.  This sketch uses the same software for all the nodes in this
// system.  Doing so greatly simplifies testing.  The hardware itself specifies
// which node it is.
//
// This is done through the role_pin
//

// The various roles supported by this sketch
typedef enum { role_remote = 1, role_led } role_e;

// The debug-friendly names of those roles
const char* role_friendly_name[] = { "invalid", "Remote", "LED Board"};

// The role of the current running sketch
role_e role;

//
// Payload
//

uint8_t button_states[num_button_pins];
uint8_t led_states[num_led_pins];

//
// Setup
//

void setup(void)
{
  //
  // Role
  //

  // set up the role pin
  pinMode(role_pin, INPUT);
  digitalWrite(role_pin,HIGH);
  delay(20); // Just to get a solid reading on the role pin

  // read the address pin, establish our role
  if ( digitalRead(role_pin) )
    role = role_remote;
  else
    role = role_led;

  //
  // Print preamble
  //

  Serial.begin(115200);
  printf_begin();
  printf("\n\rRF24/examples/led_remote/\n\r");
  printf("ROLE: %s\n\r",role_friendly_name[role]);

  //
  // Setup and configure rf radio
  //

  radio.begin();

  //
  // Open pipes to other nodes for communication
  //

  // This simple sketch opens a single pipes for these two nodes to communicate
  // back and forth.  One listens on it, the other talks to it.

  if ( role == role_remote )
  {
    radio.openWritingPipe(pipe);
  }
  else
  {
    radio.openReadingPipe(1,pipe);
  }

  //
  // Start listening
  //

  if ( role == role_led )
    radio.startListening();

  //
  // Dump the configuration of the rf unit for debugging
  //

  radio.printDetails();

  //
  // Set up buttons / LED's
  //

  // Set pull-up resistors for all buttons
  if ( role == role_remote )
  {
    int i = num_button_pins;
    while(i--)
    {
      pinMode(button_pins[i],INPUT);
      digitalWrite(button_pins[i],HIGH);
    }
  }

  // Turn LED's ON until we start getting keys
  if ( role == role_led )
  {
    int i = num_led_pins;
    while(i--)
    {
      pinMode(led_pins[i],OUTPUT);
      led_states[i] = HIGH;
      digitalWrite(led_pins[i],led_states[i]);
    }
  }

}

//
// Loop
//

void loop(void)
{
  //
  // Remote role.  If the state of any button has changed, send the whole state of
  // all buttons.
  //

  if ( role == role_remote )
  {
    // Get the current state of buttons, and
    // Test if the current state is different from the last state we sent
    int i = num_button_pins;
    bool different = false;
    while(i--)
    {
      uint8_t state = ! digitalRead(button_pins[i]);
      if ( state != button_states[i] )
      {
        different = true;
        button_states[i] = state;
      }
    }

    // Send the state of the buttons to the LED board
    if ( different )
    {
      printf("Now sending...");
      bool ok = radio.write( button_states, num_button_pins );
      if (ok)
        printf("ok\n\r");
      else
        printf("failed\n\r");
    }

    // Try again in a short while
    delay(20);
  }

  //
  // LED role.  Receive the state of all buttons, and reflect that in the LEDs
  //

  if ( role == role_led )
  {
    // if there is data ready
    if ( radio.available() )
    {
      // Dump the payloads until we've gotten everything
      while (radio.available())
      {
        // Fetch the payload, and see if this was the last one.
        radio.read( button_states, num_button_pins );

        // Spew it
        printf("Got buttons\n\r");

        // For each button, if the button now on, then toggle the LED
        int i = num_led_pins;
        while(i--)
        {
          if ( button_states[i] )
          {
            led_states[i] ^= HIGH;
            digitalWrite(led_pins[i],led_states[i]);
          }
        }
      }
    }
  }
}
`

wmarkow, 2018-01-02:
What kind of **Arduino platform core** do you use? I was successfully used **nRF24L01** chip with this **RF24** library and **NodeMCU 1.0 (ESP-12E Module)**. I'm using [this ESP8266](http://arduino.esp8266.com/versions/2.4.0/package_esp8266com_index.json) as a Arduino NodeMCU platform.

By default you do not have to override or redefine the SPI pins. SPI is supported by default by ESP8266 and should work out of the box.
I see that this platform core has the following definition of SPI pins:

```
static const uint8_t SS    = 15;
static const uint8_t MOSI  = 13;
static const uint8_t MISO  = 12;
static const uint8_t SCK   = 14;
```
So you need to use these pins to connect your nRF24 chip. For **CE** and **CS** I was using pins **D4** and **D8**, so you can create RF24 like:

```
RF24 radio(D4, D8);
or
RF24 radio(2, 15);
```

Please post the output from the compiler so we could see what exactly is the error.

Watson221, 2018-01-02:
Ok,
I'm a bit of a novice when it comes to arduino stuff. I'm also using the ESP8266 Core by ESP8266 community. So if I understand you correctly, then all I have to do to fix my error is replace my define code with your static constu code. The thing I don't understand is how to control /identify what pins are set as MOSI, MISO, and SCK  

something like

int. MOSI =5
static const uint8_t MOSI  = 13;

That's the error code I got

`Arduino: 1.6.11 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\Watson\AppData\Local\Temp\arduino_modified_sketch_613833\led_remote.pde:9:0:

C:\Users\Watson\Documents\Arduino\libraries\RF24-master/printf.h: In function 'void printf_begin()':

C:\Users\Watson\Documents\Arduino\libraries\RF24-master/printf.h:31:29: error: 'fdevopen' was not declared in this scope

   fdevopen( &serial_putc, 0 );

                             ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Avamander, 2018-01-02:
Don't include printf.h in sketches (or sketch folders) when you're compiling for the ESP8266.

Watson221, 2018-01-02:
So does the static const code define the pin for MOSI, SCK, and MISO?
like
static const uint8_t MOSI = 8;

Avamander, 2018-01-02:
The error you're having is trying to compile AVR Arduino-only code for your ESP. Don't.

Watson221, 2018-01-02:
Now I get a printf error

wmarkow could I get the code you used for the NodeMCU
The led remote example doesn't seem to be compatible with the NodeMCU.

Arduino: 1.6.11 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\Watson\Documents\Arduino\led_remote\led_remote.ino: In function 'void setup()':

led_remote:90: error: 'printf_begin' was not declared in this scope

   printf_begin();

                ^

exit status 1
'printf_begin' was not declared in this scope

Avamander, 2018-01-02:
Don't call the function.

Watson221, 2018-01-02:
So I should completely get rid of the printf function and all of its branches?

Avamander, 2018-01-02:
Yes.

wmarkow, 2018-01-02:
>So does the static const code define the pin?
static const uint8_t MOSI = 8;

@Watson221, I just copied the static definitions from the Arduino esp8266 core to show you how they define pins for SPI. The defines are already done in the Arduino platform core provided by esp8266 team. When you choose in **Arduino IDE** a proper platform (esp8266) and a proper board (NodeMCU) then everything should work automatically. You can take a look how the esp8266 team define the pins for NodeMCU.; the entry point is here on their [pins_arduino.h github](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h). If you take a look you will notice that they use a [generic/common.h file](https://github.com/esp8266/Arduino/blob/master/variants/generic/common.h). SPI defines are already there. 

>led_remote:90: error: 'printf_begin' was not declared in this scope
   printf_begin();

As @avamander suggested, just remove from your code (or comment it out) the methods that uses printing (like printf_begin()).

Watson221, 2018-01-02:
So how can I see which pins are defined to do what in the Arduino Core?

In order to control the the pins do I need to install that  pins_arduino.h github library?

Avamander, 2018-01-02:
@Watson221 You don't control the pins. If you wish you can bitbang SPI, but that's an entirely other topic. It isn't a library _per se_, pins_arduino.h is a file that contains the pin definitions of the Arduino ESP8266 core you must have to compile Arduino libraries for ESP8266.

Watson221, 2018-01-02:
So the pins_arduino.h tells me how the pins translate from arduino to NodeMCU?

Watson221, 2018-01-02:
That code compiles just fine

```
#define D6 13  // SCK
#define D7 12  // MISO
#define D8 11  // MOSI

#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"

//
// Hardware configuration
//

// Set up nRF24L01 radio on SPI bus plus pins 9 & 10 (CE & CS)

RF24 radio(3,2);

// sets the role of this unit in hardware.  Connect to GND to be the 'led' board receiver
// Leave open to be the 'remote' transmitter
const int role_pin = A0;

// Pins on the remote for buttons
const uint8_t button_pins[] = { 8,9 };
const uint8_t num_button_pins = sizeof(button_pins);

// Pins on the LED board for LED's
const uint8_t led_pins[] = { 4,5 };
const uint8_t num_led_pins = sizeof(led_pins);

//
// Topology
//

// Single radio pipe address for the 2 nodes to communicate.
const uint64_t pipe = 0xE8E8F0F0E1LL;

//
// Role management
//
// Set up role.  This sketch uses the same software for all the nodes in this
// system.  Doing so greatly simplifies testing.  The hardware itself specifies
// which node it is.
//
// This is done through the role_pin
//

// The various roles supported by this sketch
typedef enum { role_remote = 1, role_led } role_e;

// The debug-friendly names of those roles
const char* role_friendly_name[] = { "invalid", "Remote", "LED Board"};

// The role of the current running sketch
role_e role;

//
// Payload
//

uint8_t button_states[num_button_pins];
uint8_t led_states[num_led_pins];

//
// Setup
//

void setup(void)
{
  //
  // Role
  //

  // set up the role pin
  pinMode(role_pin, INPUT);
  digitalWrite(role_pin,HIGH);
  delay(20); // Just to get a solid reading on the role pin

  // read the address pin, establish our role
  if ( digitalRead(role_pin) )
    role = role_remote;
  else
    role = role_led;

  //
  // Print preamble
  //

  Serial.begin(115200);

  //
  // Setup and configure rf radio
  //

  radio.begin();

  //
  // Open pipes to other nodes for communication
  //

  // This simple sketch opens a single pipes for these two nodes to communicate
  // back and forth.  One listens on it, the other talks to it.

  if ( role == role_remote )
  {
    radio.openWritingPipe(pipe);
  }
  else
  {
    radio.openReadingPipe(1,pipe);
  }

  //
  // Start listening
  //

  if ( role == role_led )
    radio.startListening();

  //
  // Dump the configuration of the rf unit for debugging
  //

  radio.printDetails();

  //
  // Set up buttons / LED's
  //

  // Set pull-up resistors for all buttons
  if ( role == role_remote )
  {
    int i = num_button_pins;
    while(i--)
    {
      pinMode(button_pins[i],INPUT);
      digitalWrite(button_pins[i],HIGH);
    }
  }

  // Turn LED's ON until we start getting keys
  if ( role == role_led )
  {
    int i = num_led_pins;
    while(i--)
    {
      pinMode(led_pins[i],OUTPUT);
      led_states[i] = HIGH;
      digitalWrite(led_pins[i],led_states[i]);
    }
  }

}

//
// Loop
//

void loop(void)
{
  //
  // Remote role.  If the state of any button has changed, send the whole state of
  // all buttons.
  //

  if ( role == role_remote )
  {
    // Get the current state of buttons, and
    // Test if the current state is different from the last state we sent
    int i = num_button_pins;
    bool different = false;
    while(i--)
    {
      uint8_t state = ! digitalRead(button_pins[i]);
      if ( state != button_states[i] )
      {
        different = true;
        button_states[i] = state;
      }
    }

    // Send the state of the buttons to the LED board
    if ( different )
    {
      printf("Now sending...");
      bool ok = radio.write( button_states, num_button_pins );
      if (ok)
        printf("ok\n\r");
      else
        printf("failed\n\r");
    }

    // Try again in a short while
    delay(20);
  }

  //
  // LED role.  Receive the state of all buttons, and reflect that in the LEDs
  //

  if ( role == role_led )
  {
    // if there is data ready
    if ( radio.available() )
    {
      // Dump the payloads until we've gotten everything
      while (radio.available())
      {
        // Fetch the payload, and see if this was the last one.
        radio.read( button_states, num_button_pins );

        // Spew it
        printf("Got buttons\n\r");

        // For each button, if the button now on, then toggle the LED
        int i = num_led_pins;
        while(i--)
        {
          if ( button_states[i] )
          {
            led_states[i] ^= HIGH;
            digitalWrite(led_pins[i],led_states[i]);
          }
        }
      }
    }
  }
}
````

Avamander, 2018-01-02:
> So the pins_arduino.h tells me how the pins translate from arduino to NodeMCU?

Pretty much.

Watson221, 2018-01-03:
I finally more or less understand how to make this library work!
Thanks guys for all the help!

Watson221, 2018-01-11:
Hey guys,
I'm having some problems again. I couldn't figure out how to use the pins_arduino.h library. I'm really struggling getting these 2 rf modules to talk to each other. I've tried tons of different examples from all over the internet and this version is the one I could actually understand. Thanks for helping!

This is the code I have for the NodeMCU
```
#define D6 13  // SCK
#define D7 12  // MISO
#define D8 11  // MOSI

#include "nRF24L01.h"
#include "RF24.h"
#include "SPI.h"
byte  data [ 1 ] ;
boolean  var ;

int LEDup = 4;
int LEDdown = 5;

RF24 radio(3, 2); // CE,CS
const uint64_t pipe = 0xE6E6E6E6E6E6;

void setup(void) {

  radio.begin();
  radio.openReadingPipe(1, pipe);
  radio.startListening();

  pinMode(LEDup, OUTPUT);
  pinMode(LEDdown, OUTPUT);

  Serial.begin(9600);
  delay(100);
}

void loop(void) {
  if ( radio . available ( ) ) {
    radio . read ( data ,  1 ) ;
    if ( data [ 0 ]  ==  0 ) {
      digitalWrite ( LEDdown ,  LOW ) ;
      digitalWrite ( LEDup , LOW ) ;
    }
    if ( data [ 0 ]  ==  1 ) {
      digitalWrite ( LEDup ,  HIGH ) ;
    }
    if ( data [ 0 ]  ==  2 ) {
      digitalWrite ( LEDdown ,  HIGH ) ;
    }
  }
}
```

And for the Arduino
```
#include <nRF24L01.h>
#include <RF24.h>
#include <RF24_config.h>

boolean powerState  = LOW;
int photocell = 0;
int upDetect = 7;
int downDetect = 9;
int manualUp = 10;
int manualDown = 3;
int Relay = 6;
int upswitchReading;
int downswitchReading;
int photocellReading;
int manualUpReading;
int manualDownReading;
byte data [1];

RF24 radio(2, 4);
const uint64_t pipe = 0xE6E6E6E6E6E6; //address

void setup(void) {
  pinMode (upDetect, INPUT_PULLUP);
  pinMode (downDetect, INPUT_PULLUP);

  pinMode (manualDown, INPUT_PULLUP);
  pinMode (manualUp, INPUT_PULLUP);
  pinMode (Relay, OUTPUT);
  pinMode (5, OUTPUT);
  digitalWrite (5, HIGH);

  radio.begin();
  radio.openWritingPipe(pipe);

  Serial.begin(9600);
  delay(100);

}

void loop(void) {
  Serial.println("Up Switch Reading = ");
  upswitchReading = digitalRead(upDetect);
  Serial.println(upswitchReading, DEC);
  delay(1000);

  Serial.println("Down Switch Reading = ");
  downswitchReading = digitalRead(downDetect);
  Serial.println(downswitchReading, DEC);
  delay(1000);

  Serial.println("Manual Down Switch = ");
  manualDownReading = digitalRead(manualDown);
  Serial.println(manualDownReading, DEC);
  delay(1000);

  Serial.println("Manual UP Switch = ");
  manualUpReading = digitalRead(manualUp);
  Serial.println(manualUpReading, DEC);
  delay(1000);

  Serial.println("  Photocell reading = ");
  photocellReading = analogRead(photocell);
  Serial.println(photocellReading);
  delay(1000);

 //Open/Close

if (photocellReading < 400) {
  powerState = HIGH;
}
else if (photocellReading > 400) {
  powerState = LOW;
}
if (manualDownReading == 0) {
  powerState = HIGH;
}
else if (manualUpReading == 0) {
  powerState = LOW;
}

digitalWrite(Relay, powerState);

  //Radio TX

  if ( digitalRead ( 7 ) == LOW  ||  digitalRead ( 9 ) == LOW ) {
    data [ 0 ] = 0 ;
  }
  if ( digitalRead ( 7 ) == HIGH ) {
    data [ 0 ] = 1 ;
  }
  if ( digitalRead ( 9 ) == HIGH ) {
    data [ 0 ] = 2 ;
  }
  radio . write ( data ,  1 ) ;
}
```

Avamander, 2018-01-11:
> I've tried tons of different examples from all over the internet and this version is the one I could actually understand. 

What part about the default unmodified examples is unclear? Maybe I can improve them.

closed 2018-01-11 by Avamander

darrahts, 2019-02-07:
> Don't include printf.h in sketches (or in sketch folders) when you're compiling for the ESP8266.

This worked for me. I know its closed but incase others come here