nRF24/RF24 #400: Nodemcu + nRF24L01 Crash (isolated problem)

JLuisAlcaraz, 2018-01-02:
Hi all!
I isolated the problem from lib NR24 crash. In order to do things easy, I connect the Nodemcu V3 Lolin to USB, nothing more, no nRF24L01 board , just the Nodemcu. I download the RF24 lib ([https://github.com/nRF24/RF24](url) and load the example sketch "Getting started". Then compile and load on the board. When I press 'T' in order to start emitting, the program crashes.

Now,if I open RF24_config.h and look for "#if defined (ARDUINO_ARCH_ESP8266)" (line 134), and change next 2 lines:

  #include <pgmspace.h>
  #define PRIPSTR "%s"

By these ones (from RF24_config.h of 1.3.0 version):

  #define PSTR(x) (x)
  #define printf Serial.printf
  #define sprintf(...) os_sprintf( __VA_ARGS__ )
  #define printf_P printf
  #define strlen_P strlen  
  #define PROGMEM
  #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
  #define pgm_read_word(p) (*(p))
  #define PRIPSTR "%s"

Then compile and load on board. The program dont crash.  The problem is enable PROGMEM, I think.

Can someone do this test and share results (if crash or not)?

Avamander, 2018-01-02:
The library has been tested to work without any additional modifications. You have to fix your IDE/development environment to compile the library for the ESP properly. Progmem support has been ported to the ESP and your ESP crashing on using it clearly indicates your environment being extremely out-of-date or again, broken.

Duplicate of: https://github.com/nRF24/RF24/issues/398

closed 2018-01-02 by Avamander

JLuisAlcaraz, 2018-01-02:
I did a fresh install od arduino IDE 1.8.5, and add the ESP8266 boards.
Please, can You explain me what I need to do?

Avamander, 2018-01-02:
Provide information about what ESP core are you using and wipe all of your modifications to any of the libraries you've made.

JLuisAlcaraz, 2018-01-02:
Thank you very much, I think I see the problem. The ESP core was 2.3.0, and I see there is a 2.4.0. I installed the 2.4.0 core and now it works properly!
Thanks again :)