nRF24/RF24 #401: There is an issue about the "toggle_feature()" after mcu reset

WindyYam, 2018-01-04:
The toggle_feature() in the init must be called when the feature is "toggled off" otherwise it get turned off. This happens when I uses a bk2425 chip and press the arduino reset button, the toggle_feature runs second time and it's turned off in my case it blocks in write forever.

according to datasheet when feature is toggled off the register "FEATURE" could not be written as anything written and then read will get 0. so there should be something like this:

//test if feature activate is toggled
  write_register(FEATURE, 0x01);
  if(!read_register(FEATURE))
	  toggle_features();

Avamander, 2018-01-11:
Usually people don't reset their MCU without resetting the peripherals. Have you thought about doing that instead?

WindyYam, 2018-01-12:
> Usually people don't reset their MCU without resetting the peripherals. Have you thought about doing that instead?

For what I googled there is no way to reset nrf24l01 other than manage it's power supply. So to keep "rf24.begin" behaves the same after every reset the "toggle_feature()"  should be taken into consideration .

wmarkow, 2018-08-15:
I think that calling ```begin()``` should always push the RF24 chip (or a set of its registers) into the same state.

>Usually people don't reset their MCU without resetting the peripherals.

@Avamander , from one point of view I could say that calling ```rf24.begin()`` should programmatically reset RF24 chip. We could call that a **software reset**.

wmarkow, 2018-08-16:
Another question is: "what ```toggle_features()``` actually does? It is defined as:
```
void RF24::toggle_features(void)
{
    beginTransaction();
	_SPI.transfer( ACTIVATE );
    _SPI.transfer( 0x73 );
	endTransaction();
}
```
where variable ```ACTIVATE``` is defined as:
```
#define ACTIVATE      0x50
```

If I understand correctly, this method tries to execute ```ACTIVATE``` command with ```0x73``` as a parameter. In [this RF24 datasheet](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf) on the page 48 (Section 8.3.1 SPI Commands) there is no such a command which **command word** has value ```0x50``` (such a command doesn't exist). Or am I missing something? So what actually ```toggle_features()``` does?

wmarkow, 2018-08-16:
I see that ```toggle_features()``` has been introduce by commit 79628e7849 in 2011.

WindyYam, 2018-08-16:
You can see the ACTIVATE command on [nRF24L01 Product Specification V2 datasheet](https://www.nordicsemi.com/eng/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf) Page 46, it's basically a switch for some features

wmarkow, 2018-08-16:
Thanks, this link is more helpful :)

This command complicates a bit the logic. We can not track the value of that command in local variable to cache if the features are active or not, because after CPU reset (when RF24 is still powered up) we will loose that value anyway.
We need some kind of sophisticated mechanism to find out in runtime if those feature are active or not. Moreover I would get rid of this ```toggle_features()``` method (it is private so we can do it without changing the API) and introduce a new called ```activate_features()```; it is more descriptive and we have sure that - after we call it - the features will be always activated.

WindyYam, 2018-08-16:
There is actually a trick to show that, from datasheet it says "The R_RX_PL_WID, W_ACK_PAYLOAD, and 
W_TX_PAYLOAD_NOACK features registers are 
initially in a deactivated state; a write has no 
effect, a read only results in zeros on MISO"
so if ACTIVATE is off those register bit can not be modified and read always get zero. so the trick is

//test if feature activate is toggled
write_register(FEATURE, 0x01);
if(!read_register(FEATURE))
toggle_features();

this will ensure ACTIVATE is on

wmarkow, 2018-08-16:
@WindyYam, in my previous comment I have forgot to mention your proposed fix from your first post.
We write 0x01 to FEATURE register and read back its value:
 * if the read value is 0x01 then the features are already activated
 * if the read value is 0x00 then the features are deactivated and we need to activate them 

The side effect of this is that the method will activate the features and by the way will enable **EN_DYN_ACK** as well.

wmarkow, 2018-08-16:
@WindyYam, I have created a commit in my **fix_for_#401** branch of my forked version of RF24 library. Could you please review and retest?

wmarkow, 2018-08-18:
I wanted to retest the solution proposed by @WindyYam. I'm a bit surprised because I'm not able to reproduce the original issue. It looks like on my nRF24L01 chips the **features** are always activated and ```toggle_features()``` does nothing. Resetting the Arduino board has no impact - it always works. I even commented out this ```toggle_features()``` in ```RF24.begin()``` and the **dynamic payload length** and **payload with ACK** seem to work correctly. The ```RF24printDetails``` method after my setup always shows something like this:
```
DYNPD/FEATURE	 = 0x3f 0x06
```
so the **features** seem to be enabled even if ```toggle_features()``` has been commented out.

Maybe my nRF24L01 chips confirm to the [Product Specification V1](https://components101.com/sites/default/files/component_datasheet/nRF24L01%20Datasheet.pdf)? According to V1 specs, the **FEATURE** register is accessible always (page 63 - actually there is no information that I need to unlock access to that register) and there is no need to send **ACTIVATE** command. V1 specs knows nothing about **ACTIVATE** command.
However [nRF24L01 Product Specification V2 datasheet](https://www.nordicsemi.com/eng/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf) knows about **ACTIVATE** command (page 46) and on page 58 there is annotation on **FEATURE** register, that **ACTIVATE** command must be send to unlock access to that register.

Or maybe my devices are a cheap clones of the original device?

Very interesting.

WindyYam, 2018-08-19:
@wmarkow I dont have genuine nrf24 modules, I have bunch of "nrf24l01" that turn out some of them to be bk2425, others most probably si24. The issue happens when I'm using the no-ack feature(enableDynamicAck(false) and write with the no ack flag)

To be better acceptable maybe add function like "bool isFeatureActivated()" ?

WindyYam, 2018-08-20:
@wmarkow I think that has gone way too complex. Just read the register before write 0x01, and then read
1. if read is 0, the written 0x01 would do nothing(see datasheet, write has no effect), and we know it's deactivated;
2. if read is something else, write back its value before, and we know it's activated;

`bool isFeatureActivated(){
    uint8_t features = read_register(FEATURE);
    write_register(FEATURE, 0x01);
    if(!read_register(FEATURE)){   //got zero, deactivated
        return FALSE;
    }else{  //activated
        write_register(FEATURE, features);
        return TRUE;
    }
}`

wmarkow, 2018-08-20:
@WindyYam, you are right. My solution is too way complex. Yours is much simpler. Now we have an example how simple things can be over complicated :)

TMRh20, 2020-08-11:
Bit of an old issue, but I've taken a closer look at it and unfortunately I'm probably more confused now. I made the toggle_features function public, and toggled it manually to see what was going on.

1. Essentially, the toggle_features() function seems to have no real effect on my nrf24l01+ radios
2. Removing the function call from begin() _mostly_ makes no difference. I can disable/re-enable ack payloads and DPL at will.
3. I was able to get the radio into a state where upon reset, calling begin failed to reset the FEATURE register to 0, but its been difficult to recreate consistently. Writing anything but 0 always seems to work.

So I'm thinking the makers of these devices decided to ditch the ACTIVATE command and just make things writable. I don't have any bk2425 or SI24R1 to test with though.

So I guess the toggle_features command should probably be left as-is for the nrfs, since there is inconsistent behaviour without it, and adding code to the begin() function to support an obscure device is out of the question. It may be possible to add another function or #define that enables the full functionality per https://github.com/nRF24/RF24/issues/401#issuecomment-414181150

This is the one and only time I've ever heard reference to the bk2425 but I am a bit familiar with the SI24R1s, so I'd like to know how they behave in regard to the ACTIVATE command prior to making changes. If anybody with SI24R1s would like to test this out, I would mainly like to know the following:
1. What happens when toggle_features is removed from begin()? Does the FEATURES register get reset to 0? Is it still writable after?
2. What happens when toggle_features is called during runtime? Any change in enabling ACK payloads or DPL?

2bndy5, 2020-10-20:
@Avamander 
>Usually people don't reset their MCU without resetting the peripherals.

This simply isn't true. The MCUs' reset buttons don't typically affect the board's regulators. Most often the reset button focuses only on the board's CPU's enable pin (which does not invoke a power-on-reset condition in the nRF24L01). Please correct me if I misunderstood this comment.

@TMRh20 This is why adjusting the `RF24::begin()` is the *only* course of action to address this issue (for any type module connected -- including non-plus variants of NRF24L01).

@wmarkow I like your solution, but adding another function (even if only private) that is only called once is a bit excessive.

I've had to address the un-intuitive manner of the reset button on my CircuitPython library. You can see how I've addressed this issue in this library (on my fork) with the 2 above mentioned commits (beware the second commit fixes faulty logic in the first). Subsequently, I added a `_is_p_variant` private boolean to expedite additional functionality for #640 (another issue related specifically to the non-plus nRF24L01).

Avamander, 2020-10-20:
> This simply isn't true. The MCUs' reset buttons don't typically affect the board's regulators. Most often the reset button focuses only on the board's CPU's enable pin (which does not invoke a power-on-reset condition in the nRF24L01). Please correct me if I misunderstood this comment.

@2bndy5 The meaning was that it's a weird setup that doesn't run initialization/reset code after a reset.

2bndy5, 2020-10-20:
Ah, thanks for that clarification. I agree, it is weird that RF24 doesn't brute force a soft reset on init, but that would require dumping a whole lot more info to the nRF24L01 registers. I'm afraid @TMRh20 might err against this behavior as it seems to compromise the "optimized" description of this repo. BTW I did [employ a soft reset for my CircuitPython library](https://github.com/2bndy5/CircuitPython_nRF24L01/blob/70082fe08b1e798148af3e3e9a576d9ba85dfb93/circuitpython_nrf24l01/rf24.py#L52-L135), but I captured the RX addresses during the lib's init (related to #496), instead of resetting them.

2bndy5, 2020-12-14:
resolved by #691

feels good to close this one 😄

closed 2020-12-14 by 2bndy5