nRF24/RF24 #403: Help making sense of the library NodeMCU

Watson221, 2018-01-11:
Hey guys,
I'm having some problems again. I couldn't figure out how to use the pins_arduino.h library. I'm really struggling getting these 2 rf modules to talk to each other. I've tried tons of different examples from all over the internet and this version is the one I could actually understand. Thanks for helping!

This is the code I have for the NodeMCU
```
#define D6 13  // SCK
#define D7 12  // MISO
#define D8 11  // MOSI

#include "nRF24L01.h"
#include "RF24.h"
#include "SPI.h"
byte  data [ 1 ] ;
boolean  var ;

int LEDup = 4;
int LEDdown = 5;

RF24 radio(3, 2); // CE,CS
const uint64_t pipe = 0xE6E6E6E6E6E6;

void setup(void) {

  radio.begin();
  radio.openReadingPipe(1, pipe);
  radio.startListening();

  pinMode(LEDup, OUTPUT);
  pinMode(LEDdown, OUTPUT);

  Serial.begin(9600);
  delay(100);
}

void loop(void) {
  if ( radio . available ( ) ) {
    radio . read ( data ,  1 ) ;
    if ( data [ 0 ]  ==  0 ) {
      digitalWrite ( LEDdown ,  LOW ) ;
      digitalWrite ( LEDup , LOW ) ;
    }
    if ( data [ 0 ]  ==  1 ) {
      digitalWrite ( LEDup ,  HIGH ) ;
    }
    if ( data [ 0 ]  ==  2 ) {
      digitalWrite ( LEDdown ,  HIGH ) ;
    }
  }
}
```

And for the Arduino
```
#include <nRF24L01.h>
#include <RF24.h>
#include <RF24_config.h>

boolean powerState  = LOW;
int photocell = 0;
int upDetect = 7;
int downDetect = 9;
int manualUp = 10;
int manualDown = 3;
int Relay = 6;
int upswitchReading;
int downswitchReading;
int photocellReading;
int manualUpReading;
int manualDownReading;
byte data [1];

RF24 radio(2, 4);
const uint64_t pipe = 0xE6E6E6E6E6E6; //address

void setup(void) {
  pinMode (upDetect, INPUT_PULLUP);
  pinMode (downDetect, INPUT_PULLUP);

  pinMode (manualDown, INPUT_PULLUP);
  pinMode (manualUp, INPUT_PULLUP);
  pinMode (Relay, OUTPUT);
  pinMode (5, OUTPUT);
  digitalWrite (5, HIGH);

  radio.begin();
  radio.openWritingPipe(pipe);

  Serial.begin(9600);
  delay(100);

}

void loop(void) {
  Serial.println("Up Switch Reading = ");
  upswitchReading = digitalRead(upDetect);
  Serial.println(upswitchReading, DEC);
  delay(1000);

  Serial.println("Down Switch Reading = ");
  downswitchReading = digitalRead(downDetect);
  Serial.println(downswitchReading, DEC);
  delay(1000);

  Serial.println("Manual Down Switch = ");
  manualDownReading = digitalRead(manualDown);
  Serial.println(manualDownReading, DEC);
  delay(1000);

  Serial.println("Manual UP Switch = ");
  manualUpReading = digitalRead(manualUp);
  Serial.println(manualUpReading, DEC);
  delay(1000);

  Serial.println("  Photocell reading = ");
  photocellReading = analogRead(photocell);
  Serial.println(photocellReading);
  delay(1000);

 //Open/Close

if (photocellReading < 400) {
  powerState = HIGH;
}
else if (photocellReading > 400) {
  powerState = LOW;
}
if (manualDownReading == 0) {
  powerState = HIGH;
}
else if (manualUpReading == 0) {
  powerState = LOW;
}

digitalWrite(Relay, powerState);

  //Radio TX

  if ( digitalRead ( 7 ) == LOW  ||  digitalRead ( 9 ) == LOW ) {
    data [ 0 ] = 0 ;
  }
  if ( digitalRead ( 7 ) == HIGH ) {
    data [ 0 ] = 1 ;
  }
  if ( digitalRead ( 9 ) == HIGH ) {
    data [ 0 ] = 2 ;
  }
  radio . write ( data ,  1 ) ;
}
```

Avamander, 2018-01-11:
Is your issue https://github.com/nRF24/RF24/issues/399 solved? And please describe the symptoms of it "not working".

Watson221, 2018-01-11:
The 2 modules will not communicate. The NodeMCU won't receive anything from the Arduino, not matter what example code (the ones included in the library or otherwise). My conclusion is that the NodeMCU won't talk to the rf24l01 module. The code does compile, which was the issue in #399. The wiring on the Arduino is up to spec, I think the issue with the NodeMCU is in the both the wiring and/or code.

Avamander, 2018-01-11:
> I think the issue with the NodeMCU is in the both the wiring and/or code.

Maybe it's the code that specifies the wiring are you 100% sure you're using ESP8266's SPI pins and the right CE and CS (defined in the code)?

> The wiring on the Arduino is up to spec

Might worth checking again. At the bottom of this page there's a table that contains the common values http://tmrh20.github.io/RF24/ .

> The NodeMCU

Just saying, NodeMCU is the firmware running on the ESP8266, not the device itself.

Watson221, 2018-01-11:
I just verified the pins the the NodeMCU again and they match the pins in the code. In that link you gave me the AT tiny pins match the pins that I have on the NodeMCU, so can I use those pins instead? And would I have to alter the code in order to do that?

Watson221, 2018-01-11:
What part about the default unmodified examples is unclear? Maybe I can improve them.

For the getting started example I found the setup easy to follow. I just got lost in the loop. I didn't know what it was sending and where that sending code was ( so I could modify it) . I think It would be easier to understand if the transceiver and receiver code were placed in separate examples.

Avamander, 2018-01-11:
That is not right at all. ESP8266 SPI SCK is D5, MOSI is D7 and MISO is D6, CE and CS you can pick yourself. Also just use the built-in examples, the ones floating around on the internet aren't usually updated at all, most of the built-in examples have been confirmed working on all these platforms. 

See this line https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino#L16 for CE and CS pin definition.

>  didn't know what it was sending and where that sending code was ( so I could modify it) .

I recommend you learn how pointers work then.

> I think It would be easier to understand if the transceiver and receiver code were placed in separate examples.

They sending and receiving parts are well described by the comments and are fully modular, meaning there's only the effort of understanding that code needed which you would need to do still when they were in different files.

Watson221, 2018-01-11:
Ok thanks for the advice! So is the code that I have right now work with your library?

Watson221, 2018-01-11:
Sorry for the typo. So does the code that I have right now work with your library?*

Avamander, 2018-01-11:
Possibly, but I can't be fully certain. Using built-ins is better for that exact reason.

Watson221, 2018-01-11:
Ok I'll give it a test run and see if it works.

Watson221, 2018-01-11:
My code doesn't seem to be doing anything after I rewired the NodeMCU.

Watson221, 2018-01-11:
I'm going to try it with the LED remote example

wmarkow, 2018-01-11:
In your setup() method call radio.isChipConnected() and inspect what it is returning. If you have **false** then you probably have some hardware issues. Do like this:

```
void setup(void) {
  Serial.begin(9600);
  delay(100);

  radio.begin();
  bool result = radio.isChipConnected();

  Serial.println(result);
}
``` 

You can do this for Arduino and for NodeMCU. Then you will see what board has no connectivity to nRF24 chip.

Watson221, 2018-01-11:
The LED Example didn't work. I got a reading of true from the Arduino and no reading from the NodeMCU, as in nothing is getting printed

Avamander, 2018-01-11:
@Watson221 So you rewired the module, right? Picked two pins for CE and CS and set them in the code, right? Also make sure you fully read the example and make sure the radionumber is right, both can't have the same radio number https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino#L13 .

Watson221, 2018-01-11:
Yes I triple checked the ce and cs in hardware and in the code for both the NodeMCU and the Arduino.

Watson221, 2018-01-11:
I also made sure the radio number was different

Avamander, 2018-01-11:
So after that you uploaded the example code accordingly to both modules and only the Arduino gives any output?

Watson221, 2018-01-11:
Yes, now after unplugging the NodeMCU and replugging it in I'm getting a lot of symbols.

Avamander, 2018-01-11:
Make sure you have set your serial terminal to 115200 baud.

Watson221, 2018-01-11:
I do have it set to 115200

Avamander, 2018-01-11:
Then there's something else totally broken in your setup.

Watson221, 2018-01-11:
It now reads this
 
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
0
STATUS		 = 0x3ffef380 RX_DR=3ffef370 TX_DS=4 MAX_RT=0 RX_P_NO=4022f90e TX_FULL=3ffef380
���?	 = 0x3ffef3603ffef3603ffef3603ffef3603ffef360 0x3ffef3603ffef3603ffef3603ffef3603ffef360
��?	 = 0x3ffef380 0x3ffef380 0x3ffef380 0x3ffef380
��?	 = 0x3ffef3603ffef3603ffef3603ffef3603ffef360
��?	 = 0x3ffef380 0x3ffef380 0x3ffef380 0x3ffef380 0x3ffef380 0x3ffef380
��?	 = 0x3ffef380
��?	 = 0x3ffef380
��?	 = 0x3ffef380
��?	 = 0x3ffef380
��?	 = 0x3ffef380
��?	 = 0x3ffef380 0x3ffef380
Data Rate	 = ��?
Model		 = ��?
CRC Length	 = ��?
PA Power	 = ��?

Watson221, 2018-01-11:
I had the wrong terminal opened! That's what I got from the NodeMCU

wmarkow, 2018-01-11:
It may be some hardware issues with nRF24 chip interface to NodeMCU and radio.begin() stucks. Please print something to serial before radio.begin():

```
void setup(void) 
{
  Serial.begin(9600);
  delay(100);

  Serial.prinltn("Test");
  radio.begin();
  bool result = radio.isChipConnected();

  Serial.println(result);
}
```

And @Watson221, one more question: what kind on NodeMCU board you have? v0.9 or E12? Please type in google **nodemcu pinout** and in the results switch to **Images** tab and find your board. Then you will see what pins are used for SPI.

Avamander, 2018-01-11:
Don't connect CE and CS to reserved pins (GPIO 0, 2 or 15) on your ESP8266.

Watson221, 2018-01-11:
https://circuits4you.com/wp-content/uploads/2017/12/nodemcu-pinout.png

That's my NodeMCU I think it's v.09 but I'm not sure. GIO translates to pin 3, which I'm using for CE. I've switched it over to 1.

Watson221, 2018-01-11:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�

Watson221, 2018-01-11:
I get that now

Watson221, 2018-01-11:
Do I hook up the mosi, miso, ect.  pins on that diagram to the rf module?

Avamander, 2018-01-11:
@Watson221 
Have you not already? I told you the same pins the diagram contains. As the diagram shows, SPI SCK is D5, MOSI is D7 and MISO is D6.

Check out https://github.com/esp8266/Arduino/issues/2414 and https://forum.arduino.cc/index.php?topic=408876.0 for more information about that boot mode and possible causes.

Please keep in mind that board is actually called ESP(8266)-12E the software on it is called NodeMCU.

Watson221, 2018-01-11:
Yes I have the ESP(8266)-12E connected to the hmosi, hsclk, etc.

Watson221, 2018-01-11:
Thanks for the information on proper terminology

Watson221, 2018-01-11:
You know I don't think all this hassle is worth it! I think I'm just going to order another Arduino uno and use that instead! This is crazy!

wmarkow, 2018-01-12:
>Please keep in mind that board is actually called ESP(8266)-12E the firmware on it is called NodeMCU, it's just easier to understand you when you use the correct terms.

@Avamander , I know that the main chip on those boards is ESP8266, but I always thought that those boards are called **NodeMCU**. That's probably because in Arduino IDE you can choose the following boards from the **Tools -> Board** menu:
 * NodeMCU 0.9 (ESP-12 Module)
 * NodeMCU 1.0 (ESP-12E Module)

closed 2019-05-10 by TMRh20