nRF24/RF24 #407: RPI double RF24 in same bus not working

ludiazv, 2018-01-18:
Hi,
   I tried connect two RF24 on the same bus (SPI1) on a raspberry PI using CS0 & CS1 and dedicated GPIO pins for CE control. No IRQ connected.

  I try to run two complementary gettingstarted  on for sending the other for receiving. When I run the processes the printDetails() function show valid configuration on both devices. However, after start ping and pong_back but after a few failed attempts of sending the transmitter is blocked (no msg show but simply stop iterating). On the receiver side, only random payload is received. 
Wiring is correct and power supply is good with 2 caps on each radio. They are able to transmit at the same time without any block. In addition, the PA_MIN is selected to minimize power impact on the test.

I tried also use SPIDEV version with similar results.

Is actually possible to run 2 radios on the same bus?

Avamander, 2018-01-18:
Should be, the devices respect SS (people are using Ethernet modules on the same bus for ex.). Do you have an oscilloscope to verify that the rPI isn't pulling two SS's up at the same time? Also can you please make sure it's not your code trying to communicate with both devices simultaneously on the same bus?

ludiazv, 2018-01-19:
I will check with  a logic analyzer whats is happening. My guess is that at some point there is conflict.
Thanx

ludiazv, 2018-01-22:
After the verification with the logic analyzer it seems that is not a SS line conflict but somehow a dead lock in the mutex used by the RpI library.  I do not why but it seems not working well for inter-process sync. There is a lot fo controversy about using static variables on Dynamic libraries. Maybe changing the mutex attributes will solve the problem.

With SPIDEV **it worked** fine but reboot is needed after rebuild of the library with SPIDEV option. So I would recommend to used SPIDEV in this scenario.

Some addtional comments that could be useful:
SPI SPEED is not selectable with RPi driver:
``
_spi_speed variable is not used in trasaction init. 

see:

  inline void RF24::beginTransaction() {
    #if defined (RF24_SPI_TRANSACTIONS)
    _SPI.beginTransaction(SPISettings(**RF24_SPI_SPEED,** MSBFIRST, SPI_MODE0));
	#endif
    csn(LOW);
  }
``

For SPIDEV driver:

- Abuse of perror + abort(): Abort will stop the running process, this is not good practice for a library. A library should report the error to running process so it can be handled.  As the library is C++ perror+abort should be subtitured by custom exception that can be cached by library user.

- Selectable SPI Speed: Original radios are designed for 10Mhz. SPIDEV on modern Pis will setup the SPI speed of 12Mhz with default installation. This could cause problems specially if long traces or cables are used. With clones this could be worse. A simple modification of the library could be made to select the SPISpeed on linux systems.

What do you think?

Regards,

Avamander, 2018-01-22:
@ludiazv Unfortunately this is getting out of my zone-of-comfort as I do not own a development ARM board. Someone else needs to chime on how to resolve these issues permanently.

ludiazv, 2018-01-24:
I would be able to send a PR for SPIDEV improvement. For RPi SPI driver It would have to wait util I have a available RPI for propper testing.

Regards,

closed 2020-03-29 by TMRh20