nRF24/RF24 #408: ESP8266 cannot send&receive but Arduino can with same code

Zephael86, 2018-01-20:
Hi Everybody,

I'm strugling to attach an nRF24 module to my ESP8266 (NodeMCU v3).
I would like to create a relay for nRF chips toward my IP network.

I have 2 devices. One Arduino that functions as a listener, and one ESP8266 mentioned above.
All nRF24 chips that I use for this project have a 400V 10uF Capacitor.
All chips are directly attached with jumper cables to the micro controller, and not through a breadboard (but they are not soldered).

_I also have some LED Lampions and a transmitter available from an old project. This is mentioned because as of today they are still working (same channel is used on all nRF24 chips) = Transmitter is still working._

Below code (after changing the CE, CSN ports; device role; packet structure) works on the new Arduino. It means, that the old Transmitter is able to blink the LED attached to it, therefore the correct channel, address and pipes are used (at least for the listener) in the new code.

The problem is, that the ESP8266 seemingly does not send or receive anything.

- nRF24 chips were switched back and fort, replaced multiple times. Even used ones without capacitor.
- I have 5 times checked the radio chip leg layout.
- I am not using a "timer" right now, because the micro controllers are not doing anything else
- NodeMCU V1.0 (ESP-12) is selected as ESP8266 board for compile
- Sender and Receiver are in range. Just 3-5 cm apart.
- printDetails() and isChipConnected() returns real data on ESP8266.

_By the way, printDetails() does not print anything on Arduino for me._

Serial output at start:
```
Arduino:
--------
nRF24 - WiFi Bridge started
printDetails: 
isChipConnected(): 1

ESP8266
-------
nRF24 - WiFi Bridge started
printDetails: 
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe24e4e8b8b 0xe24e4e8b8b
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe24e4e8b8b
RX_PW_P0-6	 = 0x08 0x08 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x5a
RF_SETUP	 = 0x03
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW
isChipConnected(): 1
```

I am asking your help here to find out what could be the problem, because I am out of ideas. Buy another ESP8226?

And here is the code what I am using right now both for Arduino and for ESP8266:

```

#include <SPI.h>             //SPI
#include <RF24.h>               // nRF24 Radio library
#include <nRF24L01.h>

//Arduino friendly names to the ESP8266 GPIO pins
#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)

bool isSender = 1;

//#########
//# nRF24 #
//#########
#define CE_PIN  4 /*ESP8266: 4 | Arduino: 7*/
#define CSN_PIN 15 /*ESP8266: 15 | Arduino: 8*/
RF24 radio(CE_PIN, CSN_PIN); // Create a Radio

// define a structure for the payload
typedef struct{     // bytes:
  uint8_t data1;    // 1
  uint16_t data2;   // 2
  uint32_t data3;   // 4
  // sum: 7 bytes out of 32 bytes
} strctPL;

// structures to hold radio packages
strctPL dataPacket;

// Create pipes for nRF24 communication
// NOTE: the "LL" at the end of the constant is "LongLong" type
//const uint64_t pipe = 0xE8E8F0F0E1LL; // Define the transmit pipe
const uint64_t pipes[2] = {/* Reserved for unicast */0, /* Broadcast address */0xe24e4e8b8bLL};

//Pin of led to blink
uint8_t ledtoblink;

void setup() {
  if (isSender == 1)  { /*ESP8266*/ ledtoblink = LED_BUILTIN; }
  else { /*Arduino extra led*/ ledtoblink = 10;}
    
  Serial.begin(115200);
  Serial.println("nRF24 - WiFi Bridge started");

  radio.begin();
  radio.setChannel(90);                     // Set Radio channel to 90. 2400 Mhz + 90x1Mhz = 2490 MHz.
  radio.setAddressWidth(5);                 // Use 5 bytes address width
  radio.setPALevel(RF24_PA_LOW);           // Set Power Amplifier (PA) level to -6dBM. (RF24_PA_MIN, RF24_PA_LOW, RF24_PA_HIGH, RF24_PA_MAX = -18dBm, -12dBm,-6dBM, and 0dBm)
  radio.setDataRate(RF24_1MBPS);            // Set the transmission data rate to 1MBit/s
  radio.setCRCLength(RF24_CRC_16);          // Set the CRC length.
  radio.setAutoAck(false);                  // Disable auto-acknowlede packets. Needed for broadcast.
  radio.setPayloadSize(sizeof(dataPacket));  //Set Payload size to the size of the struct 'dataPacket'. Must be not more than 32 bytes.

  //Define the pipes
  // NOTE: the "LL" at the end of the constant is "LongLong" type

  //Pipe 0 is also used by the writing pipe. So if you open pipe 0
  //for reading, and then startListening(), it will overwrite the
  //writing pipe. Ergo, do an openWritingPipe() again before write().
  
  // listen on the broadcast address
  radio.openReadingPipe(1,pipes[1]);
  // write to broadcast address
  radio.openWritingPipe(pipes[1]);

  //start listening on radio
  radio.startListening();

  //Setting some junk default test values
  dataPacket.data1 = 5;
  dataPacket.data2 = 20;
  dataPacket.data3 = 300000;

  Serial.println("printDetails: ");
  radio.printDetails();
  Serial.print("isChipConnected(): ");
  Serial.println(radio.isChipConnected());
}

void loop() {
  radioReader();
  delay(1000);
}

void radioReader()
{
  if (isSender == 1)
  {
    //Sending data
    //Intended code part for ESP8266
    Serial.println("Love is in the air");
    radio.stopListening();
    radio.enableDynamicAck();
    dataPacket.data1 = 1;
    for (uint8_t i = 0; i<3; i++)
    {
      radio.write( &dataPacket, sizeof(dataPacket), 1 );
    }
    radio.startListening();
  }
  else
  {
    //Listener stuff
    //for Arduino

    if (radio.available())
    {
      // Read the data payload until we've received everything
      // Radio has 3x32 bit buffer.
      // Communication is built on the broadcast of the same package
      // High probability that all 3 packages contain the same message
      while (radio.available())
      {
        Serial.println("Found new package.");
        // Fetch the data payload, TODO: handle duplicate packages
        radio.read( &dataPacket, sizeof(dataPacket) );

        if (dataPacket.data1 == 1)
        {
          Serial.println("DISCO");
        
          for (uint8_t i = 0; i<3; ++i)
          {
            delay(200);
            digitalWrite(ledtoblink, HIGH);
            delay(200);
            digitalWrite(ledtoblink, LOW);
          }
        }
        //printPackage();
      }
    }
  }
}
```

Avamander, 2018-01-20:
Update NodeMCU? The latest is 2.1.0.

Zephael86, 2018-01-20:
Even if it is programmed from the Arduino IDE?
I thought that Arduino IDE overwrites the default firmware.

Avamander, 2018-01-20:
You said "NodeMCU V1.0", if you're using it somehow then it's old.

Zephael86, 2018-01-21:
Arduino IDE: 1.8.1
Arduino ESP Core: 2.4.0 updated via "Boards manager"
nRF24 lib: TMRh20 1.3.0 updated via "Library manager"

According to the Arduino ESP Core [documentation](https://arduino-esp8266.readthedocs.io/en/2.4.0/boards.html) I am using the fitting (NodeMCU 1.0) board profile for my micro controller.

Avamander, 2018-01-21:
So now, take the pingpair example, modify only the radio number, check your wiring and upload it to both the Arduino and the ESP. Just to make sure it isn't your code.

Zephael86, 2018-01-21:
Thank you for the tip!
Pingpair_ack works. Typical round-trip delay: 560-620 microsec depending from the micro controllers.
Then it means, that the code is faulty somewhere. I have to start from scratch the nRF chip's configuration.

I also found this [printf issue](https://github.com/nRF24/RF24/issues/188). Described workaround worked for me.

Avamander, 2018-01-22:
@Zephael86 I can do it myself. The printf issue exists because it's hard to solve consistently, we'd need a good cross-platform print library for that :D.

closed 2018-01-22 by Avamander