nRF24/RF24 #412: Arduino nano (clone) and this library

giotta1982, 2018-02-15:
I'm trying to make a nano arduino work with this library, but no luck.

When I open the serial monitor, the dwarf freezes and no signs of life. I'm sure the NRF module works, because I've tested it with another arduino (UNO).
If I try to change the library with that of ManiacBug, the arduino nano does not freeze anymore, but I can not get anything from the other NRF module.
If I use the radio.printDetails () command, I see all incorrect values, and if I try to change them, nothing happens.

I do not have any capacitors between Vcc and GND, I checked the connections and they are ok.

Does anyone know how to solve this problem?
Thank you

wmarkow, 2018-02-15:
Take a look [here](https://github.com/nRF24/RF24/issues/403#issuecomment-357043895) and then [here](https://github.com/nRF24/RF24/issues/311#issuecomment-283482827).

giotta1982, 2018-02-16:
Thank you for your answer, really.
First of all I wanted to tell you that I forgot something, the Arduino Nano is mounted on a Shield Sensor like this:

https://www.amazon.it/gp/product/B01149ELU0/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1
(I hope not to break any rules)

I tried the follow code
:
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

#define CE_PIN 9
#define CSN_PIN 10

RF24 radio (CE_PIN, CSN_PIN);

void setup (void) {
  Serial.begin (9600);
  delay (100);

  radio.begin ();
  bool result = radio.isChipConnected ();

  Serial.println (result);
}

void loop () {
  
}

on the Arduino Nano (clone) + NRF and the result was 0 (false).

I then tried the same NRF module on the Arduino Uno and the result was 1, and the module communicates perfectly with another mounted on another Arduino UNO (so I think I can eliminate the hypothesis that the module is faulty).
I do not understand what's wrong with the Arduino Nano clone.

I then tried the code of the second link and I suffer from the exact same problem as that user. I think the Arduino Nano enters an infinite loop (which makes it look like a freeze state) and I then followed the TMRh20 instructions and took the comment on the #define FAILURE_HANDLING line in the RF24_config.h file.

Now from the serial monitor I read:

RF24 / examples / GettingStarted
*** PRESS 'T' to begin transmitting to the other node
Now sending
1test 2failed test
test 3Sent 40592, Got response 0, Round-trip delay 107260 microseconds
Now sending
1test 2failed test
test 3Sent 1158348, Got response 0, Round-trip delay 101572 microseconds
Now sending
1test 2failed test
3Sent 2272500 test, Got response 0, Round-trip delay 101544 microseconds
Now sending
1test 2failed test
3Sent 3386628 test, Got response 0, Round-trip delay 101524 microseconds
Now sending
1test 2failed test
3Sent 4500728 test, Got response 0, Round-trip delay 101524 microseconds
Now sending
1test 2failed test
3Sent 5614832 test, Got response 0, Round-trip delay 101548 microseconds
Now sending
1test 2failed test
test 3Sent 6728956, Got response 0, Round-trip delay 101516 microseconds
Now sending

Can you explain to me what all this means please?

wmarkow, 2018-02-16:
I think you have some hardware issues between Arduino Nano and nRF24L01 chip. Arduino is connected to nRF24L01 by SPI interface; something is wrong and I can guess (by static source code analysis and your logs), that - when communication with the device by SPI - it gets zeros from the nRF24L01 chip. For example: the library want to get the **status register** from the chip and it reads always 0x00 value.

That's why it is in the infinite loop in the ```bool RF24::write(const void* buf, uint8_t len, const bool multicast)``` method when trying to send the packet. In that method there is a **while** loop that checks for some bits to be set in the **status register**. Since you have some hardware issues, the value of **status register** is always zero and the **while** loop doesn't break. But when you enabled the **FAILURE_HANDLING** flag, this **while** loop will finally break after 95 milliseconds and the program doesn't freeze and you see this kind of entry log:

```
1test 2failed test
test 3Sent 40592, Got response 0, Round-trip delay 107260 microseconds
```

**Got response 0** is always zero, probably because of this hardware issues and it always reads zeros from SPI interface.
The Round-trip is always around 101000 microseconds, which is like 101 milliseconds; this may be the result of being 95 millis in the **while** loop while sending the data. When the **FAILURE_HANDLING** is enabled you can just inspect the public **failureDetected** variable of **RF24** object. I feel that it will be set to **true**.

Some suggestions:
 * how do you power your nRF24L01 chip? Maybe the power source (some kind of voltage regulator) has not enough power? Soldering capacitor directly on the nRF24L01 chip is recommended.
 * maybe you have incorrect wiring to Arduino Nano?
 * try use some other pins for CE and CSN, like:
```
#define CE_PIN 7
#define CSN_PIN 8
```
 * if **isChipConnected** returns **true** then you know that hardware is correct and you can go on with your software

giotta1982, 2018-02-16:
Thank you again for the help you are giving me.

I tried to connect the module to a step-down(5v-3,3v)  but .... I had no luck, ischipconnected is always false.

I also tried to change the pins of the CE / CSN but it does not work, and I checked the wiring that turns out to be correct.

What else can I try?

Avamander, 2018-02-16:
> What else can I try?

Another Arduino, another nRF24L01+, checking your wiring again.

giotta1982, 2018-02-16:
I have 6 modules (all working tested), but none of these gives true result when connected to the Arduino Nano + SensorShield.

I just tried a module (which on the Arduino Nano gave false) on a Mega Arduino and gives True result.

It is not possible that every time I attach a module to the Arduino Nano I'm wrong about the wiring;)

giotta1982, 2018-02-16:
Is it possible that something on the SPI module goes wrong on the Arduino Nano?

Avamander, 2018-02-16:
Did you decouple the 3.3V bus?

giotta1982, 2018-02-16:
OK! I got it. Thank you 2 for the time you have dedicated to me.
Sometimes I'm a little idiot

wmarkow, 2018-02-19:
@giotta1982, so what was wrong? Wiring? Power supply? Something else? :)

giotta1982, 2018-02-19:
hehe ... ok.
I kept checking if the pins were connected correctly when I realized that I had the wrong line ..... I was connected on the mass line and not on the one of the signal 8)

wmarkow, 2018-02-19:
So it was wiring. I see. These kind of issues can happen :) But hopefully you have found it.

closed 2018-03-18 by TMRh20