nRF24/RF24 #417: nRF24 Module not working with Raspberry Pi 2 Model B

Muoq, 2018-03-05:
Hi,

I have a Raspberry Pi 2 Model B running **Raspbian 4.9.59-v7+**. When I try to run the 'gettingstarted' example program and choose role 1 (ping out), I only get this message:
 `RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.`

I have looked at the two other relevant threads I could find, and tried every possible solution that worked for other people, but none of them worked for me.

- I have changed between different constructors: 
The generic constructor, 
one with **GPIO15** and **8MHz**, 
and one with **GPIO22** and **4MHz**
- I have built the libraries both using **SPIDEV** and **BCM2835**
- I have downloaded installed using the dedicated installer (install.sh) and downloaded from github and run the makefile
- I have tried both 3v3 pins on the Raspberry Pi

If I disconnect CE from the Raspberry Pi I don't get the error message anymore, but it prints this instead, repeatedly:
`failed.`
`Failed, response timed out. `
`Now sending...`
Don't know if that's a clue or not.

I am not using a breadboard. I am not using a capacitor between VCC and GND, but I am getting power directly from the Raspberry Pi, and have checked that the voltage is steady during operation.

Thank you in advance!

akatran, 2018-03-10:
I have the same problem.
Temporally worked when I put my oscilloscope between GND and CSN.
If I remove the probe it is not working.

Couldn't manage to find why!

I'll try to put a high value resistor between these two pins to see if that solves the problem.

akatran, 2018-03-11:
It is working now.

I changed the cable and re-solder some loose ends.

Muoq, 2018-03-12:
@akatran Hi

Sorry it took a while for me to get back to you.

Very cool that you have an opportunity to work with an oscilloscope to analyse the circuit. Unfortunately the only thing I'm equipped with is a power supply and an Arduino.

In any case, I see you got it working. If you don't mind my asking, what version of Raspbian are you running, if you are running Raspbian at all. I have tried with an older version of Jessie, which didn't work, and with Wheezy the entire operating system was incredibly slow, and I wasn't even able to install the RF24 library.

I guess my question really boils down to this: What the hell did you do to get it to work? I am frustrated beyond belief. I feel like I have tried everything with non-changing results; always that same pesky 'RF24 HARDWARE FAIL' error message.

Help would be greatly appreciated!

Avamander, 2018-03-12:
@Muoq 

Could you please provide a well-lighted and clear image of your wiring and overall setup?

Muoq, 2018-03-12:
@Avamander 

Sure thing!
Just to be clear what do you mean with overall setup, to make matters more efficient.

Muoq, 2018-03-12:
Ok, here's a link. Let me know if it's too unclear.

https://photos.app.goo.gl/8VMzEOvUdCOxdVK62

Avamander, 2018-03-12:
Okay, the wires do not seem too long, for now I'm going to assume you've wired up everything to the correct pin. Try these steps:
1) Attach a 10000uF electrolytic capacitor directly to the nRF module's VCC and GND (solder it on the module)
2) Check if every pin on the rPI can switch on a led (don't forget a resistor to avoid damaging your LED **and** rPI), it might be that the rPI itself has gotten hit by static and some pin is dead, seen that happen more than once.

Muoq, 2018-03-12:
Thank you very much for your answer. It didn't even cross my mind that the Raspberry Pi would be broken. I will definitely try out your advice tomorrow when I have the time.

akatran, 2018-03-12:
@Muoq Nice case! I like it.

In my case I didn't put a capacitor.

The only "special" I did is to set the same channel on the RPi and Arduinos.
Also I set the PA level to MAX on the RPi and High on Arduinos as I had many messages lost, mainly the replies.

ludiazv, 2018-04-19:
@Muoq , @akatran with PA_MAX direct power from 3.3V could cause problems as this pwr source is used by the board for HDMI, ethernet, etc... Also long wires as you show can affect to device performance. In PLA/NA versions or clones the power requirements are higher that those stated on the datasheet. To avoid power ripple, capacitors are needed.

If you want to avoid any power issues and signal problems use a nrf24l01 break outs boards and short wires as possible. This breakouts are cheap (<$2) [like](https://www.addicore.com/1x-nRF24L01-Adapter-p/ad279.htm) but you can even cheaper in ebay or aliexpress.

For Pis there are also Hats/pHats that could be even better, as they have better physical integration with the Pis. I created for my projects with dual radio suppoer you can check it [here](https://www.tindie.com/products/boros/borosrf2-dual-nrf24l01-phathat--rtc-for-pis/).

Muoq, 2018-04-19:
Hello,

I apologize for the delay. As of now it appears that my Raspberry Pi is the weak link. It is unable to read input. I have tried reading inputs directly with bash aswell as with a simple python script. None of the pins seem to work. As this is not an issue relating to this thread, I am closing it. I will reopen it if I come across something new.

Thank you all very much, your help has been highly appreciated.

closed 2018-04-19 by Muoq