nRF24/RF24 #418: RF24 NOT WORKING WITH ARDUINO DUE!!!

ripperchan, 2018-03-13:
I use SPI on the due to connect with nrf24l01. And use "RF24 radio(43,45) to set up CE and CSN.
But when i use "radio.printDetails()", there are los of "fffffff" and I know there's something wrong.
Then I looked up the resource file. But i can not find something useful.
Help me!

TMRh20, 2018-03-18:
See https://www.arduino.cc/en/Reference/SPI

Its most likely your device is connected improperly.

ripperchan, 2018-03-18:
@TMRh20 
But how about the correct way to connect?
I connected them step by step and checked again and again. So there must be something wrong in my mind.

adokulmez, 2018-11-24:
did you find any solition? Because I have same problem... It's working with arduino uno but when I try the with due it is not working :S

closed 2019-05-10 by TMRh20