nRF24/RF24 #419: Can't get it working - RPi2, Python3, Stretch

GordonTraveler, 2018-03-14:
Hi.

I have an NRF24L01+ connected to my RPi2. It was working before like charm. But then I wanted to change to RPi3 and also from Python2 to Python3 and also made some updates as well and now I can't get it back running again.

What I have now:
Raspberry Pi 2
Raspbian Stretch Lite (Kernel 4.9.59)
Python 3

I tried a lot of libraries the last days. Nothing. Always the same result: Not working.
Most of the tutorials and libraries are just outdated. 

Right now I try to install it with this instructions: https://github.com/haseebehsan/RF24Python
(Seems like they are all based on your library. Thats why I ask here)

```
$ sudo wget http://tmrh20.github.io/RF24Installer/RPi/install.sh
$ sudo chmod +x install.sh 
$ sudo ./install.sh
```
Installing fine. No Problems.

```
$ sudo git clone https://github.com/haseebehsan/RF24Python
$ cd ./RF24Python/pyRF24/pyRF24
$ sudo python setup.py install
```
Not fine. "FileNotFoundError: [Errno 2] No such file or directory: '../Makefile.inc'" There is a Makefile 2 directories below. But if I change the line in the setup.py then the error is 

```
Traceback (most recent call last):
  File "setup.py", line 39, in <module>
    process_configparams()
  File "setup.py", line 19, in process_configparams
    identifier, value = line.split('=', 1)
ValueError: not enough values to unpack (expected 2, got 1)
```

So I skip this step. (Skipping install steps is probably never good...)

```
$ cd ../pyRF24Network
$ sudo python3 setup.py install
```
Ends up in an error:
```
running install
running build
running build_ext
building 'RF24Network' extension
creating build
creating build/temp.linux-armv7l-3.5
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c pyRF24Network.cpp -o build/temp.linux-armv7l-3.5/pyRF24Network.o
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
creating build/lib.linux-armv7l-3.5
arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/pyRF24Network.o -lrf24network -lboost_python3 -o build/lib.linux-armv7l-3.5/RF24Network.cpython-35m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lboost_python3
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-g++' failed with exit status 1
```
(If I compile it with python 2, then it is working. With Python3 not)

Last, but not least:
```
$ cd ../pyRF24Mesh
$ sudo python3 setup.py install
```
Result is the same as above. With Python2 is it working, with Python3 not.
```
running install
running build
running build_ext
building 'RF24Mesh' extension
creating build
creating build/temp.linux-armv7l-3.5
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c pyRF24Mesh.cpp -o build/temp.linux-armv7l-3.5/pyRF24Mesh.o
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
creating build/lib.linux-armv7l-3.5
arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/pyRF24Mesh.o -lrf24mesh -lrf24network -lboost_python3 -o build/lib.linux-armv7l-3.5/RF24Mesh.cpython-35m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lboost_python3
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-g++' failed with exit status 1
```

I am really at the end of my nerves (and running out of time). Can someone help me to get it back running please. In general and especially with Python3.
I am not a big Linux or Hardware expert. Please tell me what I can do to check something or to provide you with more information. Or maybe there is a working tutorial with Python3.

conmarap, 2018-03-14:
First off https://github.com/haseebehsan/RF24Python is a different repository.
Secondly, you don't need sudo to clone a git repository.

Anyway, if you have already cloned the previously mentioned repo, you need to cd into it

``` sh
cd RF24Python

# Build it.
./configure && make

# Build the Python modules.
cd pyRF24
python setup.py build
```

[image]
No errors

GordonTraveler, 2018-03-14:
As I said. With Python2 is no problem to build/install it. But try with Python3.

I know it is a different repository. But the basis is this one, isn't it?
But maybe here is a Python wrapper too? I think I saw one somewhere. I saw so many pages and repositories the last hours/days... I lost overview. Sorry.

Maybe the best would be to start from scratch with this repo, if there is a Python3 wrapper/support. Can you tell me how to install it? And then step by step. I also red somewhere that there is a problem with the Kernel 4.* and SPI. Is that fixed already?

conmarap, 2018-03-14:
You need to link the following library, but worked like a charm:

``` sh
~/RF24Python/pyRF24 $ sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_python-py35.so /usr/lib/arm-linux-gnueabihf/libboost_python3.so

~/RF24Python/pyRF24 $ python3 setup.py build
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c pyRF24.cpp -o build/temp.linux-armv7l-3.5/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/pyRF24.o -lrf24 -lboost_python3 -o build/lib.linux-armv7l-3.5/RF24.cpython-35m-arm-linux-gnueabihf.so
```

GordonTraveler, 2018-03-14:
Not here.
```

pi@RPI2:~ $ cd ~/RF24Python/pyRF24
pi@RPI2:~/RF24Python/pyRF24 $ sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_p                                                                                                                                                              ython-py35.so /usr/lib/arm-linux-gnueabihf/libboost_python3.so
ln: die symbolische Verknüpfung '/usr/lib/arm-linux-gnueabihf/libboost_python3.s                                                                                                                                                              o' konnte nicht angelegt werden: Die Datei existiert bereits
pi@RPI2:~/RF24Python/pyRF24 $ python3 setup.py build
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -march                                                                                                                                                              =armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/                                                                                                                                                              local/include -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c                                                                                                                                                               pyRF24.cpp -o build/temp.linux-armv7l-3.5/pyRF24.o
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC                                                                                                                                                               but not for C++
arm-linux-gnueabihf-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/pyRF24.o -lrf24 -lboost_python3 -o build/lib.linux-armv7l-3.5/RF24.cpython-35m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lboost_python3
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-g++' failed with exit status 1
```

conmarap, 2018-03-14:
Do you have libboost-dev installed?

GordonTraveler, 2018-03-15:
Seems like yes.
I found this script: https://github.com/G-Node/nixpy/blob/master/scripts/findboost.py
I tried it with "python" and "python3". Both was fine.

```
pi@RPI2:~/RF24Python/pyRF24 $ python findboost.py --selftest
Python version: [2.7]
Checking: Truelibboost_python.so                  ok
Checking: Truelibboost_python3.so                 ok
Checking: Truelibboost_python.dylib               ok
Checking: Truelibboost_python3.dylib              ok
Checking: Truelibboost_python-py27.so             ok
Checking: Nonelibboost_python-py27.so.1.55.0      ok
Checking: Truelibboost_python-py34.so             ok
Checking: Truelibboost_python-py35.so             ok
Checking: Truelibboost_python.lib                 ok
Checking: Truelibboost_python-mt.dylib            ok
Checking: Truelibboost_python-vc120-mt-1_57.lib   ok
Checking: Truelibboost_python-vc120-mt-1_61.lib   ok
Checking: None/usr/lib/libboost_python.a          ok
pi@RPI2:~/RF24Python/pyRF24 $ python3 findboost.py --selftest
Python version: [3.5]
Checking: Truelibboost_python.so                  ok
Checking: Truelibboost_python3.so                 ok
Checking: Truelibboost_python.dylib               ok
Checking: Truelibboost_python3.dylib              ok
Checking: Truelibboost_python-py27.so             ok
Checking: Nonelibboost_python-py27.so.1.55.0      ok
Checking: Truelibboost_python-py34.so             ok
Checking: Truelibboost_python-py35.so             ok
Checking: Truelibboost_python.lib                 ok
Checking: Truelibboost_python-mt.dylib            ok
Checking: Truelibboost_python-vc120-mt-1_57.lib   ok
Checking: Truelibboost_python-vc120-mt-1_61.lib   ok
Checking: None/usr/lib/libboost_python.a          ok
```
Also tried to locate the boost files. Seems like they are there for Python 2 and 3:

```
pi@RPI2:~/RF24Python/pyRF24/pyRF24Network $ dpkg --list | grep libboost
ii  libboost-all-dev                       1.62.0.1+b4                        armhf        Boost C++ Libraries development files (ALL) (default version)
ii  libboost-atomic-dev:armhf              1.62.0.1+b4                        armhf        atomic data types, operations, and memory ordering constraints (default version)
ii  libboost-atomic1.62-dev:armhf          1.62.0+dfsg-4                      armhf        atomic data types, operations, and memory ordering constraints
ii  libboost-atomic1.62.0:armhf            1.62.0+dfsg-4                      armhf        atomic data types, operations, and memory ordering constraints
ii  libboost-chrono-dev:armhf              1.62.0.1+b4                        armhf        C++ representation of time duration, time point, and clocks (default version)
ii  libboost-chrono1.62-dev:armhf          1.62.0+dfsg-4                      armhf        C++ representation of time duration, time point, and clocks
ii  libboost-chrono1.62.0:armhf            1.62.0+dfsg-4                      armhf        C++ representation of time duration, time point, and clocks
ii  libboost-context-dev:armhf             1.62.0.1+b4                        armhf        provides a sort of cooperative multitasking on a single thread (default version)
ii  libboost-context1.62-dev:armhf         1.62.0+dfsg-4                      armhf        provides a sort of cooperative multitasking on a single thread
ii  libboost-context1.62.0:armhf           1.62.0+dfsg-4                      armhf        provides a sort of cooperative multitasking on a single thread
ii  libboost-coroutine-dev:armhf           1.62.0.1+b4                        armhf        provides a sort of cooperative multitasking on a single thread (default version)
ii  libboost-coroutine1.62-dev:armhf       1.62.0+dfsg-4                      armhf        provides a sort of cooperative multitasking on a single thread
ii  libboost-coroutine1.62.0:armhf         1.62.0+dfsg-4                      armhf        provides a sort of cooperative multitasking on a single thread
ii  libboost-date-time-dev:armhf           1.62.0.1+b4                        armhf        set of date-time libraries based on generic programming concepts (default version)
ii  libboost-date-time1.62-dev:armhf       1.62.0+dfsg-4                      armhf        set of date-time libraries based on generic programming concepts
ii  libboost-date-time1.62.0:armhf         1.62.0+dfsg-4                      armhf        set of date-time libraries based on generic programming concepts
ii  libboost-dev:armhf                     1.62.0.1+b4                        armhf        Boost C++ Libraries development files (default version)
ii  libboost-exception-dev:armhf           1.62.0.1+b4                        armhf        library to help write exceptions and handlers (default version)
ii  libboost-exception1.62-dev:armhf       1.62.0+dfsg-4                      armhf        library to help write exceptions and handlers
ii  libboost-fiber-dev:armhf               1.62.0.1+b4                        armhf        cooperatively-scheduled micro-/userland-threads (default version)
ii  libboost-fiber1.62-dev:armhf           1.62.0+dfsg-4                      armhf        cooperatively-scheduled micro-/userland-threads
ii  libboost-fiber1.62.0:armhf             1.62.0+dfsg-4                      armhf        cooperatively-scheduled micro-/userland-threads
ii  libboost-filesystem-dev:armhf          1.62.0.1+b4                        armhf        filesystem operations (portable paths, iteration over directories, etc) in C++ (default version)
[... cut ...]
And from here: http://tmrh20.github.io/RF24/Python.html... Step 1 and 2 was done before already. Step 3 "http://tmrh20.github.io/RF24/Python.html":

```
pi@RPI2:~/RF24/pyRF24 $ ./setup.py build
Traceback (most recent call last):
  File "./setup.py", line 39, in <module>
    process_configparams()
  File "./setup.py", line 14, in process_configparams
    with open('../Makefile.inc') as f:
IOError: [Errno 2] No such file or directory: '../Makefile.inc'
```
In "/RF24" is a "Makefile", but without ".inc". Shall I use this one? But I guess it is for something else, isn't it?

conmarap, 2018-03-15:
Did you do `./configure && make` in the ./RF24 folder?

GordonTraveler, 2018-03-15:
No. I didn't (as I said... I am a beginner with this... sorry). Now I did and the  `./configure && make` went good... seems like.

[image]

After that I could do `./setup.py build`. That was fine too. But I need it in Python3. So I did `Python3 setup.py build`. And this failed again (/usr/bin/ld: cannot find -lboost_python3).

[image]

It can not find the lboost_python3. But why?
Btw... I don't have a path "/usr/bin/ld".

conmarap, 2018-03-15:
`ld` is your linker, so if `build-essential` is installed, then that is installed as well. You can do: `which ld` to find its path.

As for your issue: It seems like python can't find the shared library for libboost. Could you do the following and attach the output?

``` sh
ls /usr/lib/arm-linux-gnueabihf/libboost_python*
```

Or

``` sh
find /usr -name "libboost_python*"
```

GordonTraveler, 2018-03-15:
Hi conmarap. Thank you for your help.

Here is the output of this 3 commands:
[image]

conmarap, 2018-03-16:
No problem. Ok awesome! In the output from `ls` the red line means that the symbolic link you performed last time didn't work so it's a broken link. Try removing that file (the link) and redo that. Then make sure it's cyan like the one below it. That should work.

GordonTraveler, 2018-03-16:
Hi conmarap.

It was the broken link. Damn... such a simple thing. Took me several days of headache. 
I removed the old link, set a new one and the compiling errors disappeared. Took me now still an 2 hours to figure out that it works a bit different now. But finally it is running again. Can't tell you how happy I am. Now I can concentrade on more important things again :-)
Thank you very much for your help and patience.

closed 2018-03-16 by Avamander

conmarap, 2018-03-16:
No problem mate. Good luck on your projects.