nRF24/RF24 #422: ESP8266 NRF integration trouble

gberthoud, 2018-03-19:
Hi,

i'm trying to connect to a ADAFRUIT HUZZAH ESP8266 BREAKOUT an NRF24L01 module, and use it with this library.

First, the wirring:
ESP8266 12 -> MISO
ESP8266 13 -> MOSI
ESP8266 14 -> CLOCK
ESP8266 15 -> CS/SS
ESP8266 02 -> CSE

The Sketch
`#include <SPI.h>
#include <RF24.h>
#include <ESP8266WiFi.h>
RF24 radio(2, 15);
void setup() {
  Serial.begin(115200);
  yield();
  radio.begin();
  radio.printDetails();
}

void loop() {
}`

The result:
STATUS		 = 0x3ffef3c0 RX_DR=3ffef3b0 TX_DS=4 MAX_RT=0 RX_P_NO=4022f64e TX_FULL=3ffef3c0
P⸮⸮?	 = 0x3ffef3a03ffef3a03ffef3a03ffef3a03ffef3a0 0x3ffef3a03ffef3a03ffef3a03ffef3a03ffef3a0
P⸮⸮?	 = 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3a03ffef3a03ffef3a03ffef3a03ffef3a0
P⸮⸮?	 = 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0
P⸮⸮?	 = 0x3ffef3c0 0x3ffef3c0
Data Rate	 = P⸮⸮?
Model		 = P⸮⸮?
CRC Length	 = P⸮⸮?
PA Power	 = P⸮⸮?

Environnement:
> Arduino IDE 1.8.5
> Windows 10
> rf24 1.3.1
> esp8266 2.3.0

i try with several NRF24L01 module, with the same result.

Is this a HW issue, or the Library?

JLuisAlcaraz, 2018-03-23:
I have a NodeMCU Lolin, and a Wemos Mini D1 (booth with ESP8266) working with a nRF24L01+ board. My CE/CSN pins are 0,16:

RF24 radio(0,16);

I like to test if the chip is there,by using:

 Serial.println(radio.isChipConnected());

And the capacitor on the power pins is necessary. Youll find some boards works witout it, but another ones dont, so I put a 10uF on every rf24 board (I spend a day with a problem where the connection between 2 rf24 boards fails randomly every second or so... it was the capacitor!! I put one and solved!

Also check you have the last ESP8266 core, and the lib updated.

docteurzoidberg, 2018-08-16:
got exact same issue on a wemos d1 mini. i tried all the different pin configurations

docteurzoidberg, 2018-08-16:
i got radio.isChipConnected() = false every time with all possible pinouts

closed 2020-03-29 by TMRh20