nRF24/RF24 #423: pyRF24 SPIDEV undefined symbol "vtable for SPI"

sergiy-nazarenko, 2018-03-25:
Hello!

I compiled librf24 and which is works fine with nrf module.
But with py module something wrong. Import RF24 gives me an error
ImportError RF24.so undefined symbol _ZTV3SPI

I would be very appreciate for any hints how fix that

TMRh20, 2020-08-11:
Old issue, no idea what to suggest beyond trying again with current instructions and code. Closing.

closed 2020-08-11 by TMRh20

ErikApption, 2021-12-04:
I'm having this issue with latest v1.4.2 compiled with spidev on RPI2 with latest bullseye distribution

2bndy5, 2021-12-04:
@ErikApption Please open a new issue with more detail than this post offered. We have issue templates for a reason. Answering all the questions (especially the steps to reproduce) in the bug template will help us help you.