nRF24/RF24 #424: Library segfaults when not running as root on Raspberry Pi

paulirwin, 2018-04-05:
I assume this affects other Linux distros besides Raspbian on the Pi, but I haven't tested it as I don't have a Linux device with GPIO/SPI handy. This was encountered on the latest version of Raspbian on the Pi 3.

The library (including all of its examples) does not work on the Pi unless you're running as root/su, which may be common knowledge, but it is not mentioned in the [RPi documentation page](http://tmrh20.github.io/RF24/RPi.html). It is subtly shown to use sudo on the [Linux documentation page](http://tmrh20.github.io/RF24/Linux.html) but I did not discover this until after I found the solution to the problem. Also I was [not the only one](https://raspberrypi.stackexchange.com/questions/80326/rf24l01-rf24-creation-segmentation-fault-in-raspberry/82244?noredirect=1#comment130147_82244) that had this problem.

The result of running it as non-root on the Raspberry Pi is a segfault, which is extremely user-unfriendly. Ideally, the library should detect if SPI is unavailable and throw a meaningful exception.

It appears from the [code comments here](https://github.com/nRF24/RF24/blob/master/utility/RPi/bcm2835.h#L63-L67) that this should be as simple as checking the return value for zero [here](https://github.com/nRF24/RF24/blob/d89ff7e1b7021d10dfaabc5fadbf4bc63a0edb56/utility/RPi/spi.cpp#L16). I am not a C++ expert but I think I can try my hand at submitting a PR for this, I just wanted to go ahead and get an issue logged in case someone had a better solution in the meantime. Also, the RPi documentation should probably be updated with this requirement.

akatran, 2018-04-15:
Maybe someone with more knowledge could help on this.

Debugging the library (non root user) I saw something that it is does not looks good.

This is a backtrace of the failing function ....
`
#0  bcm2835_peri_read (paddr=paddr@entry=**0xffffffff**) at utility/RPi/bcm2835.c:124
#1  0x76fa5cf8 in bcm2835_peri_set_bits (paddr=**0xffffffff**, value=0, mask=12) at utility/RPi/bcm2835.c:195
#2  0x76fa5a9c in SPI::beginTransaction (settings=...) at utility/RPi/spi.cpp:23
#3  0x76fa3d88 in beginTransaction (this=0x24318 <radio>) at RF24.cpp:70
#4  RF24::write_register (this=this@entry=0x24318 <radio>, reg=reg@entry=0 '\000', value=value@entry=12 '\f') at RF24.cpp:194
#5  0x76fa5890 in RF24::begin (this=0x24318 <radio>) at RF24.cpp:638
#6  0x76f6d5a8 in RF24Mesh::begin(unsigned char, rf24_datarate_e, unsigned int) () from /usr/local/lib/librf24mesh.so.1
#7  0x76f5828c in RF24Gateway::begin(bool, bool, unsigned short, unsigned char, rf24_datarate_e, unsigned char) () from /usr/local/lib/librf24gateway.so.1
#8  0x76f58114 in RF24Gateway::begin(unsigned char, unsigned char, rf24_datarate_e) () from /usr/local/lib/librf24gateway.so.1
#9  0x00012c74 in main ()
`
At the first two function calls the paddr is 0xffffffff which does not seems to be right.

Even if there is no alternative to find if the SPI or the bcm can access the hardware, it can easily detected from paddr.

akatran, 2018-04-15:
While digging...

```
(gdb) print bcm2835_spi0
$24 = (volatile uint32_t *) 0xffffffff
```

akatran, 2018-04-19:
As the bcm2835 could not initialize the spi0 because of privileges we could make a check if we actually have root access.
As SPI is not available the radio functions will not work so the program will not run.
Instead of having a segmentation fault we could just print a 

> 3 0x76fa3d88 in beginTransaction (this=0x24318 <radio>) at RF24.cpp:70

We can easily detect root access there with something like :

```
  inline void RF24::beginTransaction() {
    #if defined (RF24_SPI_TRANSACTIONS)
    if (geteuid() == 0){
        _SPI.beginTransaction(SPISettings(RF24_SPI_SPEED, MSBFIRST, SPI_MODE0));
    }else{
        fprintf(stderr, "Root access required!\n") ;
        exit(-1);
    }
    #endif
    csn(LOW);
  }
```

akatran, 2018-04-19:
I'm going to create a PR for that!

closed 2019-05-10 by TMRh20