nRF24/RF24 #432: `radio.write(...)` always returns false

NoeelMoeskops, 2018-04-30:
Hi,

I can't seem to get my [NRF24L01+PA/LNA](https://www.aliexpress.com/item/1sets-Special-promotions-1100-meter-long-distance-NRF24L01-PA-LNA-wireless-modules-with-antenna/32341792715.html?spm=a2g0s.9042311.0.0.N7QNqb) to work. When i upload the gettingstarted script it always fails to write on the pipeline. `radio.write( &start_time, sizeof(unsigned long) )` always returns false. I ~quadruple~ hextuple checked my wireing, but everything seems to be in order. I use an external power supply for the RF24 and my host in an Arduino Nano. Please note that I also tried [maniacbugs lib](https://github.com/maniacbug/RF24) with the same result. So it might be hardware related but I really don't know where to look.

Avamander, 2018-04-30:
What kind of external power supply are you using? A linear 3.3V regulator from 5V?

NoeelMoeskops, 2018-04-30:
linear 3.3v, also `radio.printDetails();` returns:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x27
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```

NoeelMoeskops, 2018-04-30:
When I change the CE and CSN pin to 9 and 10. One node fails to write, but the other node [still receives incoming messages](https://imgur.com/a/p5SZ5cL)

Avamander, 2018-04-30:
Try decoupling the power bus a bit more, see if that changes things.

NoeelMoeskops, 2018-04-30:
I added two 100uF capacitors in parallel but no effect...

Avamander, 2018-04-30:
Okay, try disabling auto-ACK, lowering the power (PA) to minimum, and disable dynamic payloads.

NoeelMoeskops, 2018-05-01:
oke, i disabled auto-ACK and set the PA to min (also tried low) and disabled dynamic payloads but the message just won't arrive. It's fine sending now tho, it just fails on timeout (increased to 5 sec). I also tried it with an ATmega2560 but no luck.

Avamander, 2018-05-01:
Are you modifying the examples in any other way? Did you make sure the radio numbers are different?

NoeelMoeskops, 2018-05-02:
the only modifications I have made are:
```
  radio.setPALevel(RF24_PA_MIN);
  radio.disableDynamicPayloads();
  radio.setAutoAck(false);
  radio.setRetries(15, 15);
  printf_begin();
  radio.printDetails();
```
I also tried a different power source and uploading directly from the arduino IDE instead of using platformIO. And using release 1.3.1 instead of pulling from master but the results are all the same.

Avamander, 2018-05-02:
The radio might just be broken. Though, if you had an oscilloscope you could check the power rail for any fluctuations. But just for your own sake, are you sure that you're using the right pins on the Arduino Mega as the pins differ compared to Arduino Uno, this is literally the most common and probable error.

NoeelMoeskops, 2018-05-02:
I got an oscilloscope and the 3.3v is an extremely flat line. I could also check the output of the RF24 but i don't know what to look for so instructions would be nice :)
Please note that i use an Arduino Nano and not an Uno.
The connection on the Mega is as follows:
SCK   - 52
MOSI - 51
MISO - 50

I used the pin-out as described in the [documentation.](http://tmrh20.github.io/RF24/#Platform_Support).
is the [pin-out of my RF24](https://ae01.alicdn.com/kf/HTB1bHZ4LVXXXXXLXFXXq6xXFXXX3/205957859/HTB1bHZ4LVXXXXXLXFXXq6xXFXXX3.jpg?size=128281&height=648&width=929&hash=c02dc61d178ba0ceacd2f6ced18e3350) correct?

sergiokoval, 2018-05-08:
hi, just started playing with those modules. 

Got the same issue so decided to go from zero. 
My setup is based on two arduino nano boards.
First sample I tried is to use one way communication. 
Didn't work till I connected RF24 modules to external 3.3V supply with good filter capacitor at the output > 500uF.
Now playing with two way communication, and seems at this stage things working out well.
Working sample is here - [https://github.com/sergiokoval/eskate/tree/feature/two-way-comm](https://github.com/sergiokoval/eskate/tree/feature/two-way-comm)
Master branch contains simple one-way comm, and two-way comm is in feature branch.

Hope this somehow helps

[image]
[image]

NoeelMoeskops, 2018-05-17:
@sergiokoval Thanks for your comment. I tried your repo, but no luck. Did it with 300uF (for both RF24's) tho. Gonna see if I can get my hands on a bigger one. Is that 1000uF you got in the picture really necessary?

Avamander, 2018-05-17:
@NoeelMoeskops Pretty much.

nygma2004, 2018-07-05:
I have a similar issue. I have to nano with nRF24L01. On the sender, I am using the extra board with a separate 5V->3.3V regulator and extra caps. I have only one writing pipe open. The other nRF24L01 is also on a nano, running off the nano 3.3 line with a cap. 
The sender says transmit failed for all packets, but the receiver receives all of the messages. If I turn off the acknowledgement, nothing is received on the receiver.
Shall I just simply ignore radio.write return code in this case? I am planning a one way communication anyway.

JoseAntonio86, 2018-07-06:
Tengo el mismo problema con mis módulos NRF24l01 y 2 arduino Uno, al parecer "radio.write" sólo me da como respuesta "False", lograron solucionarlo?

void loop(void)
{
    radio.stopListening();    
    unsigned long time = millis();
    Serial.print("Enviando  ");
    Serial.println(time);
    bool ok = radio.write( &time, sizeof(unsigned long) );

    if (ok)
       { 
        Serial.println("ok...");
       }
    else
       {
        Serial.println("failed");
       }

YashiSharma98, 2018-10-19:
The main reason for this issue is 3.3V supply. For me, the 3.3V supply from Arduino Uno was causing trouble, but when I used the 3.3V supply from NodeMCU (ESP-12E Module), the issue was resolved.

JoseAntonio86, 2018-10-19:
Finalmente decidí optar por los modulos Xbee (zigbee) para la comunicación inalámbrica de mi proyecto. Igualmente muchas gracias por la información, haré una prueba con los ESP.

closed 2019-05-10 by TMRh20