nRF24/RF24 #433: nrf24l01+pa+lna(Si24R1+pa+lna) draws in PA_MAX mode only 27mA.

Pluscrafter, 2018-05-03:
I've bought a fake nrf24l01+pa+lna(Si24R1+pa+lna) and set the PALevel to PA_MAX and measure the powerconsuption with my multimeter I can only measure 27mA peak. The range of the module is ca 270m throught some trees but in line of sight. I've power the module with a raspi zero w, that should deliver 50mA. and In the datasheet the powerrating is rated to 115mA. What might be wrong?

There are the settings for my module:
`radio.begin();
radio.setPayloadSize(32);
radio.setChannel(125);
radio.setDataRate(RF24_250KBPS);
radio.setPALevel(RF24_PA_MAX);
radio.setAutoAck(true);
radio.enableDynamicPayloads();
radio.enableAckPayload();
	
radio.openWritingPipe(addresses[1]);
radio.printDetails();`

TMRh20, 2018-05-04:
You likely need an oscilloscope to accurately measure the current spikes when the device is transmitting. I’d guess it’s functioning normally.

Pluscrafter, 2018-05-07:
Ok thx. The std::system("clean"); has slow my program down.

closed 2018-05-07 by Pluscrafter