nRF24/RF24 #435: Why my program stops if my NRF is not connected?

esp13, 2018-05-12:
Hi,

I know my question is weird:

Why will my program stops if I try to "write" with my NRF while it is currently missing (unconnected)? My arduino doesn’t manage only the radio communication I doesn't want the whole system to stop for a wire/connection or NRF failure.

I see this with arduino pro mini and nano but I think this come from the library.

Is there a solution to manage the unconnected error and still continuing the program?

And why the watchdog doesn't trigger in this case?

Thank you for reading.

TMRh20, 2018-05-13:
It is due to the logic in the library. There is an option to enable "Failure Handling" in RF24_config.h

See http://tmrh20.github.io/RF24/classRF24.html#a2e40fe66d1231a333aa2534e8491f828

esp13, 2018-06-15:
Thank you for your answer,

By default, will this result in program to get stuck? Or only the communication with NRF24?

I get my whole program stopping when i try to write with NRF24 unconnected. I have to reset it (power / RST button)

wmarkow, 2018-06-15:
I think the whole program will stuck as it may hang in an endless **while** loop. From this point of view I recommend to enable "Failure Handling" in RF24_config.h (as @TMRh20 has suggested before).

TMRh20, 2019-05-10:
#494

closed 2019-05-10 by TMRh20