nRF24/RF24 #437: RF24/NodeMCU Not sending/reciving :(

ksperr, 2018-05-25:
I have a problem with the RF24 module and the NODEMCU v1 the one with white back (NodeMCU AMICA). I was working in a prototype for a project, all was ok and it was working. but recenly i bought 10 mcu more. No one is working with the same code and the same circuit. It is reciving the same voltage as it recives in the one that works, i tested the conecctions with the tester...all is ok. I tried to change bootloader and sdk...no way. It doesnt send and it doesnt recive. 
The connections are:
Rf24: d1 CE, d2 csn, d5 clk, d6 miso, d7 modi.
And a Sd card module (working) on: d5,d6,d7 and cs on d8.

ksperr, 2018-05-25:
The sd is working fine, it connects to WiFi, it connect with telegram. only thing is not working is the radio. I tried to put a led on d1 and d2 to see if something was wrong with them.  and it was ok.

Avamander, 2018-05-25:
Attach a large electrolytic capacitor (1000uF-10000uF) to the 3.3v and GND pins of the nRF24 (solder straight to the header itself) and see if that solves it.

ksperr, 2018-05-25:
ok thanks! i will try. so two electrolyc capacitor , one in gnd and another on VCC(3.3v) of 1000uf to 10000uf. I understand right?  for what is it needed?

ksperr, 2018-05-26:
1000-10000uF? are you sure? its not to much?

ksperr, 2018-05-26:
ok, i Will buy it on Monday. today I was doing some experiments and I realized that the problem maybr have something to do with the sd module. when i connect NodeMCU+rf24+sd then sd works but radio doesn't. if I take out the sd module then radio works. in the other NodeMCU it works with all connected, so it must be something in the new models. the bad regulator is the thing that regulates the power from 5v from the USB to 3.3v ? I tried to measure the voltatge and it gives the same as in the one that works.

Avamander, 2018-05-26:
If you switch the SD card modules, what happens then?

ksperr, 2018-05-26:
i have micro SD card modules and normal SD card module, with the big ones it seems to work better but it stills gives problems. I didn't had this problems with the other node MCU. it's weird because I can read the sd card, and if I take out the sd card module and put a radio module it works.. but when I try to use the two .. radio didn't work.

Avamander, 2018-05-26:
Are you sure your SD card readers are actually have a fully functioning implementations of SPI? The issue you describe sounds like it's not respecting the slave select signal or you are incorrectly handling it. Which means you might have to use SoftSPI with either one of those.

ksperr, 2018-05-28:
if I didn't connect the radio I can read the files of SD card.
 if I use radio without the sd module it works. 
but doesn't work together.
 it can't be a problem of the code (handling incorrectly spi) because it works on other NodeMCU I bought before (i tried to change bootloader, and sdk version, and it's the same) 
I didn't have time to try the capacitors solution, (I can't buy them in my town) you think it will work with 100uF that I actually have?