nRF24/RF24 #438: Segmentation fault when calling member functions, e.g. getChannel, printDetails

ngfelixl, 2018-05-25:
Hi,
I'm currently trying to build a nodejs nbind c++ addon on top of this library. The configuration and installation of this lib was successful, followed the instructions on this github page for raspberry pi. Next I importet the shared lib to my nbind project and I'm able to create RF24 instances successfully. The program is also able to share data between nodejs and c++. But when I call RF24 member functions (tested printDetails and getChannel) I get a segmentation fault like follows

```
Thread 1 "node" received signal SIGSEGV, Segmentation fault.
0xb4c15ff8 in bcm2835_peri_read () from /usr/local/lib/librf24.so.1
```

The printDetails functions breaks after the first 3 lines returning the pins and the clock speed.

Any suggestions on how to debug this?

----------------------

For completion I also share a simplified version of my code. The .cpp file looks as follows

```cpp
#include <RF24.h>
#include "nrfnode.h"

Nrfnode::Nrfnode(uint16_t cePin, uint16_t spiBus): cePin(cePin), spiBus(spiBus) {
  this->radio = new RF24(cePin, spiBus);
}

uint8_t Nrfnode::getChannel() {
  return this->radio->getChannel();
}
```

This is the corresponding header

```cpp
#ifndef NRFNODE_H
#define NRFNODE_H

class Nrfnode {
  uint8_t cePin = 22, spiBus = 0;
  RF24* radio;

  public:
    Nodenrf(uint16_t cePin, uint16_t spiBus);
    uint8_t getChannel();
};

#endif

#include "nbind/nbind.h"
NBIND_CLASS(Nrfnode ) {
  construct<uint8_t, uint8_t>();
  method(getChannel);
}
```

ngfelixl, 2018-05-25:
I have written a test program without nodejs which is working very well without segmentation fault when executing as root. Normal execution leads to the same error. Any suggestions for making it `working as non-root or letting nodejs execute its addons as root (if that's possible)?

TMRh20, 2018-05-25:
The bcm SPI driver needs root. You can set it up using spidev driver to use non root. Google for instructions.

ngfelixl, 2018-05-26:
SPI driver is working, but now every function call (printDetails etc) in nodejs prints `Error: can't send spi message`. Even **gdb** does not print any c++ related issues, but if I remove the calls it works. In the test program without node everything works as expected, even without root.
Any ideas?

mitch85, 2020-06-05:
same issue after update nodejs to 12.18.0 and nrf24 to 0.1.5. Bevor with node js v 8.x and nrf24 0.1.4 it works fine

2bndy5, 2021-03-31:
closing this as it doesn't seem to require changes on our end

closed 2021-03-31 by 2bndy5