nRF24/RF24 #440: Crash on radio.begin()

P0W, 2018-06-10:
I have connected my NRF24 module as described here http://tmrh20.github.io/RF24/RPi.html to my RPi

I cloned this repository and have Build using SPIDEV (I have ensured SPI was enabled and double checked that SPI module works, following [this](http://www.brianhensley.net/2012/07/getting-spi-working-on-the-raspberry-pi.html) tutorial  .

However when I build and launch ./gettingstarted, is abruptly terminated.

```
pi@raspberrypi:~/RF24/examples_linux $ ./gettingstarted
RF24/examples/GettingStarted/
terminate called after throwing an instance of 'int'
Aborted
```

I figured its from radio.begin();

Also if no NFR24 is connected to RPi  , the same abort happens.

What I've missed , or doing wrong ?

TMRh20, 2018-06-10:
Use “sudo rf24... “ . The driver neeeds root. The spidev driver can be set up so this is not required.

closed 2018-06-11 by P0W

Shucach, 2021-01-22:
> Use “sudo rf24... “ . The driver neeeds root. The spidev driver can be set up so this is not required.
> […](#)

How to set permission for all users? I need to start this without root.

TMRh20, 2021-01-22:
https://github.com/nRF24/RF24/issues/194#issuecomment-172883559