nRF24/RF24 #449: ESP8266 & RF24 exception

Programmer74, 2018-08-07:
Hello everyone
I think I need help with my esp8266 & rf24lib under Arduino IDE

**Source code:**
```
#include "RF24.h"

RF24 radio(4,5);

void setup(){
  Serial.begin(115200);

  radio.begin();
  radio.setAutoAck(1);
  radio.enableAckPayload();
  radio.setRetries(0,15);
  radio.setPayloadSize(1);
  radio.powerUp();
  radio.printDetails();
}

void loop(void) {
    
}
```
**Expected behaviour**
Printed radio info

**Actual behaviour**:
_serial monitor output:_

```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40209fac, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40235ff1, depc=0x00000000

Exception (3):
epc1=0x40209fac epc2=0x00000000 epc3=0x00000000 excvaddr=0x40235ff1 depc=0x00000000

ctx: cont 
sp: 3ffffc20 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffdc0:  40106f10 00002000 feefeffe 40208c10  
3ffffdd0:  3ffe8308 3ffe8308 3ffef8f0 4020904c  
3ffffde0:  3ffe8308 40201be2 0000008c 0000003e  
3ffffdf0:  3ffefa3c 3ffe8308 3ffef8f0 402094bd  
3ffffe00:  3ffe8308 00000001 3ffffe30 3ffef8f0  
3ffffe10:  0000000a 3ffe8308 3ffef8f0 4020954d  
3ffffe20:  40208bc0 3ffe8308 3ffef8f0 40209177  
3ffffe30:  3ffe8308 3ffef868 00000000 40208bb7  
3ffffe40:  3ffe8308 00000000 00000010 40208c10  
3ffffe50:  3ffe8308 feefeffe feefeffe 3ffef8f0  
3ffffe60:  3ffef826 00000000 3ffe8308 40208f48  
3ffffe70:  00000000 ffffffff 00000000 00000000  
3ffffe80:  00000000 00000000 ff302073 00000000  
3ffffe90:  00000001 0000003c 00302078 401002ac  
3ffffea0:  000001b8 0000012d 3ffee9d0 3ffe8308  
3ffffeb0:  00000004 40201be2 000001b8 40100866  
3ffffec0:  0030006c 40207844 00000000 40201be2  
3ffffed0:  3fffff50 3fffff40 00000008 40207444  
3ffffee0:  3ffef825 3fffff50 00000024 00000001  
3ffffef0:  00000012 00000002 3ffee9d0 3fffff70  
3fffff00:  4023600a 3ffef824 3ffe8308 4020652a  
3fffff10:  3fffff50 3fffff40 00000004 3fffff70  
3fffff20:  4023600a 7fffffff 3ffef824 40203d9f  
3fffff30:  3fffff50 3fffff40 00000004 40203da8  
3fffff40:  3fffff60 40235ff1 0000000a 00000002  
3fffff50:  3ffefac0 00000002 00000000 00000000  
3fffff60:  3fffdad0 00000002 3ffee808 40202a36  
3fffff70:  3fffdad0 00000000 0000000a 4020299c  
3fffff80:  00000000 00000000 3ffee808 3ffee978  
3fffff90:  3fffdad0 00000000 3ffee808 40202dd7  
3fffffa0:  3fffdad0 00000000 3ffee808 40202740  
3fffffb0:  feefeffe feefeffe 3ffee970 40203bd8  
3fffffc0:  feefeffe feefeffe 3ffe85d8 401000e5  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

**Board**: NodeMCU WeMos [this one](http://roboshop.spb.ru/NodeMCU-V3-WeMos)
**Pinout**:
```
NRF pin - ESP GPIO number - Board pin
CE - 4 - D2
CSN - 5 - D1
SCK - 14 - D5
MOSI - 13 - D7
MISO - 12 - D6
```
Tripple checked all hardware stuff, connectivity, voltages. Wires are **soldered**. Capacitor is **soldered**. Pins 4/5 are not esp system ones. Tried with other CE/CSN with no luck. SCK, MOSI & MISO are hardware SPI pins.

**Versions**
Arduino IDE: **1.8.3**
Arduino ESP compiler&libs: **2.4.2**
NRF lib: **latest at commit a871827**

What am I doing wrong?
Any help appreciated

wmarkow, 2018-08-07:
I feel like you have some hardware issues; probably SPI connectivity.
Take a look at [one of my comments here](https://github.com/nRF24/RF24/issues/360#issuecomment-405107970), especially in **setup()** method call **isChipConnected** and check its result.

When the chip is not connected correctly, then the library may stuck at those **infinite while** instructions.

Programmer74, 2018-08-08:
@wmarkow The crash seems to happen before I call isChipConnected.
I have a feeling this has something to do with SPI implementation and/or RF library.
I have tried different versions of this library (from different sources on the web) and sometimes I have symptoms close to [this](https://github.com/nRF24/RF24/issues/422), and sometimes `printDetails` goes fine, but all the hardware addresses are wrong and sending is failed.

wmarkow, 2018-08-08:
Maybe you could try to enable the ```FAILURE_HANDLING``` flag as described roughly (here)[https://github.com/nRF24/RF24/issues/435#issuecomment-388624847]? Then build the project and run it and you can see if you still have this crash (exception).

nailyk-fr, 2018-12-20:
Hello, 

(noob here)
I am commenting here because it seems to be the most recent issue regarding RF24 & `nodemcu`. 
Agreed with https://github.com/nRF24/RF24/issues/244#issuecomment-357211322 which suggest to make a dedicated FAQ

So almost same issue here as everyone: 
1. Used the `gettingstarted` sample to try getting the `RF24` & `nodemcu` working. 
Here is the 'sketch' used: [master_nodemcu.txt](https://github.com/nRF24/RF24/files/2699136/master_nodemcu.txt)
I was finally able to get some kind of communication between the RF module & `nodemcu` with this wiring diagram: http://fablab.ruc.dk/controlling-the-mi-light-from-arduino/ and now the 'isChipConnected' fct report success. 
However, using the `printDetails` function or switching the program to `transmitting` mode cause the `nodemcu` to crash. 

I check with the power rail and it looks ok. I also tried two different RF24 with the same results. 
As you can see in the sketch, I attempt to force the `ARDUINO_ARCH_ESP8266` preprocessor variable, but it was already set. Using the `FAILURE_HANDLING` preprocessor var change nothing to the output. 

After reading so many issues in your repo, I am guessing we are missing something obvious, but as it is our first setup, we do not know this step? 

P.S.: I tried to clone the library directly from your github to ensure using the latest version. 

Thanks for your time & providing the library. 
And many many thanks to take care of *each* issue in your github, first time I see this and it is amazing!

TMRh20, 2018-12-20:
> Agreed with #244 (comment) which suggest to make a dedicated FAQ

This was an issue with the earlier ESP Arduino firmware. Do you have the most recent version?

nailyk-fr, 2018-12-20:
Sorry, I am unsure what you are talking about. I was using the 'card library' version 2.4.2-beta. Since I received your mail I am trying to upgrade the `nodemcu` bootloader, but without success. 
Also I used the exact same sketch on an arduino pro-micro and everything works so far. With the nodemcu in reiceive mode, as soon as the nodemcu try to answer to arduino, it crashes. 
I am now on the 2.4.2 version of the 'board library' with exact same behaviors.

nailyk-fr, 2018-12-20:
After upgrading the 'board library' to master one, the RF24 library to master, I am still experiencing this issue. As I read a lot of online documentation I could be confused but, using this tool https://arduino-esp8266.readthedocs.io/en/2.5.0-beta2/faq/a02-my-esp-crashes.html which allow stack traces decoding, I confirm a pattern than the nodemcu is crashing into `0x40203c74: loop_wrapper() at /home/nailyk/zz_apps/arduino-1.8.4/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 125` each time. 
If I set-up the `gettingstarted` sketch in transmission mode or, I transmit with the arduino, the same function is crashing. 
I tried to follow each function of the crash, and it was mostly about `delay` functions. As the watchdog is triggered each time, I attempted to change the values:   ```
ESP.wdtDisable();
  ESP.wdtEnable(5000);
```
but with the exact same result. 

I am obviously still doing something wrong as the exact same code works fine on the arduino, but cannot find what I am doing wrong... Sorry again :disappointed:

TMRh20, 2018-12-20:
There used to be an issue with progmem causing crashes on calls to write or to radio.printdetails. If this is not the case, it’s likely your wiring or pin declarations are wrong.

nailyk-fr, 2018-12-20:
I read a lot about that `progmem` but no idea what it is. 
Indeed wiring is probably wrong. 
Now I have sometime send & receive, which mean there are also probably some power issue. Trying to solder some decoupling & power capacitors to solve this. 
When everything will be fixed, I would like to update some pages about that issue which seems to be affecting a lot of people. Where could I put it? Is FAQ good enough for you?

panosss, 2018-12-23:
I 'm also having problems using NodeMcu V3 with NRF24L01 as receiver.
It crashes:

> STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
Exception (3):
epc1=0x4020a314 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40236370 depc=0x00000000
ctx: cont 
sp: 3ffffc20 end: 3fffffd0 offset: 01a0
>>>stack>>>
BLAH BLAH BLAH....

This is how I have them connected:
NodeMCU  Arduino mini pro
3V                VCC
G	            GND
D8               CSN
D4               CE
D5               SCK
D7               MOSI
D6              MISO

This is the code I 've uploaded in the NodeMCU:
    /*
    * Arduino Wireless Communication Tutorial
    *       Example 1 - Receiver Code
    *                
    * by Dejan Nedelkovski, www.HowToMechatronics.com
    * 
    * Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
    */
    #include <ESP8266WiFi.h>
    #include <SPI.h>
    #include <nRF24L01.h>
    #include <RF24.h>    
    RF24 radio(D4,D8); // CE, CSN

    const byte address[6] = "00001";
    void setup() {
      Serial.begin(9600);
      Serial.println("1");
      radio.begin();
      Serial.println("2");
      radio.openReadingPipe(0, address);
      Serial.println("3");
      radio.setPALevel(RF24_PA_MIN);
      Serial.println("4");
      radio.startListening();
      Serial.println("5");
      
      delay(3000);
      radio.printDetails();
      delay(3000);
    }
    void loop() {
      if (radio.available()) {        
        char text[32] = "";
        radio.read(&text, sizeof(text));        
        Serial.println(text);
      }else{
        Serial.println("radio UNavailable");
      }
    }

Any help is welcome.

nailyk-fr, 2018-12-23:
At what line does it crash? 

This is how I have them connected:
NodeMCU Arduino mini pro
3V VCC
G GND
D8 CSN
D4 CE
D5 SCK
D7 MOSI
D6 MISO 

This is the code I 've uploaded in the NodeMCU:
/*
* Arduino Wireless Communication Tutorial
* Example 1 - Receiver Code
*
* by Dejan Nedelkovski, www.HowToMechatronics.com [1]
*
* Library: TMRh20/RF24, https://github.com/tmrh20/RF24/
*/
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
RF24 radio(D4,D8); // CE, CSN

const byte address[6] = "00001";
void setup() {
  Serial.begin(9600);
  Serial.println("1");
  radio.begin();
  Serial.println("2");
  radio.openReadingPipe(0, address);
  Serial.println("3");
  radio.setPALevel(RF24_PA_MIN);
  Serial.println("4");
  radio.startListening();
  Serial.println("5");

  delay(3000);
  radio.printDetails();
  delay(3000);
}
void loop() {
  if (radio.available()) {        
    char text[32] = "";
    radio.read(&text, sizeof(text));        
    Serial.println(text);
  }else{
    Serial.println("radio UNavailable");
  }
}

Any help is welcome.

wmarkow, 2018-12-23:
@panosss 

>I 'm also having problems using NodeMcu V3 with NRF24L01 as receiver.

So I understand that NodeMCU is a receiver.

>NodeMCU Arduino mini pro

So I understand that Arduino pro mini is a transmitter.

>This is the code I 've uploaded in the NodeMCU:
>```char text[32] = "";
>   radio.read(&text, sizeof(text));
>```

I'm not quite sure but you may suffer a data integration issue. On Arduino Pro Mini the ```sizeof(char)``` is probably 1 byte, while on NodeMCU it may by 2 bytes (the size depends on the compiler; Arduino and NodeMCU uses different compilers). It may be that Arduino Pro Mini sends 32 bytes while NodeMCU wants to read 64 bytes.

panosss, 2018-12-26:
I 've also been trying with two Arduinos mini pro without success...
So, before  going crazy  :)  I 'm thinking of using a NodeMCU and a Wemos.
At NodeMCU I 'll connect them like this for use with TMRh20 's libraries:

> NRF24L01....NodeMCU
GND.........G
3.3V........3V
CSN.........D8
CE..........D4
SCK.........D5
MOSI........D7
MISO........D6

> 
NRF24L01....Wemos
CSN.........D8
CE..........D1
SCK.........D5
MOSI........D7
MISO........D6

Are they correct for use with TMRh20 's libraries?

nailyk-fr, 2018-12-26:
Problem is not the library, I got it working with an arduino nano & nodemcu (see my github repos) but mostly the wiring. 
Looks like you are using the right SPI bus, but pinout wiring is confusing. 
To debug I used the nodemcu with lot of information's on the serial, and changed the wiring 'live' to find the right pinout, most problematic one was CE (chip enabled). 
Good luck!

panosss, 2018-12-26:
I read [here](https://github.com/nRF24/RF24/issues/403) 'Don't connect CE or CS to reserved pins (GPIO 0, 2 or 15) on your ESP8266'. GPIO0=D3, GPIO2=D4, GPIO15=D8 (for both NodeMCU & Wemos)

So I chandged CE and CSN:
> NRF24L01....NodeMCU
    GND....................G
    VCC....................3V
    CSN....................D2
    CE.......................D1
    SCK....................D5
    MOSI................D7
    MISO................D6

>  NRF24L01....Wemos
    GND....................G
    VCC....................3.3V
    CSN....................D2
    CE.......................D1
    SCK....................D5
    MOSI................D7
    MISO................D6

panosss, 2019-01-01:
So many days passed (btw I wish you a happy new year) and still haven't managed to make it work!!
Is there someway to diagnose the NRF24L01 module? A test code to check if the Arduino connect to the NRF module?

nailyk-fr, 2019-01-01:
Yes, there are some functions. One to test if communication is OK, one to report basics information. 
Sorry, no computer here, but you can check this ( https://github.com/TMRh20/tmrh20.github.io/pull/7 ) PR which contain some basic troubleshooting information.

panosss, 2019-01-01:
In the above mentioned link I didn't find any info on how to test the communication. (between NRF24L01 and arduino, I suppose). Maybe you posted the wrong link?

nailyk-fr, 2019-01-01:
You are right, this is unreadable, sorry ( https://github.com/TMRh20/tmrh20.github.io/pull/7/files#diff-4dbad6e5213af349294ac35ac1a4daa7R88 )
I just saw something about the serial line working at twice the speed. Are you sure you set up the right arduino/ESP, as this serial issue could be caused by a wrong CPU clock speed?

panosss, 2019-01-01:
Both of my Arduinos are 3.3V 8MHz.
In the program, for both of them I use `Serial.begin(9600);`.

When I open the serial monitor, for the one of them I have to set it at 19200 baud and for the other at 4800 baud!!!
None is communicating at 9600 baud.
 I have no idea why is this happening.

(I tested both with the code from the link I posted. None of them seems to communicate!!)

gc61, 2019-02-27:
After days of failed attempts I finally managed to make the NRF24L01 successfully work with NodeMcu.
The problem as already reported by @panosss is that the gpio15 (D8) of NodeMcu is reserved and it would be good not to use it. In fact, however, it can be used if a PullDown resistor is inserted on this pin.
Likewise, gpio2 (D4) is also a special pin that should be maintained in High mode during start-up, and a PullUp resistor (untested solution) should be inserted.
I have seen that in many sketches found around the Internet many use the D4 as CE and the D8 as CSN and the pullup and PullDown resistances are not mentioned and this I think is the reason for the many failures in this type of connection between the modules.
By simply changing D1 as CE and D2 as CSN the module immediately worked.
I have not yet done a connection test with complex data and I will do it in the next days. Later I will report the results obtained
For now I have tested the operation through these 2 simple sketches, where one NRF24L01 is connected to the NodeMcu in transmission and the other module is connected to an Arduino Nano in reception. The connection also worked by inverting the parts, that is NodeMcu in reception and Arduino in transmission.

I hope to have been useful for those who had problems in making the connection with these modules.

These are the sketches used to test the modules:

`/+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
// NodeMcu transmitter module code
//+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#define D1 5 // CE to pin3 NRF
#define D2 4 // CSN to pin4 NRF
#define D5 14 // SCK to pin5 NRF
#define D6 12 // MISO to pin7 NRF
#define D7 13 // MOSI to pin6 NRF
//#define D8 15 // CSN to pin4 NRF onlyif is present a PullDown resistor 3.3/4.7 Kohm

typedef struct{
   char text[28]="";
   long count=0;
} Data;

RF24 radio(D1, D2); //CE-CSN
//RF24 radio(D1, D8); //CE-CSN
const byte rxAddr[6] = "00001";

void setup()
{
   Serial.begin(115200);
   radio.begin();
   radio.setRetries(15, 15);
   radio.openWritingPipe(rxAddr);

   radio.stopListening();
}

void loop()
{
   static long counter=0;
   counter++;
   Data data;
   char text[]="Hello world=>";
   strcat(data.text,text);
   data.count=counter;
   radio.write(&data, sizeof(data));
   Serial.println((String) counter +" -> "+ data.text+" "+data.count);
   delay(100);
}

//+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
// Arduino Code testing
//+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(7, 8);
typedef struct{
   char text[28]="";
   long count=0;
} Data;

const byte rxAddr[6] = "00001";
void setup()
{
  Serial.begin(115200);
  radio.begin();
  radio.openReadingPipe(0, rxAddr);
  radio.startListening();
}

void loop()
{
 static long counter=0;
 if (radio.available())
 {
  counter++;
  Data data;
  radio.read(&data, sizeof(data));
  Serial.println((String) counter +" -> " + data.text + " "+ data.count);
 }
}`

closed 2019-05-10 by TMRh20