nRF24/RF24 #450: Installation Failed on Rasspberry-Pi 3b

alimp5, 2018-08-08:
```
root@Hero-Fi:~/ali/RF24# make all
arm-linux-gnueabihf-g++ -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
In file included from /usr/include/stdio.h:859:0,
                 from utility/RPi/RF24_arch_config.h:8,
                 from utility/includes.h:7,
                 from RF24_config.h:43,
                 from RF24.cpp:10:
/usr/include/arm-linux-gnueabihf/bits/stdio.h: In function ‘int vprintf(const char*, __gnuc_va_list)’:
/usr/include/arm-linux-gnueabihf/bits/stdio.h:36:56: sorry, unimplemented: Thumb-1 hard-float VFP ABI
 vprintf (const char *__restrict __fmt, _G_va_list __arg)
                                                        ^
Makefile:42: recipe for target 'RF24.o' failed
make: *** [RF24.o] Error 1
```
note: I'm using USB-Serial converter. (I attached NRF24l01 via usb port):
```
root@Hero-Fi:~/ali/RF24# lsusb 
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```

ludiazv, 2018-09-27:
The configure script do not detect the rpi3 it fallback to RPi 1/Zero compilation ARMV6. You can force to compile with option --soc=BCM2836 (RPI2).

On the other hand, AFAIK this library do not support USB inteface. In order to work with the library you need to wire the radios to the SPI pins of the Pi. 

Regards,

OrtiMussPrime, 2019-03-11:
I got the same issue as @alimp5,

I tried both of @tomcatGu and @ludiazv solutions, nevertheless changing the CFLAG in the _Makefile.inc_ does not help, as well as the --soc=BCM2835 (RPi3) option (`./configure --driver=SPIDEV --soc=BCM2835`). 

I edited the _Makefile.inc_ with `sudo nano Makefile.inc` from:
`CFLAGS=-march=armv6zk -mtune=arm1176jzf-s [...]`
to:
`CFLAGS=-marm -march=armv6zk -mtune=arm1176jzf-s [...]`

however everytime I execute `sudo make install -B` and inspect the Makefile.inc after the same error message, the `-marm` flag has been removed, net even properly added.
```
root@kali:~/Desktop/RF24# ./configure --driver=SPIDEV --soc=BCM2835
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.
root@kali:~/Desktop/RF24# sudo make install -B
[Running configure]
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
./configure: line 66: warning: command substitution: ignored null byte in input
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=armv7l.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[OK] Finished.
arm-linux-gnueabihf-g++ -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
In file included from /usr/include/stdio.h:870,
                 from utility/RPi/RF24_arch_config.h:8,
                 from utility/includes.h:7,
                 from RF24_config.h:43,
                 from RF24.cpp:10:
/usr/include/arm-linux-gnueabihf/bits/stdio.h: In function 'int vprintf(const char*, __gnuc_va_list)':
/usr/include/arm-linux-gnueabihf/bits/stdio.h:39:60: sorry, unimplemented: Thumb-1 hard-float VFP ABI
 vprintf (const char *__restrict __fmt, __gnuc_va_list __arg)
                                                            ^
make: *** [Makefile:42: RF24.o] Error 1
```

Seems like it does ignore the set driver via the config file. I tried it with the MRAA driver as well.
Nevertheless the same result.

Additionally I should mention that I have [connected the radio via SPI pins](https://tmrh20.github.io/RF24/RPi.html) to my RPi3.
Since both solutions didn`t work for me you guys got any other possible solutions?

TMRh20, 2020-03-29:
Old issue, but for the record, after running configure, sudo make install should be run without the -B switch, else the configure script is re-run with default values.

Fixed in https://github.com/nRF24/RF24/commit/b323f535aa9644d988370fc6989a2ef9f85aa469

closed 2020-03-29 by TMRh20