nRF24/RF24 #451: AckPayload Issue

FaridGad, 2018-08-12:
Hi everyone,

I seem to be having an issue with using the AckPayload to establish a two way connection between two arduinos connected to their own nRF24L01 modules. The first Arduino (Uno), the Master, sends data to the slave (Arduino Mega 2560) which in return sends data back to the master.

**Master code:**
```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <printf.h>

 int msg[1] = {1};
int rec[1] = {5};
bool stat = true;
RF24 radio(7,10);
const uint64_t pipe[1] = {0xF0F0F0F0E1LL};

void setup()
{
  Serial.begin(9600);
  radio.begin();
  delay(100);
  radio.setAutoAck(true);
  radio.enableAckPayload();
  radio.enableDynamicPayloads();
  radio.stopListening();
  radio.openWritingPipe(pipe[0]);
  radio.setRetries(15,15);
  Serial.flush();
 }
void loop()
{
  radio.flush_rx();
if(stat)
{
    if(radio.write(msg,sizeof(msg)))
    {
      Serial.print( msg[0] );
      Serial.println("...tx success");
      if(radio.isAckPayloadAvailable())
      {
        radio.read(rec,sizeof(int));
        Serial.print("received ack payload is : ");
        Serial.println(rec[0]);
      }
      else
      {
        Serial.println("AckPayload not available....");
      }
      msg[0]+=3;;
    if(msg[0]>=100)
    {msg[0]=1;}
    }
 }
}
```

**Slave code:**
```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <printf.h>
#include <avr/wdt.h>

const uint64_t pipe[1]= {0xF0F0F0F0E1LL};
RF24 radio(7,30);
int rec[1] = {2};
int red;

//void(* resetFunc) (void) = 0;

void setup()
{
  Serial.begin(9600);
  radio.begin();
  delay(100);
  radio.setAutoAck(true);
  radio.enableAckPayload();
  radio.enableDynamicPayloads();
  radio.openReadingPipe(1,pipe[0]);
  radio.startListening();
  radio.writeAckPayload( 1, rec, sizeof(int) );
  radio.setRetries(15,15);
}

void loop()
{
  if ( radio.available() ) {
    radio.writeAckPayload( 1, rec, sizeof(int) );
    radio.read( &red,sizeof(red) );rec[0]+=2;
    Serial.print("integer got is : ");
    Serial.println(red);
}
else
{
  Serial.println("Radio not available");
}
}
```

The code works and data is exchanged back and forth between the Arduinos. However, if I reset or reupload the slave code, the `radio.isAckPayloadAvailable()` method fails and the master doesn't receive any data back from the slave. I have tried using the `radio.flush_rx()` and `radio.flush_tx()` to no avail. The only way to get the two way communication working again is if I reset the slave a second time. I know it seems like a non-issue, but the communication between the arduinos is a small part in a much larger project and every component needs to run perfectly for the project to work smoothly.

Thanks in advance for your assistance.

TMRh20, 2018-08-15:
Does it work every time if you power cycle the arduino? Where did u get your modules?

FaridGad, 2018-08-15:
It works alternately. Power cycle once, it works. Power cycle again, it doesn't. And so on and so forth.
The ` if(radio.isAckPayloadAvailable())` is the one causing this problem. It alternates between 'true' and 'false' every time I power cycle the arduino.

I got the modules from the university. So I'll ask them about it.

FaridGad, 2018-08-15:
@wmarkow I commented the `toggle_features()` in the `rf24.begin()` function and now it works! Slave transmits successfully to the master, even after resetting!

Thanks for the help!

wmarkow, 2018-08-16:
@FaridGad, ```toggle_features()``` should be called only once, to activate some features that are normally deactivated after power up the RF24 chip. It looks a bit strange that it works now when you totally commented out this method.

In issue #401 I have created a commit in my branch. Could you please use the RF24v library from that branch and retest if it now works? Just read the whole #401 and you will know what to do.

FaridGad, 2018-08-22:
@wmarkow Sorry about the late reply. Thesis work and whatnot. You were right when you mentioned that the `toggle_features()` should be called once. When I disconnected both the arduinos from the PC and reconnected them again, there was no data transmitted. I had to add `toggle_features()` so that data transmission can start and then removed it so that the data transmission can continue even after resetting the slave via the reset button (Complicated, I know).

I checked the commit and made the changes to my RF24v library. And everything is working fine now.

I appreciate all the work you've put into this. Thanks a lot!

closed 2019-05-10 by TMRh20