nRF24/RF24 #456: NRF24L01 fails to write

nrf24Noob, 2018-10-03:
Hello, I bought 3 kuman nrf24l01 modules for my arduinos uno. I can't get any of them to work.
radio.printDetails() of the first module:
`STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xe8e8f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x21
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN
`
radio.printDetails() of the second module:
`STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xe8e8f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x21
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN
`

When I try to use the getting started sketch it just keeps to fail transmitting and times out.
The modules are wired to arduinos' 5V since kuman in their website claims the modules can be wired directly to arduinos' 5V.

Any advice?

Avamander, 2018-10-03:
The power is definetly wrong if you're speaking of these modules: http://www.kumantech.com/kuman-3pcs-nrf24l01-24ghz-antenna-wireless-transceiver-module-for-arduino-raspberry-pi-k19_p0233.html

nrf24Noob, 2018-10-03:
@Avamander I'm talking about these modules: http://www.kumantech.com/kuman-nrf24l01palna-antenna-wireless-rf-transceiver-module-for-arduino-ky67_p0387.html

Avamander, 2018-10-03:
And for those 5V is too much as input voltage as well. 
> ● Operating voltage: 3 ~ 3.6V Max

nrf24Noob, 2018-10-03:
@Avamander it says
* The module has 5V tolerant inputs which allows direct connection of SPI pins to the Arduino.

Avamander, 2018-10-03:
Yep, but that's 5V tolerant logic input, not power input.

Avamander, 2018-10-03:
A 3.3V one that provides 500mA or more.

PhillyNJ, 2018-10-03:
> which external regulator should I use?

These modules must use 3.3v. They will not work at 5v. 

Regards,

Phil

Sent from Phil's iPhone

closed 2019-05-10 by TMRh20