nRF24/RF24 #458: can't get pingpair_dyn.py

erpadmin, 2018-10-11:
I modified pingpair to use radio = RF24(22, 24);
the are the board pins for CE then CSN

pi@pystick:~/RF24/examples_linux $ python pingpair_dyn.py
pyRF24/examples/pingpair_dyn/
Traceback (most recent call last):
  File "pingpair_dyn.py", line 67, in <module>
    radio.begin()
RuntimeError: unidentifiable C++ exception

erpadmin, 2018-10-11:
`
pi@pystick:~/RF24 $ make all && sudo make install
make: Nothing to be done for 'all'.
[Installing Libs to /usr/local/lib]
[Installing Headers to /usr/local/include/RF24]
`

`
pi@pystick:~/RF24/pyRF24 $ python setup.py build && sudo python setup.py install
running build
running build_ext
running install
running bdist_egg
running egg_info
writing RF24.egg-info/PKG-INFO
writing top-level names to RF24.egg-info/top_level.txt
writing dependency_links to RF24.egg-info/dependency_links.txt
reading manifest file 'RF24.egg-info/SOURCES.txt'
writing manifest file 'RF24.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_ext
creating build/bdist.linux-armv6l/egg
copying build/lib.linux-armv6l-2.7/RF24.so -> build/bdist.linux-armv6l/egg
creating stub loader for RF24.so
byte-compiling build/bdist.linux-armv6l/egg/RF24.py to RF24.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying RF24.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
writing build/bdist.linux-armv6l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating 'dist/RF24-1.3.1-py2.7-linux-armv6l.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing RF24-1.3.1-py2.7-linux-armv6l.egg
Removing /usr/local/lib/python2.7/dist-packages/RF24-1.3.1-py2.7-linux-armv6l.egg
Copying RF24-1.3.1-py2.7-linux-armv6l.egg to /usr/local/lib/python2.7/dist-packages
RF24 1.3.1 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/RF24-1.3.1-py2.7-linux-armv6l.egg
Processing dependencies for RF24==1.3.1
Finished processing dependencies for RF24==1.3.1
`

olixr, 2018-10-23:
I went through all the installation steps as well. Trying to run the pingpong example on RPi 2 b+, I encountered same error:

`
    radio.begin()
RuntimeError: unidentifiable C++ exception
`
Do we have any idea what to fix?

closed 2020-03-29 by TMRh20