nRF24/RF24 #460: Mesh Problems with RF24

pmcgillan, 2018-10-24:
Recently, I started replacing my existing network of NRF24L01 data collectors with a MESH network. The existing system was not the most reliable and I need to send from a couple of sensors that were at the extreme distance. All radios have been modified with a 10 µf capacitor and a dipole antenna. Even with the mods, they could not send more than randomly to the older master station. Now I am getting fairly reliable data, but not perfect. Some stations seem to go on break for hours at a time, before putting themselves back online. The master is running on an RPi, with latest Raspbian. All the data collectors are running on Unos.

Avamander, 2018-10-24:
How have you configured the radios?

TMRh20, 2018-10-24:
> Here is the UNO version of one of the remotes.
>
> Attic.txt

Sounds like there might be some interference. I would try running the radios on a different channel and see if that clears it up. Default is 97.
  
Use mesh.begin(<channelNum>)

Avamander, 2018-10-24:
You can also try decoupling more, 10uF isn't much. A few of my setups have 10mF to get stable results.

pmcgillan, 2018-10-24:
I put mesh.begin(99); in all the correct places. Seems to be working better already. Time will tell.

closed 2019-05-10 by TMRh20