nRF24/RF24 #464: Blank payload issue in multiceiver mode test

nkkim91, 2018-11-04:
Hello, 

As mentioned, tried to test with multiple Arduino + nRF24L01p devices for "multiceiver" mode operation. but, only up to 3 TX devices are working fine. Once 4th TX device is added, it receives the "Blank payload" unlike the others. (Definitely not a device specific issue of the 4th TX device)

Have 4 TX mode Arduino Uno and 1 RX mode Arduino Uno

- Uno board is not the genuine, but compatible one and have verified it with other examples
- Used nRF24L01p which are genuine and also copy. but, couldn't see any difference btw. two so far.
- Added 10 uF (Tantal) on VCC(3.3V)
- USB power through USB hub connected to 5V (3A) power supply

The test code was slightly customized from the example code for the multiceiver mode test.

https://github.com/nkkim91/nrf24l01p_multiceiver

- 250 Kbps (just in case)
- PA High
- 3 sec delay for tx
- 16 bytes payload
- split the code for rx and tx 
  * nrf24l01p_multiceiver/nrf24l01p_multiceiver_tx
  * nrf24l01p_multiceiver/nrf24l01p_multiceiver_rx
- EEPROM to store the RF address and channel info.

The code hasn't clean up so. could have some problem. but, at least working up to 3 TX + 1 RX.

I'd like to test with the max. 5 TX devices as noted in the spec. of the chip (nRF24l01p). but, tied up with the issue (Blank Payload) when adding 4th device.

Could you comment on my issue if you are aware of anything like this ?

nkkim91, 2018-11-06:
Looks the issue is connected to the max. number of TX ACK Payload FIFOs which is 3 in nRF24l01p.

When the TX device sends the packet in every 5 seconds which is longer than initial trial (3 sec.), the "Blank payload" or "Write failed" (happens infrequently compared to blank payload issue) is seen less frequently. but, 5 seconds delay/period for TX isn't acceptable solution. And still would like to know how to resolve the issue with reasonable solution but increasing the delay from TX.

Handling the ack payload in the isr of RX could be the solution ?

TMRh20, 2020-08-11:
This is a problem with coding and design rather than an issue with the library. Ack payload functionality is there and it works.

One main thing to consider is that as per the examples, ack payloads should be written to the FIFO **before** a normal payload is received. Although certain MCUs may be quick enough in some cases, it is not reliable and not supported otherwise.

Closing issue.

closed 2020-08-11 by TMRh20