nRF24/RF24 #466: Wemos D1 mini + NRF24L01 + Nano - auto ack fail

bgray817, 2018-11-19:
I have a process utilizing Blynk on Wemos w NRF24L01+ to receive/store user requests. An Arduino Nano w NRF24L01+ comes out of deep sleep on intervals, writes a message to Wemos to check ofor work. Wemos should Auto Ack with response.

This process is not working. When I stripped it down and usd a very simple Tx/Rx program pair, I found that the Ack appeared to work for three tx's, then it would fail. This is a consistent result. I am posting the code and resulting Serial output below. I have read online that the Wemos/ESP8266 "has issues" with auto ack. Can someone help me understand/solve this issue? Thanks in advance.

Tx Master Nano code:
`// SimpleTxAckPayload - the master or the transmitter

#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

#define CE_PIN   9
#define CSN_PIN 10

//const byte slaveAddress[5] = {'R','x','A','A','A'};
const uint64_t slaveAddress = 0xABCDABCD71LL;
RF24 radio(CE_PIN, CSN_PIN); // Create a Radio

char dataToSend[10] = "Message 0";
char txNum = '0';
int ackData[2] = {-1, -1}; // to hold the two values coming from the slave
bool newData = false;

unsigned long currentMillis;
unsigned long prevMillis;
unsigned long txIntervalMillis = 1000; // send once per second

//===============

void setup() {

    Serial.begin(9600);
    Serial.println(F("Source File /mnt/sdb1/SGT-Prog/Arduino/ForumDemos/nRF24Tutorial/SimpleTxAckPayload.ino"));
    Serial.println("SimpleTxAckPayload Starting");

    radio.begin();
    radio.setDataRate( RF24_250KBPS );

    radio.enableAckPayload();

    radio.setRetries(3,5); // delay, count
    radio.openWritingPipe(slaveAddress);
}

//=============

void loop() {

    currentMillis = millis();
    if (currentMillis - prevMillis >= txIntervalMillis) {
        send();
    }
    showData();
}

//================

void send() {

    bool rslt;
    rslt = radio.write( &dataToSend, sizeof(dataToSend) );
        // Always use sizeof() as it gives the size as the number of bytes.
        // For example if dataToSend was an int sizeof() would correctly return 2

    Serial.print("Data Sent ");
    Serial.print(dataToSend);
    if (rslt) {
        if ( radio.isAckPayloadAvailable() ) {
            radio.read(&ackData, sizeof(ackData));
            newData = true;
        }
        else {
            Serial.println("  Acknowledge but no data ");
        }
        updateMessage();
    }
    else {
        Serial.println("  Tx failed");
    }

    prevMillis = millis();
 }

//=================

void showData() {
    if (newData == true) {
        Serial.print("  Acknowledge data ");
        Serial.print(ackData[0]);
        Serial.print(", ");
        Serial.println(ackData[1]);
        Serial.println();
        newData = false;
    }
}

//================

void updateMessage() {
        // so you can see that new data is being sent
    txNum += 1;
    if (txNum > '9') {
        txNum = '0';
    }
    dataToSend[8] = txNum;
}
`

Rx Slave Wemos code;
`
// SimpleRxAckPayload- the slave or the receiver

#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

//#define CE_PIN   4
//#define CSN_PIN 15
#define CE_PIN   0
#define CSN_PIN 16
//#define CE_PIN   9
//#define CSN_PIN 10

//const byte thisSlaveAddress[5] = {'R','x','A','A','A'};
const uint64_t thisSlaveAddress = 0xABCDABCD71LL;
RF24 radio(CE_PIN, CSN_PIN);

char dataReceived[10]; // this must match dataToSend in the TX
int ackData[2] = {109, -4000}; // the two values to be sent to the master
bool newData = false;

//==============

void setup() {

    Serial.begin(9600);

    Serial.println("SimpleRxAckPayload Starting");
    radio.begin();
    radio.setDataRate( RF24_250KBPS );
    radio.openReadingPipe(1, thisSlaveAddress);

    radio.enableAckPayload();
    radio.writeAckPayload(1, &ackData, sizeof(ackData)); // pre-load data

    radio.startListening();
}

//==========

void loop() {
    getData();
    showData();
}

//============

void getData() {
    if ( radio.available() ) {
        radio.read( &dataReceived, sizeof(dataReceived) );
        updateReplyData();
        newData = true;
    }
}

//================

void showData() {
    if (newData == true) {
        Serial.print("Data received ");
        Serial.println(dataReceived);
        Serial.print(" ackPayload sent ");
        Serial.print(ackData[0]);
        Serial.print(", ");
        Serial.println(ackData[1]);
        newData = false;
    }
}

//================

void updateReplyData() {
    ackData[0] -= 1;
    ackData[1] -= 1;
    if (ackData[0] < 100) {
        ackData[0] = 109;
    }
    if (ackData[1] < -4009) {
        ackData[1] = -4000;
    }
    radio.writeAckPayload(1, &ackData, sizeof(ackData)); // load the payload for the next time
}`

Tx serial output:
`Source File /mnt/sdb1/SGT-Prog/Arduino/ForumDemos/nRF24Tutorial/SimpleTxAckPayload.ino
SimpleTxAckPayload Starting
Data Sent Message 0  Acknowledge data 102, 0

Data Sent Message 1  Acknowledge data 102, 0

Data Sent Message 2  Acknowledge data 102, 0

Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed
Data Sent Message 3  Tx failed`

Rx serial output:
`Data received Message 0
 ackPayload sent 105, -4004
Data received Message 1
 ackPayload sent 104, -4005
Data received Message 2
 ackPayload sent 103, -4006
Data received Message 3
 ackPayload sent 102, -4007
Data received Message 3
 ackPayload sent 101, -4008
Data received Message 3
 ackPayload sent 100, -4009
Data received Message 3
 ackPayload sent 109, -4000
Data received Message 3
 ackPayload sent 108, -4001
Data received Message 3
 ackPayload sent 107, -4002
Data received Message 3
 ackPayload sent 106, -4003`

Avamander, 2018-11-19:
You might have counterfeit modules that mess up autoack or dynamic payloads. Disable both and see if that helps.

bgray817, 2018-11-19:
Looks like I would disable auto ack by issuing `setAutoAck (false)`
How would I disable dynamic payloads?

What exactly will these accomplish? Will the Write continue to get a response in the ack payload?

Thanks for responding.

Avamander, 2018-11-19:
> What exactly will these accomplish?

Disabling features that are broken on your devices will simply avoid the issue.

> Will the Write continue to get a response in the ack payload?

Not automatically.

You can test individual features with the examples, every single one should run flawlessly.

bgray817, 2018-11-20:
Tried to compile the pingpair_ack.ino example for Wemos and got the following:
`In file included from C:\Users\PATRIC~1\AppData\Local\Temp\arduino_modified_sketch_131254\sketch_nov19a.ino:20:0:

C:\Users\Patrick Gray\Documents\Arduino\libraries\RF24/printf.h: In function 'void printf_begin()':

C:\Users\Patrick Gray\Documents\Arduino\libraries\RF24/printf.h:31:29: error: 'fdevopen' was not declared in this scope

   fdevopen( &serial_putc, 0 );`

TMRh20, 2018-11-21:
On the tx master, try changing this:
``` int ackData[2] = {-1, -1}; // to hold the two values coming from the slave ```
to this
``` int ackData[4] = {-1, -1}; // to hold the two values coming from the slave ```

and edit the showData function:
```         Serial.print(ackData[0]);
        Serial.print(", ");
        Serial.println(ackData[2]); 
```

You can see the problem by printing the sizeof(ackData); on the ESP device vs the nano.
On the 32 bit device, it is 8 bytes, and on the nano it is 4.

If this is not accounted for, the incoming data is not cleared from the radio, (because the entire payload is not read out) and the RX buffer fills up, leading to the erroneous behavior after 3 payloads.

wmarkow, 2018-11-21:
@TMRh20, @bgray817, maybe it would be good to replace on both master and slave this:
```int ackData[2]```
with:
```int16_t ackData[2]```
?

**int16_t** will always have two bytes no matter which compiler/device we have. In such a case the size of ```ackData``` will be always 4 bytes.

bgray817, 2018-11-22:
Thank you so much! Success.

closed 2019-05-10 by TMRh20