nRF24/RF24 #467: nrf24l01 mesh using atmega8L

Zuputmer, 2018-12-04:
Hey, i need some help in getting this library to work with the atmega8L. Can you please advise on how i could port this library to get it to work with it? I plan to use embedded c on the atmega.

Zuputmer, 2018-12-05:
Any help would be greatly appreciated. I am trying to use this with the atmega8l but cannot get it to work

akatran, 2018-12-07:
@Zuputmer One magic function `GPIO_getPort` does the port selection in file 
      [RF24/utility/ATXMegaD3/gpio_helper.c](https://github.com/nRF24/RF24/blob/master/utility/ATXMegaD3/gpio_helper.c)

It is a simple if waterfall that returns a pointer to the port.

The magic with xmega is that ports are something like "symmetric" which means that all ports have the same properties.

Good news are that you can also do the same.

Your if waterfall will be much smaller as the atmega8L has 3 ports (PB, PC, PD), and if you use my code as a starting point, adjust the offsets to the *pin_bm.

Hope that helped!

Zuputmer, 2018-12-07:
@akatran thank you for the help. I just found out that the mesh library comes to 12kb when compiled onto a program, and my atmega8L has only 8kb of onboard flash memory. I will thus move to the 328p, which is already supported by the library. This should make my life easier :)

closed 2018-12-10 by Zuputmer